joan2937/pigpio #263: Baud Rate

cc96cc, 2019-02-05:
How can I set a much higher baud rate using pigpio.bb_spi_open?
I would like baud rates greater than 50 MHz.

guymcswain, 2019-02-05:
Not much higher than maybe 50KHz.  This comes from my experience using bit-bang serial.  The maximum sample rate for bb I/O is 1MHz so theoretically, the highest signal rate would be 500KHz.  However, due to sampling jitter, created by high traffic on the internal DMA channels, to get a reliable (error-free) signal you need to be well under 100MHz.

cc96cc, 2019-02-06:
Ok is there any other way to get the baud rate up to 50 MHz?

joan2937, 2019-06-16:
@guymcswain is correct.

There is no way 50 MHz will work on the Pi.  I think even the Pi's UART hardware is limited to 4 MHz.

closed 2019-06-16 by joan2937