joan2937/pigpio #266: patch for pigpiod to work with 64bit gentoo

zwim, 2019-02-27:
Hello,

I am using a gentoo 64 bit system an an rpi 3 model B V1.2 (based on sakaki https://github.com/sakaki-/gentoo-on-rpi3-64bit).
With the pach applied below for pipgoid.c everything works fine for me :-)

As I am not familiar with github i have appended the patch below and as a zip file.
[pigpio.c.patch.zip](https://github.com/joan2937/pigpio/files/2912023/pigpio.c.patch.zip)

Despite not having a different linux distribution, I cannot check if the patch breaks something on 32bit.
Please could someone confirm, that the patch does not break 32bit systems?

Thanks and have joy with the patch

---------------------------------------

--- pigpio/pigpio.c	2019-02-27 20:24:18.389655115 +0100
+++ pigpio-v68/pigpio.c	2019-02-27 20:19:07.712302977 +0100
@@ -1511,6 +1511,47 @@
    return valid;
 }
 
+
+
+static unsigned get_dt_ranges(const char *filename, unsigned offset)
+{
+   unsigned address = ~0;
+   FILE *fp = fopen(filename, "rb");
+   if (fp)
+   {
+      unsigned char buf[4];
+      fseek(fp, offset, SEEK_SET);
+      if (fread(buf, 1, sizeof buf, fp) == sizeof buf)
+         address = buf[0] << 24 | buf[1] << 16 | buf[2] << 8 | buf[3] << 0;
+      fclose(fp);
+   }
+   return address;
+}
+
+unsigned bcm_host_get_peripheral_address(void)
+{
+   unsigned address = get_dt_ranges("/proc/device-tree/soc/ranges", 4);
+   return address == ~0 ? 0x20000000 : address;
+}
+
+unsigned bcm_host_get_peripheral_size(void)
+{
+   unsigned address = get_dt_ranges("/proc/device-tree/soc/ranges", 8);
+   return address == ~0 ? 0x01000000 : address;
+}
+
+unsigned bcm_host_get_sdram_address(void)
+{
+    unsigned address = get_dt_ranges("/proc/device-tree/axi/vc_mem/reg", 8);
+    return address == ~0 ? 0x40000000 : address;
+}
+
+
+
+
+
+
+
 /* ----------------------------------------------------------------------- */
 
 static char * myTimeStamp()
@@ -13338,6 +13379,54 @@
       }
 
       fclose(filp);
+
+
+      if ( piCores == 0)
+          {
+	  if ( !strncasecmp("processor\t: 0", buf, 10) )
+	  	{
+	  	piCores=1;
+                piCores = 1;
+                pi_peri_phys = 0x20000000;
+                pi_dram_bus  = 0x40000000;
+                pi_mem_flag  = 0x0C;
+                }
+
+	  else if ( strncasecmp( "processor\t: 3", buf, 10) )
+	  	{
+	  	piCores=4;
+		pi_peri_phys = bcm_host_get_peripheral_address();
+		pi_dram_bus  = bcm_host_get_sdram_address();
+		pi_mem_flag  = 0x04;
+		}
+	  }
+
+printf("peripheral address %x\n", pi_peri_phys);
+printf("sdram_address %x\n", pi_dram_bus);
+      printf("piCores: %d\n", piCores);
+
+      if ( rev == 0 )
+      	  {
+	  filp =  fopen( "/proc/device-tree/system/linux,revision", "r" );
+	  if (filp != NULL)
+      		{  
+            	while (fgets(buf, sizeof(buf), filp) != NULL)
+	            {  
+		    rev = (unsigned)buf[0];
+		    rev <<= 8;
+		    rev |= (unsigned)buf[1];
+		    rev <<= 8;
+                    rev |= (unsigned)buf[2];
+		    rev <<= 8;
+		    rev |= (unsigned)buf[3];
+		    }
+		}
+	  fclose( filp );
+	  }
+	
+      printf("revision: %8x\n", rev );
+
+
    }
    return rev;
 }

cinderblock, 2019-04-22:
Welcome to github. Might I humbly suggest that you edit your issue and put your **code** into a code block with triple backticks: (` ``` ` ). Note it looks like you tried to put the code into a block format but inadvertently put each line of code into its own block making it very difficult to read. Please use the "Preview" functionality to ensure that your code is readable. You might also add a `C++` or `patch` language tag to get proper highlighting.

While you're at it, you might consider adding a description of the changes and why they are necessary. Most patches without any justification seem to just get lost in the noise.

PS. I'm just a humble githubitizen and have no affiliation with pigpio.

PPS. I had no idea there was a big Gentoo community for the Pi. I love Gentoo and got it working on my own forever ago once. Have stayed on Debian images since to save the headache but always considered the possibility of going back... With distcc and proper cross compiling setup it might not be so bad!

sakaki-, 2019-12-05:
FYI, the current [gentoo-on-rpi-64bit](https://github.com/sakaki-/gentoo-on-rpi-64bit) release (v1.5.2) contains `pigpio` bundled (with the `pigpiod` service activated by default), and has (I believe) all necessary [patches](https://github.com/sakaki-/genpi64-overlay/blob/61973d3039e6393b1b8fc29cd844a6c7545b2b08/dev-libs/pigpio/pigpio-71-r1.ebuild#L22) for 64-bit operation (on RPi3B, 3B+ and 4B) applied.

It replaces there `wiringpi`, now [deprecated](http://wiringpi.com/wiringpi-deprecated/) by its author.

Best, sakaki

github-k8n, 2019-12-05:
Interesting.

@joan2937 , any chance that those patches referenced above in the post of @sakaki- can be included in the code?
Otherwise @sakaki- , do you think it would make sense to include them in the official gentoo version? (I'd prefer to have them included upstream (aka here...) but will include them in the official ebuild if needed, need to bump to 72 anyways ... )
I guess those patches do not break anything on the 32bit versions?

sakaki-, 2019-12-07:
I think just a v72 verbump should do it, the patch I included was just an additional required commit wrt v71.

Oh and also btw, I found I needed to set the shipped `/etc/conf.d/pigpiod` file to include:

```bash
PIGPIOD_OPTS="-l -n 127.0.0.1"
```

to have the service work out-of-the-box with `pigs`.

github-k8n, 2019-12-07:
https://github.com/gentoo/gentoo/pull/13905
(waiting for it to be comitted)

(also added the -n parameter)

Might be better if pigpiod would allow to specify IPv6 addresses with -n (currently only allows ipv4), to make it use IPv6 you have to ditch the -n parameter which makes it listen only on "::1" but allows connections from any IP (if that really is what "-n" does ...)

I was originally under the wrong impression that it would listen on both or that the "::1" listener should also work for ipv4 (similar to what it does when you bind to "::"), that is not the case and both 127.0.0.1 and "::1" should need to be set up automatically to listen when specifying "-l"

Though it seemed to work fine for me when only using (-l) , maybe because i am also using IPv6 in my network?

guymcswain, 2020-01-04:
@github-k8n , 
Let me explain that Joan does not have time to work on pigpio.  I'm helping as a collaborator on pigpio but admit this patch is a lot for me to wrap my head around and I'm currently prioritized on bug fixes and, secondarily, on small enhancements that have wide community support.

It would be helpful to me if you could explain if these changes are specific to gentoo-64bit or are they applicable to all 64-bit distributions?

As for your comments above on IPv6, we are engaged with a contributor who provided basic support for IPv6 but left a hole on accepting either "::1" or "127.0.0.1" when the "-l" option is invoked.  This will be fixed.  The "-n" option is for providing a 'whitelist' of IP addresses but it was not included in the original IPv6 support.  Instead it now essentially forces IPv4 only.  Including IPv6 on "-n" would require some network stack help.
Guy

github-k8n, 2020-01-04:
Hi @guymcswain , I don't really have any insight into the code/patches at all.
Regarding the -n/-l confusion, I'm happy to specify the default config settings in the ebuild as required, however it would be nice to have this fixed and working consistently in future versions.

Regarding the patch this issue is about, I would have to point you to @sakaki- , if I understand the comment above correctly, the patch should be already included in some way or form in v72, thus in this context I think the issue could be closed?

guymcswain, 2020-01-15:
@zwim , @sakaki- ,
Can you please report the output from `/proc/cpuinfo` running 64-bit gentoo on RPi?  I'm trying to see if there is correlation with this 64-bit issue #316

sakaki-, 2020-01-15:
Sure, here's the output on an RPi4 (4GB) running v1.5.3 of [gentoo-on-rpi-64bit](https://github.com/sakaki-/gentoo-on-rpi-64bit):
```console
demouser@pi64 ~ $ cat /proc/cpuinfo
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03111
Serial		: 10000000--------
Model		: Raspberry Pi 4 Model B Rev 1.1
```

guymcswain, 2020-01-15:
Thank you!  These two issues are indeed related.  Two methods of patching have been offered.  I believe there is yet a third method that is more in line with the RaspberryPi.org documentation.  I'll soon propose a solution to the library owner and put that patch on the 'develop' branch for everyone to review.

guymcswain, 2020-01-18:
@zwim , @sakaki- , May I request that you download the proposed patch for this issue and test it on your systems? 
Please run the following on your raspberry pi:
```
wget https://github.com/joan2937/pigpio/archive/v7301.zip
unzip v7301.zip
cd pigpio-7301
make
sudo make install
```
Test your system as normal.  If for some reason it fails, capture the initialization log by doing the following:
1) Stop the currently running instance of pigpio.
2) From a working directory, run:
`sudo pigpiod -c4 -g 2> pigpiod.log`
3) Allow a few seconds to complete initialization, then Cntrl-C (SIGINT) to abort.
4) Run
`head pigpiod.log`

If there was any error in the hardware revision decode it should appear in this output.

Thank you very much!

zwim, 2020-01-19:
Hello,

I have tested the v7301.zip on a  Raspberry Pi 3 Model B Rev 1.2 with 64-bit.
It seems to work correctly.

Thanks

guymcswain, 2020-01-19:
Excellent.  Thank you for the test confirmation.

@sakaki- , my plan is to merge this into the pigpio master branch by early Feb unless you tell me 64bit Gentoo would need it sooner.

sakaki-, 2020-01-21:
Tested your v7301.zip today on RPi3B+, RPi4B; seems working OK.

@guymcswain happy to wait for your merge in early Feb; this will be before my next planned release of the [gentoo-on-rpi-64bit](https://github.com/sakaki-/gentoo-on-rpi-64bit) image (v1.5.4) goes into soft freeze.

commit bab05ceb (2020-02-03): Merge pull request #321 from joan2937/Release-v74

Release v74: closes #274, closes #298, closes #266, closes #316, closes #223.
closed 2020-02-03 by guymcswain