joan2937/pigpio #267: Pigpio on more Raspberry's

djdacore, 2019-03-06:
Hi Joan!

I have the following situation.
**1 Raspberry (pi1)** as the receiver (8 port relays with lights connected)
**1 Raspberry (pi2)** sender (with 3 input buttons)
**1 Raspberry (pi3)** sender (with 3 input buttons)

I would like to send signals with the inputs from pi2 and pi3 to one receiver (pi1). 
So both (pi2 and pi3) are in the same network connected with PIGPIO(D) via port 8888 to pi1.
All is working well. When I send inputs on pi2 or pi3 the lights will go on. But, when I start both of the scripts (same scripts) on both pi's the same time it won't work. The lights will no longer go on and off. You can hear the relay clicking, but nothing is happening anymore.

I restart one of the pi's (so with only one pi connected to the receiving pi) and all is working again. So for a reason my scripts are conflicting eachother? Is there a way to get it working with 2 sending pi's and only one receiving pi?

the script I am using:

```
#!/usr/bin/env python
import pigpio
import os
import time

#set the pigpio pi's to the network
pi1 = pigpio.pi()
if not pi1.connected:
   os.system('sudo shutdown -r now')

pi2 = pigpio.pi('192.168.178.80')
if not pi2.connected:
   os.system('sudo shutdown -r now')

#set the inputs on the local Pi
pi1.set_glitch_filter(18, 300000)
pi1.set_glitch_filter(23, 300000)
pi1.set_glitch_filter(24, 300000)
pi1.set_mode(18,pigpio.INPUT)
pi1.set_mode(23,pigpio.INPUT)
pi1.set_mode(24,pigpio.INPUT)
pi1.set_pull_up_down(18, pigpio.PUD_UP)
pi1.set_pull_up_down(23, pigpio.PUD_UP)
pi1.set_pull_up_down(24, pigpio.PUD_UP)

#Testing 1 2 3 if all the lights go on/off
pi2.write(4, 1)
pi2.write(17, 1)
pi2.write(18, 1)
pi2.write(22, 1)
pi2.write(23, 1)
pi2.write(27, 1)
pi2.write(24, 1)
time.sleep(3)
pi2.write(4, 0)
pi2.write(17, 0)
pi2.write(18, 0)
pi2.write(22, 0)
pi2.write(23, 0)
pi2.write(27, 0)
pi2.write(24, 0)

#Waiting for an INPUT of buttons (while loop)
while True:
        if pi1.read(18):
             pi2.write(17, 0)
             pi2.write(21, 0)
        else:
             pi2.write(17, 1)
             pi2.write(21, 1)
             time.sleep(5)
             pi2.write(17, 0)
             pi2.write(21, 0)

        if pi1.read(23):
             pi2.write(17, 0)
             pi2.write(27, 0)
        else:
             pi2.write(17, 1)
             pi2.write(27, 1)
             time.sleep(5)
             pi2.write(17, 0)
             pi2.write(27, 0)

        if pi1.read(24):
             pi2.write(17, 0)
             pi2.write(22, 0)
        else:
             pi2.write(17, 1)
             pi2.write(22, 1)
             time.sleep(5)
             pi2.write(17, 0)
             pi2.write(22, 0)

GPIO.cleanup()
```

guymcswain, 2019-03-07:
Pigpio(d) is working perfectly!
Hint:  When you push a button on one pi, what is the other pi doing?

djdacore, 2019-03-07:
The other pi has a few Ports connected to a relay with lights connected.
It has also an i2c 4x20 display connected. 
But no matter what i connect, all is not working from two pi’s with the same script running to one receiving pi.
Even if I connect only one lightbulb to the relay. It looks like that the receiving pi can only handle one sending pi?

guymcswain, 2019-03-07:
What I'm trying to point out is that your script on the second pi is always writing 0 to the relays - I assume the switches are momentary type.  So the first pi will write 1 but immediately the second pi will over write the relay to 0 again.  I suspect that if you press the same switch on both pi simultaneously that you will observe the light to come on.

djdacore, 2019-03-07:
Very good idea, I was not thinking about that. Thanks for your help I will try this tonight immediately!

djdacore, 2019-03-07:
Just to let you know that it works great now!

guymcswain, 2019-03-07:
I'm glad I could help.  You may have recognized now that the 'while forever' loop is not a good way to implement your idea.  Pigpio provides callbacks for asynchronous events - like button press - that you should explore.

joan2937, 2019-06-12:
Thanks for resolving that issue @guymcswain.

closed 2019-06-12 by joan2937