joan2937/pigpio #269: GPIO for bitbanged slave?

selli69, 2019-03-13:
Hello!

I want to use a bitbanged I2C slave driven by a pigpio/python script for emulating devices used by codesys IEC PLC System as I2C master running on the same PI.

The python example on the website for "bsc_i2c" offers exactly the framework I want to start with. My problem is that I only have a RPI V1, which has no GPIO19.

The doc states: "This example assumes GPIO 2/3 are connected to GPIO 18/**19**. "
But i have to connect 2/3 to 18/**17** (or other GPIO)

Where can I change the GPIO used by the slave?

Thanks forward!

selli69, 2019-03-13:
Answer was given on StackExchange.

"The answer is no.

The SPI/BSC Slave peripheral is only available on fixed GPIO. In the case of I2C this is GPIO18 for SDA and GPIO19 for SCL.

See pages 102 and 160 of BCM2835 ARM Peripherals."

 I close this issue.

closed 2019-03-13 by selli69

guymcswain, 2019-03-14:
I'm not understanding what your requirement is but with bit-bang I2C you have the flexibility to choose the gpio pins.

selli69, 2019-03-20:
Hi Guy!

My requirement is to loopback the "original" RPi I2C master to a slave on the same device. A bitbanged slave seems not possible, only a "real" HW I2C slave on the designated pins. But It's only a problem with  old RPIs, so i ordered a newer one.

Vortm4x, 2024-03-03:
RPI 1 model b rev 1 has BCM2835 chip that supports BSC interface, BUT IT DOESN'T HAVE PINS FOR THAT, there is gpio 18, that has ALT functionality for BSC (and also I2C) slave, but there is no gpio 19, board doesn't have a BSC data at all, therefore bit bang slave is still an issue.

guymcswain, 2024-03-03:
This old thread is somewhat confusing - some of that confusion I may have contributed to.  There are no "bitbanged slave" APIs in this library.   You can, however, have a bitbanged master communicate with the BSC peripheral which provides hardware slave functionality for either I2C or SPI.   In either case, the fact that your version of Pi does not support BSC I/O makes 

> "therefore bit bang slave is still an issue."

 a mute point.