joan2937/pigpio #273: get_servo_pulsewidth throws GPIO is not in use for servo pulses

andrewg78, 2019-04-10:
Python 2.7 Raspberry Pi 3.
When we call get_servo_pulsewidth just after the raspberry PI is booted up, I got:
"
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1663, in get_servo_pulsewidth
    return _u2i(_pigpio_command(self.sl, _PI_CMD_GPW, user_gpio, 0))
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 975, in _u2i
    raise error(error_text(v))
	error: 'GPIO is not in use for servo pulses'
"
When I use set_servo_pulsewidth, even once, method get_servo_pulsewidth starts to work correctly

jveber, 2019-04-15:
And you think it is not a correct behavior? Why?

joan2937, 2019-06-12:
As @jveber indicates this is correct and expected behaviour.  A GPIO can be in several modes, not just servo mode.

closed 2019-06-12 by joan2937