joan2937/pigpio #274: Delay on reboot/shutdown

steuck13, 2019-04-23:
Greetings,

Im running pigpiod on the following setup?
- Raspbian 4.14.98-v7+
- Raspberry Pi 3 Model B Rev 1.2

And I created the following service to start the daemon (called by basic.target) and to keep it running for as long as possible on shutdown/reboot:

```
[Unit]
Description=Daemon required to control GPIO pins via pigpio
DefaultDependencies=false

[Service]
EnvironmentFile=/etc/default/config_file
ExecStart=/usr/local/bin/pigpiod -n localhost -n 127.0.0.1
ExecStartPost=/bin/bash -c 'pigs proc $SCRIPT'
ExecStartPost=/bin/systemctl start leds_service
ExecStop=/bin/bash -c "kill -TERM -- $(pidof pigpiod)"
Type=forking
SyslogIdentifier=pigpiod
Restart=always
RestartSec=2
```
I 'm having no problem with the startup. The daemon starts, load my script, and starts led_service. However, when rebooting/shutting down It is taking up to 90 seconds in most cases. There are rare cases when it boots/shutdown normally, but in general it keeps hanging.

Is this expected behavior?Is something wrong with my service?

Thank you

guymcswain, 2019-04-23:
I'm not sure if pigs is forkable.  Can you try the '-g' option on pigpiod and make everything non-forking.

steuck13, 2019-04-24:
Greetings,
Thank you for your answer.

Systemd documentation says:

> If set to forking, it is expected that the process configured with ExecStart= will call fork() as part of its start-up. The parent process is expected to exit when start-up is complete and all communication channels are set up. The child continues to run as the main service process, and the service manager will consider the unit started when the parent process exits. This is the behavior of traditional UNIX services. If this setting is used, it is recommended to also use the PIDFile= option, so that systemd can reliably identify the main process of the service. systemd will proceed with starting follow-up units as soon as the parent process exits.

The forked process is the one specified at `ExecStart`, right?Am I missing something?

I tried to remove everything leaving only the following with basic.target starting this service:

```
[Unit]
Description=Daemon required to control GPIO pins via pigpio
DefaultDependencies=false

[Service]
ExecStart=/usr/local/bin/pigpiod -n localhost -n 127.0.0.1
Type=forking
```

And the problem persists. I could add the `-g` flag but it is common for daemons to fork. Not only common, it is an expected behavior (pigpiod default is to fork). I need it to do this, so not forking is not an option...

obs: `DefaultDependencies` is set to false so the daemon keeps running until late shutdown

guymcswain, 2019-04-24:
I'm no systemd expert.  I run pigpiod service with `-g` and don't specify `Type=forking` which seems to work fine although I haven't measured the time it takes to shutdown.  But I haven't seen it hang when shutting down.

steuck13, 2019-04-25:
Greetings,
Thank your for posting your service.

I tried to run it like this:
```
[Unit]
Description=Pigpio daemon
DefaultDependencies=false
StartLimitIntervalSec=60
StartLimitBurst=5
StartLimitAction=reboot

[Service]
Type=simple
ExecStart=/usr/local/bin/pigpiod -g -n localhost -n 127.0.0.1
Restart=always
RestartSec=10
```

And the problem is still there. As far as I know, settings `DefaultDependencies` to false is needed when you want something to run on early boot or on late shutdown, but it doesn't seem to work really well with `pigpiod`.

joan2937, 2019-06-12:
I will not be making any changes in this area.  Mainly because I'm not familiar with system services.

A 90 second delay sounds like a network timeout.

SlySven, 2019-06-13:
The `PIDFile=` thingummybob might help with running `pigpiod` - it uses one which was at `/var/run/pigpio.pid` IIRC - but my installation is rather mature now and I do not know if that is currently how it is done.

guymcswain, 2020-01-03:
@steuck13 , Were you able to resolve this issue?  Did you attempt what @SlySven suggests?  Do something like:
`ExecStopPost=/bin/rm /var/run/pigpio.pid`
or
`ExecStartPre=/bin/rm /var/run/pigpio.pid`

The existence of the file **/var/run/pigpio.pid** will prevent the daemon from starting.

guymcswain, 2020-01-05:
How did you get this service to run at startup/reboot without an `[Install]` section?  I'm trying to reproduce your issue by writing a simple script of my own.

guymcswain, 2020-01-10:
@steuck13 , please retest with the patch for this issue in the 'develop' branch of this repo.

iovsiann, 2020-07-23:
FYI, shutdown issue still exists, here is a fix - just add KillSignal=SIGKILL. That kills pigpiod using SIGKILL instead of SIGTERM.

[Unit]
Description=Pigpio daemon

[Service]
Type=forking
ExecStart=/usr/bin/pigpiod
KillSignal=SIGKILL

[Install]
WantedBy=multi-user.target

SlySven, 2020-07-29:
Using SIGKILL is really not a good idea IMHO - SIGKILL terminates an application immediately *without* it being able to do any clean up - and one of the clean up processes of any daemon using a PID file ***is to remove that file so it is not seen by the* next *invocation of the daemon which causes it to either give up because it thinks there is still a daemon around OR (if it is more sophisticated) to examine that PID file - see what the number inside is and then check to see whether there is a process running with that ID and then abort if it is - which, if it is a PID file that hasn't been cleaned up, could be a totally unrelated process that just happens to have that PID number...***

If SIGTERM isn't doing it then by all means consider a *stronger* one but at least use one that still allows for some cleanup!

guymcswain, 2020-07-30:
Thanks @SlySven  for your comment.  I'd be happy to reopen and investigate this if someone could provide evidence that a problem exists.  My testing has always been remote/headless and I can't say I've noticed a difference rebooting/shutting down with or without pigpiod running.  There always seems to be black out/no comm period lasting over a minute.

iovsiann, 2020-07-30:
@guymcswain, here's what I do to reproduce the issue. Boot a Raspberry Pi 4B to a latest, fresh Raspbian image:

Baseline
````
$ sudo shutdown now
  // shutdown completes in around 3..5 seconds
  // i.e. Raspberry Pi 4B on-board green LED blinks 6 times indicating it's ok to power off the board
````

Reproduce issue
````
$ sudo pigpiod
$ sudo shutdown now
  // shutdown completes in, I guess, around 90 second
  // i.e. Raspberry Pi 4B on-board green LED blinks 6 times indicating it's ok to power off the board
````
Hope this helps.

Nodi-Rubrum, 2020-08-27:
@guymcswain 
Any progress on this?  I am seeing this across multiple installations, different models of Pi devices.

guymcswain, 2020-08-28:
@Nodi-Rubrum , no I have not (I've been on a pigpio hiatus for most of August).  I just happened to be looking at the raspberrypi.org forums today and found a reference to a similar issue.  In that post, a suggested workaround is to edit the file `/etc/systemd/system.conf` and change the `DefaultTimeoutStopSec` parameter.  This approach would affect all running services so a better approach might be to add `TimeoutStopSec=` under the `[Service]` section of the `pigpiod.service` file as follows:
```
[Service]
Type=forking
ExecStart=/usr/bin/pigpiod
TimeoutStopSec=15
```
Change `TimeoutStopSec=` to your preference - the default is 90 seconds.  Note that this is essentially a `SIGKILL` but only after some amount of delay - so it could be subject to issues pointed out by @SlySven.

Could you please test this to see if it works for you?

Nodi-Rubrum, 2020-08-28:
Sure, will check out the above.  Do you not want to test one change at a time?  Enabling forking and setting the timeout stop is two changes at once?  Or is that no an issue (for testing) in this case?

guymcswain, 2020-08-28:
> Enabling forking and setting the timeout stop is two changes at once?

Good catch.  I was just copying the section used by @iovsiann thinking it was 'standard' except for the added SIGKILL line.  Yes please just test the single line change with `TimeoutStopSec=15`.

Nodi-Rubrum, 2020-08-28:
30+ year IT guy... at lot of engineering design and code review over the years... so QA/QC mind-set is my default.  :)

Nodi-Rubrum, 2020-08-28:
Unfortunately, TimeoutStopSec is being ignored.  There are a number of google matches to why and how this can happen.  Odd.  But what is more odd, if you will, is when I restored the original unit file, with no changes, and tested again, I did not get the pigpiopd hang issue.  Really odd.

guymcswain, 2020-08-28:
Odd indeed.  Can you tell the versions of pigpio and OS you are running?  I'll try to get some time this weekend to attempt to reproduce.

Also, are you running shutdown/reboot from a locally attached console or remote?

guymcswain, 2020-08-28:
This is the standard service file:
```
[Unit]
Description=Pigpio daemon

[Service]
Type=forking
ExecStart=/usr/bin/pigpiod

[Install]
WantedBy=multi-user.target
```

Nodi-Rubrum, 2020-08-28:
Now that is interesting... because all the documentation I have found, suggests quite a different configuration.  Don't get me wrong, I am not debating the above, just wondering why the difference?  Many examples show no forking, show explicit kernel networking overrides, etc.

guymcswain, 2020-08-28:
... and what does `pigpiod -v` report?  Most recent version is 77.

guymcswain, 2020-08-28:
> Many examples show no forking, show explicit kernel networking overrides, etc.

I have used similar examples myself but the important difference is to invoke pigpiod with the -g option (don't fork) in these cases.

guymcswain, 2020-08-28:
Another thing to test:   Add `PIDFile=/var/run/pigpio.pid` to the service section.

It occurs to me that since the service is named `pigpiod.service`, systemd may incorrectly guess the path to the pid file is '/run/pigpiod.pid'.  In that case, systemd *may* wait for the default timeout of 90 seconds before sending SIGKILL.  This addition removes any ambiguity.

alrabe, 2020-08-28:
> Another thing to test: Add `PIDFile=/var/run/pigpio.pid` to the service section.

Hi 

I have a similar problem on my PI4. The pigpio service cannot be stopped but only every now and then. I added your `PIDFile` line to the service section. After a few working attempts the process could not be stopped again. This time by chance while I tried to stop it manually using `systemctl stop`.

```
Aug 29 01:21:14 retro systemd[1]: /etc/systemd/system/pigpiod.service:6: PIDFile= references path below legacy directory /var/run/, updating /var/run/pigpio.pid → /run/pigpio.pid; please update the unit file accordingly.
Aug 29 01:21:36 retro systemd[1]: Stopping PWM cooler fan controll...
Aug 29 01:21:36 retro systemd[1]: fan_control.service: Main process exited, code=killed, status=15/TERM
Aug 29 01:21:36 retro systemd[1]: fan_control.service: Succeeded.
Aug 29 01:21:36 retro systemd[1]: Stopped PWM cooler fan controll.
Aug 29 01:21:36 retro systemd[1]: Stopping Pi GPIO daemon...
Aug 29 01:23:06 retro systemd[1]: pigpiod.service: State 'stop-sigterm' timed out. Killing.
Aug 29 01:23:06 retro systemd[1]: pigpiod.service: Killing process 371 (pigpiod) with signal SIGKILL.
Aug 29 01:23:06 retro systemd[1]: pigpiod.service: Main process exited, code=killed, status=9/KILL
Aug 29 01:23:06 retro systemd[1]: pigpiod.service: Failed with result 'timeout'.
Aug 29 01:23:06 retro systemd[1]: Stopped Pi GPIO daemon.
```

Btw. the `fan_control.service` relies on gpio.

guymcswain, 2020-08-29:
@alrabe , can you change the line to:
`PIDFile=/run/pigpio.pid`
Thanks.

Nodi-Rubrum, 2020-08-29:
Just for reference post the working unit file?  Just would like to compare to what I am testing?  I am not seeing the issue on my Pi4, but am still seeing it on one of my Pi3 devices.  Doing more testing.

guymcswain, 2020-08-29:
```
[Unit]
Description=Pigpio daemon

[Service]
Type=forking
PIDFile=pigpio.pid
ExecStart=/usr/bin/pigpiod

[Install]
WantedBy=multi-user.target
```

guymcswain, 2020-08-31:
For reference, a fresh install of Raspberry Pi OS - `Version: August 2020` - has this content in the pigpiod.service file:
```
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```
The same expected behavior is observed - the daemon is shutdown within a few seconds and does not experience the 90 second delay described in this issue.

Nodi-Rubrum, 2020-09-01:
wow this is a quite a bit different from past variants.  Will test this as well, but suspect it will work as expected.  The only change I can think of is limiting he network interface, say -n 127.0.0.1 -n localhost.

Nodi-Rubrum, 2020-09-01:
When I try to use the above, the service seems to hang on start.  I never get a prompt back.

guymcswain, 2020-09-01:
The -l option is just that, optional.  The distro supposedly requires this as a default for "security".

Nodi-Rubrum, 2020-09-04:
I also hit this error...
`root@pi3modelb1:~# apt install pigpio
Reading package lists... Done
Building dependency tree
Reading state information... Done
pigpio is already the newest version (1.64-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@pi3modelb1:~# /usr/local/bin/pigpiod -l
root@pi3modelb1:~# 2020-09-04 18:36:44 initMboxBlock: init mbox zaps failed
Can't initialise pigpio library
`

Nodi-Rubrum, 2020-09-04:
Well no pid file...
# find / | grep pigpio.pid

returned nothing.

I removed the -l option and added -n 127.0.01 -n localhost.  Tried the reboot, that cleared the previous error.  So far so good.  This error only happened on one Pi device, the couple of others, no issues as of yet.

But my additional question still is pending... what unit file is the correct unit file?

/etc/systemd/system/pigpiod.service
/lib/systemd/system/pigpiod.service

I removed the user space one, i.e. /etc/systemd/system.  Running under the ### /lib/systemd/system/pigpiod.service file.

guymcswain, 2020-09-05:
> But my additional question still is pending... what unit file is the correct unit file?
/etc/systemd/system/pigpiod.service
/lib/systemd/system/pigpiod.service

The service file under `/etc/systemd/system` will override `/lib/systemd/system`.  It seems recent OS versions are installing the pigpio package service file under `/lib/systemd/system`.  I usually download the latest pigpio version as a zip file and install the service under `/etc/systemd/service` since the OS package version of pigpio is usually several revisions behind.

Nodi-Rubrum, 2020-09-06:
Cool.  By the way, have not seen any slow down or shut down hangs since the latest set of changes to the unit file.

knutegit, 2020-09-13:
Posted at guym's request.

[debug.log](https://github.com/joan2937/pigpio/files/5214284/debug.log)

guymcswain, 2020-09-17:
@knutegit ,
From this log I see pigpio exit successfully  after receiving the TERM signal.  Something else is causing your system to hang.

Nodi-Rubrum, 2020-09-19:
I have had 3 Pi devices running the latest configuration we worked up in this discussion, I have not see a delay in shutdown for some time now.

closed 2020-09-29 by guymcswain

pompushko, 2024-01-09:
Well, greetings :)

```
pigpiod -v
79
```

```
cat /lib/systemd/system/pigpiod.service
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```

Fresh install in Pi Zero... still get this error :)

```
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
```

guymcswain, 2024-01-09:
Can you try adding this override 

```bash
[Unit]
Requires=default.target
After=default.target
[Install]
WantedBy=
WantedBy=default.target
```

commit bab05ceb (2020-02-03): Merge pull request #321 from joan2937/Release-v74

Release v74: closes #274, closes #298, closes #266, closes #316, closes #223.