joan2937/pigpio #275: pigpiod on different port

karl0ss, 2019-04-29:
I am running pigpiod on a different port to the standard 8888, using 'sudo pigpiod -p 8887' I am wanting to update my lights using pigs, but cant seem to work out how to get pigs to use the new port?

Thanks

karl0ss, 2019-05-01:
`pi@octopi:~ $ sudo pigpiod -p 8887
`
```
pi@octopi:~ $ sudo netstat -tulpn | grep :8887
tcp6       0      0 :::8887                 :::*                    LISTEN      1816/pigpiod
```
when I run 
`pigs hwver
`
or
`sudo pigs hwver
`

Terminal just hangs...

When I do -

`pi@octopi:~ $ sudo pigpiod
`
```
pi@octopi:~ $ sudo netstat -tulpn | grep :8888
tcp6       0      0 :::8888                 :::*                    LISTEN      1946/pigpiod
```

```
pi@octopi:~ $ pigs hwver
10494163
```

```
pi@octopi:~ $ cat /dev/pigerr&
[1] 1998
pi@octopi:~ $ echo "hwver" > /dev/pigpio
```

It seems that pigs just can't communicate with pigpiod when its not running on the standard port?

Thanks

joan2937, 2019-06-12:
Have you told pigs the changed port number?

`PIGPIO_PORT=8887
pigs hwver`

closed 2019-07-07 by joan2937

sawansaurabh, 2020-10-01:
Hi Joan, is there a way i can change the default port for pigspiod? i want to run it on startup always and on a different port.

Noschvie, 2020-10-01:
pigpiod accepts the following configuration options
http://abyz.me.uk/rpi/pigpio/pigpiod.html

sawansaurabh, 2020-10-01:
Hi, 
Thanks for quick reply and sorry for continuing on a closed thread.
I have gone through that documentation. I want to know how do i permanently set a different port for the daemon (change it from the default 8888 to say.. 8000). Running pigpiod -p 8000 will only keep that port until i reboot right? Sorry if it is too obvious, i might be missing something.

guymcswain, 2020-10-01:
Enable the systemd service if one already exists.
`sudo systemctl enable pigpiod`

Edit the service file to include the port option.
Then do:
'sudo systemctl daemon-reload'
Followed by:
`sudo systemctl start pigpiod'
Or just reboot.

If the pigpiod.service file doesn't exist in your /etc/systemd/service directory, copy the one in the repo under /util.  Modify accordingly.

SchulzJ-UNI, 2022-01-11:
Hi,
I'm sorry to continue to a closed threat. I have the exact same problem as @sawansaurabh.
Unfortunately, I do not have a directory that is called  "/etc/systemd/service". I can find a pigpiod.service file at "/lib/systemd/system" or a public.conf file at "/etc/systemd/system/pigpiod.service.d" but both are not editable. 
It would be really great if either @sawansaurabh could explain your solution if you fixed the problem or if @guymcswain could specify how I am able to edit the service file to include the port option.
Thanks a lot in advance!

guymcswain, 2022-01-11:
> I do not have a directory that is called "/etc/systemd/service"

The above is a typo.  It should be `/etc/systemd/system`.  There should be a file there named `pigpiod.service`.

Example showing a system that does not contain the above referenced file:
```bash
pi@blueberry:~ $ systemctl status pigpiod
Unit pigpiod.service could not be found.
```
In this case, you will need to copy the file from the /util directory of the repo.  Edit this line `ExecStart=/usr/bin/pigpiod` adding the `-p xxxx` option for the desired port number.

Resulting file in `/etc/systemd/system/pigpiod.service` should look like this:
```bash
ExecStart=/usr/bin/pigpiod[Unit]
Description=Pigpio daemon

[Service]
Type=forking
PIDFile=pigpio.pid
ExecStart=/usr/bin/pigpiod -p xxxx

[Install]
WantedBy=multi-user.target
```

SchulzJ-UNI, 2022-01-11:
Thanks for the quick response!
Unfortunately it looks a bit different at my Raspberry Pi.
My pigpiod.service file seems to be stored at /lib/systemd/system:
`pi@raspberrypi:~ $ systemctl status pigpiod
pigpiod.service - Daemon required to control GPIO pins via pigpio
   Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled; vendor preset:
  Drop-In: /etc/systemd/system/pigpiod.service.d
           \u2514\u2500public.conf
   Active: inactive (dead)
lines 1-5/5 (END)
`

SchulzJ-UNI, 2022-01-11:
Thanks for the fast response!
Unfortunately it looks differnent at my Raspberry Pi. 
My pigpiod.service file seems to be stored at **/lib**/systemd/system

`pi@raspberrypi:~ $ systemctl status pigpiod`
`pigpiod.service - Daemon required to control GPIO pins via pigpio'`
 `Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled; vendor preset:`
  `Drop-In: /etc/systemd/system/pigpiod.service.d`
           `public.conf`
   `Active: inactive (dead)`
`lines 1-5/5 (END)`

The **/lib**/systemd/system/system.config looks as the following:
`[Unit]`
`Description=Daemon required to control GPIO pins via pigpio`
`[Service]`
`ExecStart=/usr/bin/pigpiod -l `
`ExecStop=/bin/systemctl kill pigpiod`
`Type=forking`
`[Install]`
`WantedBy=multi-user.target`

and if I'm trying to change it i get the following error:
[image]
English: Document could not be saved. No rights.

Same error if I'm trying to create a file at **/etc/**systemd/system or change the public.config file at **/etc/**systemd/system/pigpiod.service.d

guymcswain, 2022-01-11:
Yes, systemd now appears to be located in other directories.  Nevertheless, wherever it is located, the same edit to the line beginning with ExecStart applies - just append -p (your port number).  And yes, you need to use 'sudo' because these are root owned directories.

SchulzJ-UNI, 2022-01-12:
Thanks a lot! The hint with 'sudo' was very helpfull.
I am now able to start the pigpiod on port 8889 at startup. The `systemctl status pigpiod.service` command also indicates that the daemon is running(see screenshot). 
BUT: when i try to test pigpiod by command `pigs t` or `pigs hwver` i get the error "socket connection failed". Which I assume needs to work to control the GPIO Pins remotely. 
I tried PIGPIO_PORT=8889, but i doesn't help.
[image]

This error does not appear if i start pigpiod via `sudo pigpiod` --> on socket 8888. 
[image]
I am really sorry for all the question and appreciate your help so far!

SchulzJ-UNI, 2022-01-13:
And how am I able to get the `export PIGPIO_PORT=8889` command done automatically at reboot?

guymcswain, 2022-01-13:
You can set environment variables within the pigpiod.service unit.  There is plenty of information on systemd available on the internet.