joan2937/pigpio #278: Raspberry pi 4

kauailabs, 2019-06-24:
I’m interested in using pigpio on Raspberry pi 4.  If you need beta testing I’m happy to help.  Existing Features used would include Aux SPI, SPI, I2C, PWM generation, Serial Port, interrupt handling from GPIO pins, and basic GPIO pin reads and writes.

Thanks for your consideration.

- scott

joan2937, 2019-06-25:
pigpio does not work on the Pi4B.

At the moment I can't get any information so do not know if this is a temporary blip or not.

I need to find out what (if any) changes have been made to the memory map and SoC hardware peripherals (e.g. DMA, PWM, PCM, GPIO, SYST, BSC).

kauailabs, 2019-06-25:
You may have already come across this, but just in case here's the info from the RPI Foundation's website:  https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf

kauailabs, 2019-06-26:
Another finding.  A commit to the [rpi kernel github pin driver code](https://github.com/raspberrypi/linux/commit/abcfd092860760087b87acbdda0963fe7906839c#diff-2a94646cafdb60e71d57b6416529081f), indicating:
(1) The new processor (everything on RPI Foundation site says BCM2711) appears to be referred to as the BCM2838.
(2)   The pin driver code was modified to change the pullup/pulldown control registers on the BCM2838.

joan2937, 2019-06-26:
Unfortunately that gives little to no useful information as far as pigpio is concerned.

It doesn't even give the new peripheral address which appears to be 0xfe000000.

joan2937, 2019-06-28:
I have made progress.

Most of the peripherals used by pigpio seem to work as before (PWM/PCM/GPIO/SYST/DMA).  I haven't tried SPI yet.

The DMA channels which were free for pigpio use (14 primary, 6 secondary) do not operate quite as before.

If you want to experiment using the peripheral address above (change in gpioHardwareRevision) and starting the daemon with -d 6 - e 14 (swapping primary/secondary) should be enough to get pigpio running at least partially.

There are PUD changes which need to be incorporated.

If I can sort out the DMA channels I'll have a look at making all the changes needed to do a new release.

My time to dedicate to pigpio is extremely limited at the moment so I can't provide an estimate of time-scale.

If it turns out to be infeasible I'll let you know.

kauailabs, 2019-06-28:
Thank you very much for all of your valuable time to look into this, Joan.

My RPI4 should arrive early next week, and I can test this out then.  My main focus will be on the Aux. SPI, and after that on the PWM Generation (using DMA, as I understand it) on the GPIO pins.  Is it possible for you to create a branch in github with your work in progress?

I will also keep searching for any new information that might be helpful and post it on this issue thread.

joan2937, 2019-07-02:
V69 released.  There will probably be problems with the DMA channels being used.  When the full peripheral specs are available they will probably indicate further software changes.

kauailabs, 2019-07-04:
Hi Joan,

Very excellent; I am very grateful for this.  I have tested this and found it to be working so far.  I'm not using the deamon, I'm using the C interface.

So far, Aux SPI is working, GPIO Outputs are working and PWM Generation on the GPIO Outputs is working.  And I've verified this on both RPI4B as well as RPI3B+, so I haven't noticed any regressions.

Up next is to  do more testing in the areas of GPIO Inputs, verifying the pullup/pulldown on the GPIO pins, and the SPI/UART/I2C interface.  As a side note, I've heard that the I2C clock stretching on the Raspberry Pi 4B (BCM2711) is working due to fixes in the silicon, so hopefully that's correct and I'll see if I can verify that, too.

jhw84, 2019-07-05:
Hi,
I've tried to use V69 on a PI4B (2GB) for using the pi as a i2c-slave (bscXfer/bsc_xfer). This does not seem to work: The data line is kept high by the pi it seems, making any communication on the i2c bus impossible. I've tried both with and without daemon. Is this at all related to choosing the right DMA channels or is this another issue all together?

joan2937, 2019-07-05:
@jhw84 I will have a look.

Could you add the result of **grep Revision /proc/cpuinfo** and **pigs hwver**

Is the daemon running okay?  If DMA fails to start the system will seem sluggish and pigpio will likely be taking 100% CPU.

jhw84, 2019-07-05:
Hi @joan2937 , thanks for the quick response. Here is the data you requested:
```
$ grep Revision /proc/cpuinfo 
Revision	: b03111
```
```
$ pigs hwver
11546897
```
I just checked: When launching pigpiod without any arguments it consumes about 7% CPU.

joan2937, 2019-07-05:
@jhw84 I tried and got the same failure for slave I2C as you.

I'm afraid slave I2C will not be possible on the BCM2711/Pi4b.  The SPI/BSC slave functionality has been removed.  GPIO 18/19 in mode ALT3 now (completely pointlessly) provide SPI device 6.  I say completely pointlessly as they already provide a different SPI interface in mode ALT4.

jhw84, 2019-07-05:
Too bad, fortunately I control both devices on the bus so I can move to a different type of interface. Hard to imagine why this choice was made, as you say there is no point to it, it only loses functionality without gain.
Many thanks for looking into this so quickly!

braincore, 2019-07-07:
I'm interested in PWM generation. @kauailabs:

1. Did you try a wide range of sample rates and frequencies?
2. Did you check the quality by examining the waveforms with an oscilloscope?

@joan2937: What were the changes to the DMA channels? Is there a detailed datasheet available yet? Any reason to worry that there might be a degradation of performance/quality w.r.t PWM?

joan2937, 2019-07-08:
@braincore 

pigpio currently defaults to primary channel 7, secondary channel 6 on the Pi4B.  Without a detailed datasheet it's pointless trying to guess why this is needed.  The detailed datasheet should be made available to us within a couple of weeks from now.

I know of no reason why PWM would be degraded from its previous performance.

All I automatically do when testing is run x_pigpio, x_pigpio.py (Python 2 and 3), x_pigpiod_if, x_pigpiod_if2, x_pigs, and x_pipe.  They showed no regression with PWM.

bludin, 2019-08-05:
Just to let you know: I've tested a number if I2C (on both the i2c-0 and i2c-1 bus) devices and a SPI devices and they work fine with pigpio on the RPi4B in my hands. I can also confirm that a device that uses clock-stretching and didn't work with the RPi3B+ is working with the RPi4B.
Hardware PWM is also working fine

CreaValix, 2019-08-07:
#284 Got pigpiod running on a RPi4 model B running tumbleweed (arm64).

Serveurperso, 2019-08-12:
Thanks a lot, DMA PWM successfully tested 24/7 on raspberry PI 4 robot

hayguen, 2019-08-15:
Are Raspi 4 internal pull up/down already working?

joan2937, 2019-08-15:
@hayguen Yes, any version **after** V68 supports the Pi4B (and pulls).

matrn, 2019-08-22:
Hello everyone. I have RPI4B 2GB and in one code I am using `hardware_PWM()` on pin `18`. In second code I have IR send example (from https://www.raspberrypi.org/forums/viewtopic.php?t=126919) with IR diode on pin `13` and it looks like when `pi.wave_chain(wave)
` comes to the scene, PWM on pin `18` stopps working - LED is off.

Does anyone knows where is the problem?

Thanks for advance.

joan2937, 2019-08-23:
@matrn  http://abyz.me.uk/rpi/pigpio/python.html#wave_chain

matrn, 2019-08-23:
@joan2937 Thanks. Is there any way to do not cancel HW PWM with `wave_chain()`?

joan2937, 2019-08-23:
@matrn No there is not, they are incompatible.

matrn, 2019-08-23:
@joan2937 That's bad. :/

JeffJassky, 2019-08-31:
Forgive my ignorance, but when I do apt-get upgrade pigpio it says `pigpio is already the newest version (1.68-3+rpi1)`

How can I get the latest version installed?

joan2937, 2019-08-31:
@JeffJassky Glad you are sorted.

For info.

If you are using the Raspbian version then `sudo apt update` followed by `sudo apt upgrade` will get the latest Raspbian version.

A later version may be available at http://abyz.me.uk/rpi/pigpio/download.html or at https://github.com/joan2937/pigpio

I try to keep the abyz.me.uk and github versions the same but occasionally there is a more recent version at github.

Please be aware of the following.

pigpio places files in the following locations
```
/usr/local/include (pigpio.h, pigpiod_if.h, pigpiod_if2.h)
/usr/local/lib (libpigpio.so, libpigpiod_if.so, libpigpiod_if2.so)
/usr/local/bin (pig2vcd, pigpiod, pigs)
/usr/local/man (man pages)
```
The (Raspbian) raspberrypi.org image containing pigpio uses different locations.
```
/usr/include (pigpio.h, pigpiod_if.h, pigpiod_if2.h)
/usr/lib (libpigpio.so, libpigpiod_if.so, libpigpiod_if2.so)
/usr/bin (pig2vcd, pigpiod, pigs)
/usr/man (man pages)
```
Mostly this doesn't matter as the /usr/local directories will generally be earlier in the search path.  The pigpio built includes, binaries, and manuals are normally found first.

However the wrong libraries may be linked during the compilation.  If this is the case remove the /usr/lib entries for libpigpio.so , libpigpiod_if.so, and libpigpiod_if2.so

SlySven, 2019-08-31:
:information_source: For the record the use of the different locations is entirely appropriate - distribution provided packages use `/usr`, `/etc` - system operators (i.e. `root` and those using `su`/`sudo`) are expected to use `local` alternatives and their choices are expected to override the distribution's defaults. The choice between them is supposed to be handed - in a C/C++ application - by the correct `-I` (for include "header" files) and `-L` (for libraries to be linked) path options to the compiler/linker...

joan2937, 2019-08-31:
@SlySven Yes, I did not mean to imply there was anything wrong with the choice of file locations.  I just meant it as a heads-up as something which might catch people out (as I myself was).

fivdi, 2019-09-07:
The documentation for [gpioGetPWMrealRange](http://abyz.me.uk/rpi/pigpio/cif.html#gpioGetPWMrealRange) states the following:
```
If hardware PWM is active on the GPIO the reported real range will be approximately 250M divided by the set PWM frequency. 
```
To be correct for the Raspberry Pi 4 it looks like it should be:
```
If hardware PWM is active on the GPIO the reported real range will be approximately 250M (375M for the BCM2711) divided by the set PWM frequency. 
```

I'd also like to mention that the [pigpio Node.js module](https://github.com/fivdi/pigpio) works well with V69 or higher of the pigpio C library on the Raspberry Pi 4.

@joan2937 Excellent job :)

Reda1000, 2019-10-25:
Hi there,

just wanted to inform that I had weird difficulties with gpio pin 26 and 20 as they were not working. While Pins 05, 06, 12, 13, 16, 19 are working as expected - am also using Node pigpio wrapper.

Could there be an issue with GPIO Pins 2x when addressing them?

I double checked with PIGPIO-Test tool:
[image]

joan2937, 2019-10-25:
digitalWrite sounds like a wiringPi command.  Have you downloaded and installed the latest version of wiringPi.

As shown in your terminal output pigpio sees nothing wrong with the GPIO.

Reda1000, 2019-10-25:
@joan2937 thanks for your quick reply. Indeed digitalWrite is not from pigpio but its from node-pigpio:
https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#digitalwritelevel

It is defined as:
`  digitalWrite(level) {
    pigpio.gpioWrite(this.gpio, +level);
    return this;
  }`

I tried finding gpioWrite in .c File but only found in Headers.

There is nothing wrong by gpiog-test but still I did not get ~5V on 26 and 20. Maybe there are other people having similar problems - I am currently using other Pins which is fine so only mentioning.

joan2937, 2019-10-25:
ALL the Pi GPIO are 3V3.  You will only get 5V on the 5V power rail (pins 2 and 4).

What do you mean by 26 and 20?

https://pinout.xyz/

pigpio only uses GPIO numbers (not physical pin numbers).

guymcswain, 2019-10-25:
@Reda1000 , assuming you are using broadcom gpio pin numbers, can you report what the levels are when you `pigs w 20 1` and `pigs w 26 1`

Reda1000, 2019-10-25:
@guymcswain @joan2937 Thank you both for your help.

I was afraid short-circuiting pins and used a cable in between like so:
[image]
One cable (yellow) has been damaged, so it gave me weird results on meter. Should have checked cable earlier... sorry for taking your time.

patrickliu-99, 2019-11-22:
Not sure if this is related, however, I was trying to use pigpio (v71) with the Pi Camera on a 4 using Python 3.

Worked at some point, and then some time later (with an apt full-upgrade because the File Explorer started crashing whenever I tried to open it, then an apt-get update and apt-get upgrade), I stopped being able to access the Pi Camera, and spent some time debugging it - one of the errors that was reported occurred in pigpio.py, so I eventually decided to try swapping the DMA channels, which seemed to do the trick?

I'm investigating further - maybe starting pigpiod on startup using rc.local is configured incorrectly and is causing the issue.

guymcswain, 2020-03-02:
The SPI/BSC slave functionality is now working(*) on Pi4 model B although the GPIO have moved from 18,19 to 10,11 respectively.  Version 7401 is available on the 'develop' branch of the pigpio GitHub repo.  See #328 for more details.

\*  Note:  SPI slave functionality does not work on any Pi model.

maheshhchandra, 2020-04-10:
Hi any way to release PCM and PWM for audio 
or restating audio after sending
gpioWaveTxSend();
audio not working even after
gpioTerminate();

guymcswain, 2020-04-10:
@maheshhchandra , your question/issue is not related to Pi4 - the subject of this thread.  Please open an separate issue if you believe there is a behavior that is inconsistent with the pigpio documentation.

hayguen, 2020-04-10:
> Hi any way to release PCM and PWM for audio
> or restating audio after sending
> gpioWaveTxSend();
> audio not working even after
> gpioTerminate();

usually, only one of PCM or PWM is used.
when using PWM (not the default), there should not be any problems with sound.
see https://github.com/joan2937/pigpio/issues/331
and https://github.com/joan2937/pigpio/pull/332

guymcswain, 2020-05-02:
For all participants on this thread:  I plan to close this issue as it was a catch-all for anything related to Pi4.  It is quite stale now and there have been multiple releases since.  Any outstanding issues should be opened in a separate thread.  Thanks.

@joan2937 , looks like we missed this input from @fivdi .  Are you ok with this change in doc?
> The documentation for gpioGetPWMrealRange states the following:

> If hardware PWM is active on the GPIO the reported real range will be approximately 250M divided by the set PWM frequency.
> To be correct for the Raspberry Pi 4 it looks like it should be:
>
> If [...]

jnakanojp, 2020-10-30:
Hi,
I finally succeeded to use Pi 4 SPI3,4,5 (6 is not checked but maybe ok).
If there is anyone try to use Pi 4 SPI 3,4,5 with PiGPIO,
please check my fork.
https://github.com/jnakanojp/pigpio/commits/spi3
https://github.com/jnakanojp/pigpio/commits/spi4
https://github.com/jnakanojp/pigpio/commits/spi5

Thanks.

lbt, 2020-11-15:
> For all participants on this thread: I plan to close this issue as it was a catch-all for anything related to Pi4. It is quite stale now and there have been multiple releases since. Any outstanding issues should be opened in a separate thread. Thanks.

Please note the text here:
  http://abyz.me.uk/rpi/pigpio/index.html
still says "At the moment pigpio on the Pi4B is experimental..." which lead me to dig deeper and eventually come here.

This looks like an incredibly supportive community so I'd hate to see people pass by because of an outdated comment :)

joan2937, 2020-11-15:
@lbt 

Okay, I have updated the web site.

guymcswain, 2020-12-18:
@joan2937 - I just noticed a new version of the BCM2711 datasheet on raspberrypi.org:

- Release 3
- Date:  Oct 16, 2020
- Description:  First public release

If you see anything affecting pigpio just open a new issue.

closed 2020-12-18 by guymcswain

tadegiac, 2022-12-19:
Hi there !!!
Sorry if this is a redundant question. I was wondering if it is still impossible to use the i2c ?

> @jhw84 I tried and got the same failure for slave I2C as you.
> 
> I'm afraid slave I2C will not be possible on the BCM2711/Pi4b. The SPI/BSC slave functionality has been removed. GPIO 18/19 in mode ALT3 now (completely pointlessly) provide SPI device 6. I say completely pointlessly as they already provide a different SPI interface in mode ALT4.
> 
> Compare page 102 of https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf to https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf

Thank you very much for your time and effort. 
<3

CreaValix, 2022-12-19:
I have a Raspberry Pi 4 Model B Rev 1.4 running as I2C slave via `pigpio` and at the same time as a I2C master via `/dev/i2c-1`.

tadegiac, 2022-12-20:
> I have a Raspberry Pi 4 Model B Rev 1.4 running as I2C slave via `pigpio` and at the same time as a I2C master via `/dev/i2c-1`.

Thank you very much for your answer, did you do any special steps to set it up? I am not a very experienced Linux user, i just now my way around. Is there a ressource where i can follow the installation steps or sth similar?

CreaValix, 2022-12-21:
No, nothing special. I followed the example code for the `bsc_i2c` Python call. I am not at home, so I can't look up the wiring...

For the master, I added `dtparam=i2c_arm=on` to get the `/dev/i2c-1` device. Then I used the python-periphery module to xfer data to my I2C slave:
```
i2c = I2C("/dev/i2c-1")
i2c.transfer(0x40, [I2C.Message([byte1, byte2, ...])])
```