joan2937/pigpio #279: Should this work on a Banana Pi M3 ?

GeorgeFlorian, 2019-06-25:
Hello !

I am trying to test if the M3's pins works or not and I've stumbled upon this library.

I followed the instruction here: https://abyz.me.uk/rpi/pigpio/download.html

And I've ran the following commands:
```
: sudo apt-get autoremove && sudo apt-get -f install && sudo apt-get update && : : sudo apt-get upgrade -y
: wget abyz.me.uk/rpi/pigpio/pigpio.zip
: unzip pigpio.zip
: cd PIGPIO
: make
: sudo make install
pi@bananapim3:~/PIGPIO$ sudo ./x_pigpio # check C I/F
2019-06-25 09:15:49 initAllocDMAMem: mbox open failed(No such device or address)
pigpio initialisation failed.
pi@bananapim3:~/PIGPIO$ sudo pigpiod
pi@bananapim3:~/PIGPIO$ 2019-06-25 09:16:49 initAllocDMAMem: mbox open failed(No such device or address)
Can't initialise pigpio library
```

And that's it. I can't make it work and I don't know what to past this point ! Any ideas ?
Thank you !

joan2937, 2019-06-25:
pigpio only works on the Raspberry Pi (it uses features unique to the Broadcom SoC).

closed 2019-06-25 by joan2937

GeorgeFlorian, 2019-06-26:
> pigpio only works on the Raspberry Pi (it uses features unique to the Broadcom SoC).

Ok. Thank you !

I've been trying to replace Raspberry Pi Zero with a Banana Pi M3 for the past 2 weeks and I can't seem to make my project work on the Banana. This is the reason I've tried your library.

I see that you seem to know the differences between a Raspberry Pi and a Banana Pi, like the ones mentioned above:
>(it uses features unique to the Broadcom SoC).

Can you, please, tell me some more "unique" things that set apart the two boards ?
It would be of a great help for me !
Thank you !

joan2937, 2019-06-26:
pigpio uses the Brodcom SoC DMA engine to provide the following functionality at little CPU cost.

1. sampling and time-stamping GPIO 0-31 every 5µs (by default, maximum sample rate is 1MHz)

1. generating hardware timed PWM/Servo pulses on GPIO 0-31

1. generating accurately timed chains of GPIO level changes

This allows pigpio to provide solutions to many software problems without the need of additional external hardware.

I suppose other SoCs may have equivalent functionality in their DMA engines so perhaps it just needs someone familiar with those SoCs to write similar software.