joan2937/pigpio #280: The Pi4B has no support for SPI/BSC SLAVE peripheral

joan2937, 2019-07-07:
See post by @jhw84 in https://github.com/joan2937/pigpio/issues/278#issuecomment-508764135_

joan2937, 2019-07-07:
The GPIO mode needed to access the peripheral have been reassigned to provide an additional master SPI interface.

The lack of support by the BCM2711 SoC used in the Pi4B will be noted in the pigpio documentation.

joan2937, 2020-02-22:
New information has come to light and the SPI/BSC SLAVE peripheral seems to work as it does on earlier Pi models with one change.

https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=265832&p=1616018#p1616018

The access to the peripheral is multiplexed to GPIO 8-11 on the Pi4B / BCM2711 SoC.

```
#define BSC_SDA_MOSI_2711 10
#define BSC_SCL_SCLK_2711 11
#define BSC_MISO_2711      9
#define BSC_CE_N_2711      8
```
My preliminary tests show that I2C seems to work as before.  I haven't tested SPI (which didn't use to work anyhow).

guymcswain, 2020-02-22:
This is great news!  Let me know if you want me to assist with the GitHub branching stuff, or if you just patch the master branch directly I'll sync it to develop.

joan2937, 2020-02-22:
All I have done is locally `git checkout -b develop`.  So far I have only changed pigpio.c and pigpio.h.  I will need to change pigs, the Python module, and pigpiod_if2 to reflect documentation updates.  While I am at it I will probably add a little more explanatory documentation in the affected files (of a non related nature).

guymcswain, 2020-02-22:
That works fine.

The pendantic approach is to create a separate 'hotfix-blah' branch push it to GitHub then later merge it to both master and develop.  But unless we had more cooks in the kitchen this seems too burdensome.

joan2937, 2020-03-01:
@guymcswain 

Okay.  I created a local branch called BCM2711-BSC-slave and made the changes I wanted. I git-added the changes and then did a `git push -u origin BCM2711-BSC-slave`.  I thought that would put my local changes to the remote.  It doesn't seem to.  It wants me to commit the changes.

```
git status -uno
On branch BCM2711-BSC-slave
Your branch is up-to-date with 'origin/BCM2711-BSC-slave'.

Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

	modified:   pigpio.3
	modified:   pigpio.c
	modified:   pigpio.h
	modified:   pigpio.py
	modified:   pigpiod_if2.3
	modified:   pigpiod_if2.c
	modified:   pigpiod_if2.h
	modified:   pigs.1
	modified:   setup.py

Untracked files not listed (use -u option to show untracked files)
```
Any idea what I should do to get the changes to the remote so I can generate a pull request?

guymcswain, 2020-03-01:
Go ahead and commit changes to your local branch:
 git add .
 git commit -m 'these are my changes...'

Now
 git push
should fail but give you instructions on how to do it.

joan2937, 2020-03-01:
I did a local commit and then pushed and it seems to have done what I expected.  The branch is populated with the changes.

Should I now create a pull request?

guymcswain, 2020-03-01:
Yes.  Do the PR with respect to 'develop' branch first to see how it all plays out.

guymcswain, 2020-03-01:
I recommend we leave it in 'develop' until we get some feedback from users before merging into master.  I just got some pi4 hardware and would like to give it a go as another check.

guymcswain, 2020-03-02:
I wonder if something changed in latest Raspbian.  I downloaded latest '2020-02-13-raspbian-buster-lite', then:
```bash
wget https://github.com/joan2937/pigpio/archive/develop.zip
unzip develop.zip
cd pigpio-develop
make
sudo make install
```
and get this error:

> ...
> if which python3; then python3 setup.py install ; fi
> /usr/bin/python3
> Traceback (most recent call last):
>   File "setup.py", line 3, in <module>
>     from distutils.core import setup
> ModuleNotFoundError: No module named 'distutils.core'
> make: *** [Makefile:107: install] Error 1

I had to `sudo apt-get install python3-distutils`.  Not sure if this is worthy of a note in the installation documention.

guymcswain, 2020-03-02:
My script that bit-bangs I2C to the BSC slave is not working on Pi4.  Previously this worked on the other raspberries.  I set up the bb I2C master for GPIO 2 (SDA) and GPIO 3(SCL) - convenient for their on-board pullup resistors.  I jumper GPIO 2 to GPIO 10 and GPIO 3 to GPIO 11 to connect to BSC slave on Pi4.

BSCX api returns reasonable status information as I set the slave address.  But when I attempt the first write on the bus, I get `PI_I2C_WRITE_FAILED` returned from the BI2CZ api .

I'll try using different gpio pins, with external pullups, for the I2C master to see if that makes any difference.

joan2937, 2020-03-02:
I'll download "develop" and try some tests on a Pi4B and a Pi3B+.

joan2937, 2020-03-02:
I ran the tests at the end of http://localhost/pigpio/pigs.html#BSCX

It seems to work okay as far as they are concerned.

Pi4B
```
pi4b: $ pigs bscx 0x130305
1 18
pi4b: $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- 13 -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi4b: $ pigs i2co 1 0x13 0
0
pi4b: $ pigs i2cwd 0 90 87 51 9 23 # write 5 bytes
pi4b: $ pigs bscx 0x130305 # check for data
6 18 90 87 51 9 23
pi4b: $  pigs bscx 0x130305 11 13 15 17 # check for data and send 4 bytes
1 262338
pi4b: $ pigs i2crd 0 4 # read 4 bytes
4 11 13 15 17
pi4b: $ pigs i2cwd 0 90 87 51 9 23 # write 5 bytes
pi4b: $ pigs bscx 0x130305 11 13 15 17 # check for data and send 4 bytes
6 262338 90 87 51 9 23
pi4b: $ pigs i2crd 0 4
4 11 13 15 17
pi4b: $ pigs bscx 0x130305 22 33 44 55 66
1 327938
pi4b: $ pigs i2crd 0 5
5 22 33 44 55 66
pi4b: $ 
```
Pi3B+
```
pi3bp: $ pigs bscx 0x130305
1 18
pi3bp: $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- 13 -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi3bp: $ pigs i2co 1 0x13 0
0
pi3bp: $ pigs i2cwd 0 90 87 51 9 23 # write 5 bytes
pi3bp: $ pigs bscx 0x130305 # check for data
6 18 90 87 51 9 23
pi3bp: $  pigs bscx 0x130305 11 13 15 17 # check for data and send 4 bytes
1 262338
pi3bp: $ pigs i2crd 0 4 # read 4 bytes
4 11 13 15 17
pi3bp: $ pigs i2cwd 0 90 87 51 9 23 # write 5 bytes
pi3bp: $ pigs bscx 0x130305 11 13 15 17 # check for data and send 4 bytes
6 262338 90 87 51 9 23
pi3bp: $ pigs i2crd 0 4
4 11 13 15 17
pi3bp: $ pigs bscx 0x130305 22 33 44 55 66
1 327938
pi3bp: $ pigs i2crd 0 5
5 22 33 44 55 66
pi3bp: $ 
```

joan2937, 2020-03-02:
The following works as expected on a Pi3B+ and Pi4B.
```python
#!/usr/bin/env python

import time
import pigpio

I2C_ADDR=0x13

def i2c(id, tick):
    global pi

    s, b, d = pi.bsc_i2c(I2C_ADDR)

    if b:

        if d[0] == ord('t'): # 116 send 'HH:MM:SS*'

            print("sent={} FR={} received={} [{}]".
               format(s>>16, s&0xfff,b,d))

            s, b, d = pi.bsc_i2c(I2C_ADDR,
               "{}*".format(time.asctime()[11:19]))

        elif d[0] == ord('d'): # 100 send 'Sun Oct 30*'

            print("sent={} FR={} received={} [{}]".
               format(s>>16, s&0xfff,b,d))

            s, b, d = pi.bsc_i2c(I2C_ADDR,
               "{}*".format(time.asctime()[:10]))

pi = pigpio.pi()

if not pi.connected:
    exit()

# Respond to BSC slave activity

e = pi.event_callback(pigpio.EVENT_BSC, i2c)

pi.bsc_i2c(I2C_ADDR) # Configure BSC as I2C slave

time.sleep(600)

e.cancel()

pi.bsc_i2c(0) # Disable BSC peripheral

pi.stop()
```
```
h=$(pigs i2co 1 0x13 0)
pigs i2cwd $h 100 i2crd $h 11 -a
pigs i2cwd $h 116 i2crd $h 9 -a
```

guymcswain, 2020-03-02:
Yes, I'm able to run the above pigs i2c commands - after enabling the I2C peripheral in /boot/config.txt.

Perhaps something in my bit-bang I2C implementation.  Thanks for the quick verification.

guymcswain, 2020-03-02:
After enabling the I2C peripheral, the bit-bang script is now passing.  I guess the on-board resisters are somehow enabled/disabled via the I2C peripheral state?  Anyway, all is working now!

joan2937, 2020-03-02:
That's odd.  Might be worth adding your test script.

guymcswain, 2020-03-02:
Here it is.  Probably not helpful to those who don't use pigpio-client.
```javascript
/*  Bit-Bang I2C and BSC Slave Example - for Raspberry Pi4 (BCM2711)
 *
 *  Connect SDA jumper wire between pin 3 (gpio 2) and pin 19 (gpio 10).
 *  Connect SCL jumper wire between pin 5 (gpio 3) and pin 23 (gpio 11).
 */
const pi = require('./pigpio-client.js').pigpio({host: '192.168.1.133'}); // remote pi4

const s = require('debug')('i2c_slave');
const m = require('debug')('i2c_master');

const ready = new Promise( (resolve, reject) => {
  pi.once('connected', resolve);
  pi.once('error', reject);
});

const I2C_Message = "Hello I2C,123456"; // BSC FIFO is 16 deep

ready.then(async (info) => {
  console.dir(info);

  /* BSC slave */

  // configure
  let rv = await pi.bscI2C(0x13);
  s('bsc rv = ', rv);
  let rv_s = rv.slice(1, 5);
  s('bsc status = ', rv_s);
  if (rv[0] > 5) s('bsc data = ', rv.slice(5).map(x=>String.fromCodePoint(x)).join(''));
  else s('bsc read fifo empty');

  pi.on('EVENT_BSC', async() => {
    let [count, ...data] = await pi.bscI2C(0x13);
    let bsc_stat = data.slice(0, 4);
    data = data.slice(4);
    if (count>5) {
      s('BSC Rx status = ', bsc_stat);
      s('BSC Rx data =', data.map(x=>String.fromCodePoint(x)).join(''));
    }

    // write data to the BSC Tx FIFO
    if (data.length) {
      [count, ...bsc_stat] = await pi.bscI2C(0x13, data);
      s('BSC Tx status: ', bsc_stat);
    }
  });

  /* master */

  let i2c = await pi.bbI2cOpen(2, 3, 50000);
  let buf = I2C_Message;
  m('Sending: ', buf);
  await i2c.write(0x13, buf);

  // wait BSC to receive then echo the message
  // (Alternatively, you could use a gpio to notify the master.)
  await new Promise( (resolve, reject) => setTimeout(resolve, 1000) );

  // read the BSC data and convert to string
  let [count, ...data] = await i2c.read(0x13, I2C_Message.length);
  m(`reading ${count} bytes: `)
  m('message = ', data.map(x=> String.fromCodePoint(x)).join(''));

  await pi.bscI2C(0); // reset BSC peripheral
  await i2c.close();
  await pi.end();
  console.log('Goodbye');
  process.exit();

}).catch(console.error);
```

guymcswain, 2020-03-02:
Oh, forgot to mention that I had a wiring error on my jumper wires!  Late night, not enough caffeine this morning.

closed 2020-03-22 by guymcswain

pakluba, 2024-01-11:
But it has!
The signals are at GPIO8-11 ALT3 (so shared with SPI0).
The IRQ line is 43.
DMA request of TX is 8
DMA request of RX is 9
Source: https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf

commit ef48a043 (2020-03-01): Merge pull request #328 from joan2937/BCM2711-BSC-slave

Bcm2711 bsc slave.  Closes #280 and #327 .