joan2937/pigpio #282: pigpiod doesn't clean up connections

pablo-mendoza, 2019-08-04:
If I have a python script that connects to pigpiod (calls `pi = pigpio.pi()`) and such application doesn't call pi.stop() eventually pigpiod  will stop accepting connections (even when all previous connections have been closed).

I know that I should call pi.stop() but in some cases it's not possible (like the application crashes), the server should do the cleanup after the connection is terminated.

joan2937, 2019-08-05:
Please provide a small script which demonstrates the problem.

guymcswain, 2019-08-05:
V68 introduced TCP keepalive probes to detect and close dead peer connections.  Which version are you using?

pablo-mendoza, 2019-08-10:
I have  V70

python script to dupe is:

```
import pigpio
while True:
        pi = pigpio.pi()
```

eventually it will starting showing this error:

> %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
> Can't connect to pigpio at localhost(8888)
> 
> Can't create callback thread.
> Perhaps too many simultaneous pigpio connections.
> %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

At that point running it again will error out since the start. Checking `netstat -anlp|grep 8888` maybe show some TIME_WAIT connection, but even waiting for those to clear won't make it work again.

guymcswain, 2019-08-10:
To be fair, this 'script' just demonstrates that once all networking resources have been exhausted you will get errors.  Not a real situation.   The TCP timeout can be adjusted if this is causing a problem for you but I'm not seeing a problem for pigpiod.

pablo-mendoza, 2019-08-10:
What's the default timeout? I have waited for a while and it never works again.

guymcswain, 2019-08-10:
That depends on how your system is configured.  I usually set the networking parameters in my systemd unit like so:
```
ExecStartPre=/sbin/sysctl -w net.ipv4.tcp_keepalive_time=300
ExecStartPre=/sbin/sysctl -w net.ipv4.tcp_keepalive_intvl=60
ExecStartPre=/sbin/sysctl -w net.ipv4.tcp_keepalive_probes=5```

pablo-mendoza, 2019-08-11:
I put in your settings, exhausted  the connections, waited for 15minutes. 

Still can't create new connections. I have waited for an hour and still can't create new connections.

That's my whole point. Old connections that are long gone seem to be leaking resources.

guymcswain, 2019-08-11:
I think you should try one connection, kill it, then verify it gets cleaned up.  Exhausting all resources may cause a different issue.

pablo-mendoza, 2019-08-11:
Well, that's the bug I'm reporting. pigpio doesn't recover after  exhausting all connections. even after those connections are closed and long gone.

guymcswain, 2019-08-11:
Ok.  My (limited) testing on v68 showed the dead connections we're cleaned up.  I'll see if I can find time to go retest

pablo-mendoza, 2019-08-11:
I think this is a new issue, as I never ran into this problem with the version that ships with raspian strech (which I think is V68)

guymcswain, 2019-08-11:
BTW, you mentioned you are connecting to localhost.  So when your script crashes is it taking down the server (pigpiod) also?  

As I recall my testing was remote and I had to crash the remote client - like using a hard reset.  That would create the dead peer connection and pigpiod would, later, remove it.  If I simply killed the process on the remote machine the os would close the socket gracefully.

So I'm interested in how you are creating the dead peer connection.

pablo-mendoza, 2019-08-11:
I just run the above script in the same machine as pigpiod. The server doesn't die.

guymcswain, 2019-08-11:
But the above script is creating some kind of resource starvation issue which is possibly an altogether different issue.

I'm interested in your original case where you create a connection then somehow terminate the process leaving a dead connection.  I understand that if this is repeated over and over that it could lead to the resource starvation issue.  

We need to create a dead connection then show that pigpiod did not clean it up.  I'm not sure I know how to do that using local host connections.

pablo-mendoza, 2019-08-11:
It's the exact same issue, just run this:

```
import pigpio
pi = pigpio.pi()
```

over and over, eventually you will run into the same problem.

joan2937, 2019-08-11:
@pablo-mendoza 

The first example just creates too many simultaneous connections.  The daemon has no way of knowing you aren't planning to use one of those connections in the future so they will all be left open until resources are consumed.

In your second example I would expect the network connection to be closed at a timeout period after each process terminates.

pigpio NEVER releases resources until explicitly told to do so.  E.g. by a connection being closed (pi.stop), a handle being closed (i2c_close), or a connection failing (network timeout).

pablo-mendoza, 2019-08-11:
My whole point is that pigpio will NEVER release the resources even after the connection failed and timed out.

guymcswain, 2019-08-11:
>  even after the connection failed and timed out.

@pablo-mendoza  -  You have not demonstrated this point!

joan2937, 2019-08-11:
We may be talking at cross purposes.

If you open a handle (I2C, SPI, serial etc) that must always be manually released.  Crashing or aborting a script will not release those resources.

The network connection (i.e. pigpio.pi) will be released after a timeout period  after crashing or aborting a script..  When you use pigpio.pi() you additionally automatically get a notification handle.  That automatic notification handle should also be released after a timeout period after crashing or aborting a script.

pablo-mendoza, 2019-08-11:
>  When you use pigpio.pi() you additionally automatically get a notification handle. That automatic notification handle should also be released after a timeout period after crashing or aborting a script.

That's what I have been trying to say all along, it doesn't get released after aborting the script. I have waited up to 1hour.

joan2937, 2019-08-11:
I'll have another look at that code.  However if it is by design (which it may be, I do not remember) nothing will be changed.

guymcswain, 2019-08-11:
There are two issues in this discussion.  The first is does pigpiod clean up dead connections.  I contend that is does because there has been no evidence presented to the contrary.   The second issue is what happens to pigpiod when you exhaust all the network connections - does it enter a state where it cannot recover and needs to be restarted.  @pablo-mendoza 'script' demonstrates the second issue.

@joan2937, what is the maximum number of sockets that will pigpiod accept?

I will offer to test a script that opens some number of connections less than the maximum, exit script without pi.stop() (or throw exception to abort), then repeat the script.  If it runs successfully then it demonstrates that connections are indeed being released (as they should in a local host where the os is responsible for opening/closing).

joan2937, 2019-08-11:
I'm not sure if there is a maximum number of sockets.  I think that will be a system limit.

@pablo-mendoza is using Python which additionally opens a "notification" socket per connection.  The number of notifications is limited to 32 (PI_NOTIFY_SLOTS, pigpio.h).  So that gives an overall limit of 32 connections from all (concurrent) Python clients.

I thought they were released at network fail, perhaps they are not.

guymcswain, 2019-08-11:
Ok, so it should be possible to open 32, crash and burn, repeat forever without ever exhausting the connection resources.

joan2937, 2019-08-11:
@guymcswain That would probably be the intention.  I'll need to look at the code when I get some time.

pablo-mendoza, 2019-08-12:
I just tested that in the version in the raspbian repository (68) it will free the resources, but in V69 and V70 it does not.

guymcswain, 2019-08-14:
Just ran this simple script on V68:
```python
import pigpio
sockets = 1
pi = pigpio.pi()

while pi.connected:
  sockets += 1
  pi = pigpio.pi()
  
print "number of sockets = ", sockets
exit()
```
The script exits with sockets = 33 and the message "Can't create callback thread.  Perhaps too many simultaneous pigpio connections."  I can run this script repeatedly with the same output - ie pigpiod isn't thrown into some unresponsive state.  When I limit the number of sockets to 32 the message goes away.

Because V68 reports socket status at the user level, I was able to see the socket connections accepted then later close after the script exited.  I observed these messages using `cat /dev/pigerr&` from the command line.   Here's the output:

> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 5
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 5
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 6
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 6
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 7
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 7
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 8
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 8
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 9
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 9
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 10
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 10
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 11
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 11
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 12
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 12
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 13
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 13
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 14
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 14
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 15
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 15
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 16
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 16
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 17
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 17
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 18
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 18
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 19
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 19
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 20
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 20
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 21
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 21
> 
> 2019-08-14 15:10:50 pthSocketThread: Connection accepted on socket 22
> 2019-08-14 15:10:50 pthSocketThread: SO_KEEPALIVE enabled on socket 22
[... 194 lines cut ...]
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 30 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 27 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 32 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 26 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 25 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 23 closed
> 2019-08pi@raspberrypi:~ $ -14 15:10:50 pthSocketThreadHandler: Socket 21 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 19 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 24 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 22 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 13 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 11 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 18 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 20 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 15 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 9 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 17 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 12 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 16 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 10 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 5 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 7 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 14 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 8 closed
> 2019-08-14 15:10:50 pthSocketThreadHandler: Socket 6 closed

pablo-mendoza, 2019-08-14:
As I said this issue is not present in V698. It's only present in V69 and V70

joan2937, 2019-08-15:
I'll find some time today to have a look.  If it's a simple error on my part (as I expect) I'll push out a new release (V71) later today.  By that I mean within 16 hours of this comment.

joan2937, 2019-08-15:
I can't create erroneous behaviour with V70.

I am using this Python script (t.py):
```python
#!/usr/bin/env python

import time
import pigpio

t = int(time.time())

pi = pigpio.pi(show_errors=False)

if pi.connected:
   print("{} good".format(t))
else:
   print("{} bad".format(t))

time.sleep(40.0)

print("{} exit".format(t))
```

I run it from this bash script (r):
```bash
#!/bin/bash

./t.py &
```
This created the following output (I aborted the script at the end).
```
$ while ((1)); do ./r; sleep 1; done
1565864654 good
1565864655 good
1565864656 good
1565864657 good
1565864658 good
1565864659 good
1565864660 good
1565864661 good
1565864662 good
1565864663 good
1565864664 good
1565864665 good
1565864666 good
1565864667 good
1565864668 good
1565864669 good
1565864670 good
1565864671 good
1565864672 good
1565864673 good
1565864674 good
1565864675 good
1565864676 good
1565864677 good
1565864678 good
1565864679 good
1565864680 good
1565864681 good
1565864682 good
1565864683 good
1565864684 good
1565864685 good
1565864686 bad
1565864688 bad
1565864689 bad
1565864690 bad
1565864691 bad
1565864692 bad
[... 687 lines cut ...]
1565865013 exit
1565865014 exit
1565865015 exit
1565865016 exit
1565865017 exit
1565865018 exit
1565865019 exit
1565865020 exit
1565865021 exit
1565865022 exit
1565865023 exit
1565865024 exit
1565865025 exit
1565865026 exit
1565865027 exit
1565865028 exit
1565865029 exit
1565865030 exit
$ 
```
This is expected behaviour.

Have you got a script I can run which demonstrates the problem you are having?

I would need a complete runnable script to duplicate the problem.

pablo-mendoza, 2019-08-15:
@joan2937 I discovered this only duplicated when I run pigpiod with the -m option (the whole reason I upgraded from V68).

So start pigpiod -m and try your same script again and you will see that after the "bad"s starts it will never recover even after the "exit"s.

joan2937, 2019-08-15:
@pablo-mendoza I'm not sure what to do about this.  All sorts of things will stop working if you use the -m option.  Alerts stop, Python callbacks stop.  I'll need to think about this some more as any "fix" to cater for using the -m option may potentially break the bulk of code which doesn't use the -m option.

joan2937, 2019-08-15:
Released V71 which should clear this issue.

closed 2019-08-15 by joan2937

pablo-mendoza, 2019-08-15:
@joan2937 I can confirm the fix for work. Thanks a bunch! Great work.