joan2937/pigpio #283: number of i2c handles

bludin, 2019-08-05:
Is there a compelling reason for limiting the maximum number of I2C-handles to 64 and if not, could it be increased (e.g. to 512)? I have a situation where this becomes limiting (a modular construct where each of the up to 48 modules with up to 10 I2C-devices receives its own I2C-bus via 6 TCA9548 muxes). I can program my way around this limitation but it would be nicer if I didn't have to

Beat

PS: Thanks for pigpio which is just wonderful!

guymcswain, 2019-12-27:
I'm looking into implementing the I2C APIs in a client and found it a bit odd that not every device address on a bus could be reached without exhausting the handle resources.  Without impacting the current APIs or the user code built on them, would it be possible to add a new API that allows changing the address on the same handle?  Something like `i2cAddress(handle, addr)`.

joan2937, 2019-12-27:
Frankly I never expected there would be a use for so many handles.  Anyhow it's something people seem to request quite often so I'll accept the demand and increase the slots to 512.

Unless someone wants a larger number?

guymcswain, 2019-12-28:
Additional handles are not necessary and would consume more memory.  A single client/context can safely share the same handle.  What is needed is a method to change the I2C_SLAVE address on the current handle.

I would be willing to provide a PR to add the API to the Python client, pigpio.c and pigs (and anything else that is impacted) if you agree.  Of course this will take me a few weeks and I may need your guidance.

joan2937, 2019-12-28:
I really don't see that the effort of creating a PR would be worthwhile.  Changing the number of I2C slots uses an extra 7k bytes of memory which is trivial compared to the effort of doing anything else.

@guymcswain I have little time to devote to pigpio.  I am quite happy to add you as a collaborator which would allow you to steer pigpio in any direction you want.

guymcswain, 2019-12-28:
@joan2937 Allow me some time to give you a more thoughtful response but in general I would be happy to help you maintain this wonderful repository.  I would feel uncomfortable, however, to steer in any direction without you providing some guiding principles or vision.  Thank you.

guymcswain, 2020-01-02:
Fixed in version V73.

closed 2020-01-02 by guymcswain