joan2937/pigpio #285: Increasing Memory Usage when continuously sampling Accelerometer

DanTheMan2000, 2019-08-08:
I'm trying to sample an I2C accelerometer every 333ms indefinitely on a Raspberry Pi. However, the memory use of the application slowly increases over time and reaches over 500MB after one week.

Notice that I only initialize the connection once with pigpio.pi() and never stop the connection, is this the problem? Does the socket continue to allocate memory every time it is used? I was trying to avoid the overhead of opening and closing the connection every 333ms.

```
import threading
import time
import pigpio

# Accelerometer Address
addr = 0x19
BUS = 1
OUT_X_H = 0x29

class AccelerometerException(Exception):
    pass

class Accelerometer:

    def __init__(self):

        self.gpio = pigpio.pi()

        if not self.gpio.connected:
            raise AccelerometerException

        self.read_acceleration()
        accelerometer_thread = threading.Timer(1, self.accelerometer_read_thread)
        accelerometer_thread.daemon = True
        accelerometer_thread.start()

    def accelerometer_read_thread(self):
        while True:
            self.read_acceleration()
            time.sleep(.333)

    def read_acceleration(self):
        h = self.gpio.i2c_open(BUS, addr)
        if h >= 0:
            self.gpio.i2c_read_byte_data(h, OUT_X_H)
            self.gpio.i2c_close(h)
```

guymcswain, 2019-08-08:
What version of pigpio are you running?  Command line: `pigpiod -v`

DanTheMan2000, 2019-08-08:
@guymcswain The version is 68 and the handle is always 2.
I changed the code to call .pi() and .stop() every time that I sample the acceleration. I'll run it for a couple of hours and let you know if the memory is still increasing. Thanks

guymcswain, 2019-08-08:
Indeed if you call .pi() more than once you'll need to call .stop().  But in the above code you are just opening then closing a network socket handle which shouldn't cause the memory to leak.

guymcswain, 2019-08-08:
Also, why do you have three open handles - 2, 1 and 0?  Is there more code executing than what we see?

DanTheMan2000, 2019-08-08:
Yes, I do have another file of code that writes to an gpio pin. It run on a different thread than the acclerometer thread.

```
            import pigpio
            gpio = pigpio.pi()
            if gpio.connected:
                gpio.write(19, value)
                gpio.stop()
            else:
                logger.error('Lighting: Cannot access gpio')
```

DanTheMan2000, 2019-08-08:
This is the latest code:

```
import logging
import platform
import threading
import time

import pigpio

logger = logging.getLogger()

# Accelerometer Address
addr = 0x19
BUS = 1

CTRL_REG1 = 0x20
CTRL_REG2 = 0x21
CTRL_REG4 = 0x23
TEMP_CFG_REG = 0x1F
OUT_X_L = 0x28
OUT_X_H = 0x29
OUT_Y_L = 0x2A
OUT_Y_H = 0x2B
OUT_Z_L = 0x2C
OUT_Z_H = 0x2D
OUT_TEMP_L = 0x0C
OUT_TEMP_H = 0x0D

# CTRL_REG1 Bit Masks
ODR_OFF = 0x00  # Power-down mode
OD_1HZ = 0x10  # Normal mode (1 Hz)
OD_10HZ = 0x20  # Normal mode (10 Hz)
OD_25HZ = 0x30  # Normal mode (25 Hz)
OD_50HZ = 0x40  # Normal mode (50 Hz)
OD_100HZ = 0x50  # Normal mode (100 Hz)
OD_200HZ = 0x60  # Normal mode (200 Hz)
OD_400HZ = 0x70  # Normal mode (400 Hz)
L_PEN = 0x08  # This bit must be set to '1' for the correct operation of the device
X_AXIS = 0x04  # X-Axis enabled
Y_AXIS = 0x02  # Y-Axis enabled
Z_AXIS = 0x01  # Z-Axis enabled

# CTRL_REG2 Bit Masks
FDS = 0x08  # Filtered data selection

# CTRL_REG4 Bit Masks
BDU = 0x80  # Block data update
RANGE_2G = 0x00  # Full scale = +/-2g
RANGE_4G = 0x10  # Full scale = +/-4g
RANGE_8G = 0x20  # Full scale = +/-8g
RANGE_16G = 0x30  # Full scale = +/-16g

# TEMP_CFG_REG Bit Masks
TEMP_ENABLE = 0xC0  # Temperature sensor enabled

class AccelerometerException(Exception):
    pass

class Accelerometer:
    operational = False

    def __init__(self):
        self.h = 0
        self.x = 0
        self.y = 0
        self.z = 0
        if platform.system() is not "Windows":
            if not self.init_accelerometer():
                logger.error('Accelerometer: Initialization failed')
                raise AccelerometerException

            self.read_acceleration()
            accelerometer_thread = threading.Timer(1, self.accelerometer_read_thread)
            accelerometer_thread.daemon = True
            accelerometer_thread.start()
            logger.info('Accelerometer: Accelerometer initialized')
            self.operational = True

        else:
            logger.warning('Accelerometer: Not available on Windows OS')
            raise AccelerometerException

    def init_accelerometer(self):

        gpio = pigpio.pi()

        if not gpio.connected:
            logger.error('Accelerometer: Cannot access gpio, initialization failed')
            return False

        try:
            h = gpio.i2c_open(BUS, addr)
            if h >= 0:

                # Check if the handle is changing
                if h != self.h:
                    logger.info('Accelerometer: Handle changed to {}'.format(h))
                    self.h = h

                # Initialize accelerometer control register 1: Normal Power Mode and 50 Hz sample rate
                gpio.i2c_write_byte_data(h, CTRL_REG1, (OD_10HZ | L_PEN | X_AXIS | Y_AXIS | Z_AXIS))
                gpio.i2c_write_byte_data(h, CTRL_REG4, BDU | RANGE_2G)
                gpio.i2c_write_byte_data(h, TEMP_CFG_REG, TEMP_ENABLE)
                gpio.i2c_close(h)
                gpio.stop()
                return True
            else:
                logger.error('Accelerometer: Cannot get I2C bus handle')
                gpio.stop()
                return False
        except Exception as e:
            logger.error('Accelerometer: ' + str(e))
            gpio.stop()
            return False

    def accelerometer_read_thread(self):
        while True:
            self.read_acceleration()
            time.sleep(.333)

    def read_acceleration(self):
        gpio = pigpio.pi()
        if not gpio.connected:
            logger.error('Accelerometer: Cannot access gpio, initialization failed')
            return False

        h = gpio.i2c_open(BUS, addr)

        if h >= 0:

            # Check if the handle is changing
            if h != self.h:
                logger.info('Accelerometer: Handle changed to {}'.format(h))
                self.h = h

            g_range = 4.0  # +/-2g
            self.x = self.twos_complement(gpio.i2c_read_byte_data(h, OUT_X_H), 8) / 255 * g_range
            self.y = self.twos_complement(gpio.i2c_read_byte_data(h, OUT_Y_H), 8) / 255 * g_range
            self.z = self.twos_complement(gpio.i2c_read_byte_data(h, OUT_Z_H), 8) / 255 * g_range

            # Read to allow registers to be updated
            gpio.i2c_read_byte_data(h, OUT_X_L)
            gpio.i2c_read_byte_data(h, OUT_Y_L)
            gpio.i2c_read_byte_data(h, OUT_Z_L)
            gpio.i2c_close(h)
            self.operational = True
        else:
            self.x = 0
            self.y = 0
            self.z = 0
            self.operational = False
            logger.error('Accelerometer: Cannot get I2C bus handle')

        # Release resources before leaving
        gpio.stop()

    @staticmethod
    def twos_complement(input_value, num_bits):
        mask = 2 ** (num_bits - 1)
        return -(input_value & mask) + (input_value & ~mask)
```

guymcswain, 2019-08-08:
> I was trying to avoid the overhead of opening and closing the connection every 333ms.

Not only is .pi() / .stop() unnecessary in read_acceleration() but so is opening / closing the handle each time you take readings.  I suggest you go back to your original code but keep an instance copy of the handle - `self.h` or whatever.  

It's not obvious to me why you are experiencing a memory leak.  Are you are sure the memory leak is associated this app's process?

joan2937, 2019-08-08:
Is that all the code?

I know it's a matter of style but personally I would only call pigpio.pi() and pi.stop() once at the start/stop of the script.  Similarly I would only open/close any handles (like the I2C handles) once per script.

DanTheMan2000, 2019-08-08:
No, that is not all the code. However, I did find the root cause of the issue using tracemalloc. The Accelerometer was inheriting another class what was causing the problems.

closed 2019-08-08 by DanTheMan2000