joan2937/pigpio #287: bad gpio errors and segmentation faults under buster

UKTailwind, 2019-08-19:
I've been using PIGPIO as part of a large application (Pi-cromite) for some time without issues under Jessie and Stretch. Just tried to port my application to Buster (Pi Zero W). PIGPIO installs and passes "sudo ./x_pigpio # check C I/F" test with both 6.8 and 7.1 but my application gives errors after compiling with either release. I'm using the C I/F. PIGPIOD is not running.

> pi@raspberrypi:~ $ sudo ./mmbasic
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioWrite: bad gpio (68)
2019-08-19 18:24:46 gpioSetMode: bad gpio (68)
2019-08-19 18:24:46 gpioSetMode: bad gpio (68)
2019-08-19 18:24:46 gpioSetPullUpDown: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioSetMode: bad gpio (68)
2019-08-19 18:24:46 gpioWrite: bad gpio (68)
2019-08-19 18:24:46 gpioSetMode: bad gpio (68)
2019-08-19 18:24:46 gpioWrite: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioSetMode: bad gpio (68)
2019-08-19 18:24:46 gpioSetMode: bad gpio (68)
Segmentation fault
pi@raspberrypi:~ 

> $

joan2937, 2019-08-19:
pigpio only supports the Pi GPIO.  No Pi has a GPIO numbered greater then 53.

UKTailwind, 2019-08-19:
I know. I don't address anything greater than 27. This is the error. Something about PIGPIO on the Pi Zero W under buster is causing the problem . This is code that has run on many peoples' Pis for an number of years under Jessie and Stretch without issues. All I did is re-compile under buster and I get this error. The error occurs with both versions 68 and 71 both of which were built under buster before compiling th eC source.

joan2937, 2019-08-19:
I don't really see what help I can give.  I have no idea what the mmbasic command you are running does.  It is not part of pigpio.

UKTailwind, 2019-08-20:
This is happening as part of initialisation - nothing to do with any Basic commands

`        gpioCfgClock(1,Option.clock,0);
        gpioCfgMemAlloc(1);
        gpioCfgDMAchannels(4,5);
        gpioInitialise();
        gpioSetPad(0,10);
        gpioSetTimerFunc(0, 10, T4Interrupt);
`

Then I loop and set all valid pins up to 27 but less the reserved I2C pins

gpioSetPullUpDown(pin, PI_PUD_UP);

I was hoping you might have some insight into why buster is behaving differently to Jessie and Stretch where I have no issues. Have you seen anything remotely similar?

joan2937, 2019-08-20:
All the above diagnostics are referring to GPIO 68.

```
pi@raspberrypi:~ $ sudo ./mmbasic
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
2019-08-19 18:24:46 gpioGetMode: bad gpio (68)
```
Those messages are apparently being generated by the mmbasic command.

I only use buster on the Pi4B.  I don't have any problems.

If you want help you need to provide a short program which I can run which demonstrates the problem (C, Python, or pigs).

UKTailwind, 2019-08-20:
I'm making some progress. The issue seems to be related to a segmentation fault which is happening on buster but not on earlier releases. I'm getting a PIGPIO diagnostic report when I close PIGPIO - dos this give any useful info?

> sudo ./mmbasic

#####################################################
pigpio version=71 internals=200
micros=1 allocMode=1 dmaInitCbs=2 DMARestarts=1
samples 0 maxSamples 0 maxEmit 0 emitFrags 0
cbTicks 14106, cbCalls 8428
pipe: good 0, short 0, would block 0
alertTicks 8173, lateTicks 0, moreToDo 262
        1         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0        12        41        54        54        46        97      2567     46486    371256     46709      2459       100        35        46        64        67        56        57        80        53        20         9         3         6         1         1         0         0         0         0         1         0         4
#####################################################

> 
I'm beginning to suspect that the issue is a timing one - I'm running the clock at maximum speed to get fastest possible I/O

        gpioCfgClock(1,Option.clock,0);

joan2937, 2019-08-20:
The diagnostics have been printed because the DMA "engine" was restarted.  It is not something which normally happens.  I'd be tempted to leave the program running for 10 minutes and see how many restarts there were.  If just the one it can be ignored.

Why the `gpioCfgMemAlloc(1);`?  If you are not doing something odd I would remove that call as using the default mbox method is a better choice.

UKTailwind, 2019-08-20:
Joan - thanks for your help - together we have solved it.

The problem is a change in Buster with DMA usage. It appears DMA 4 is no longer available and this was causing the problem. Also the old method of getting a list of DMA usage doesn't work either i.e.

cat /sys/module/dma/parameters/dmachans

I've posted a request for info on the Raspbian forum but don't expect much

closed 2019-08-20 by UKTailwind