joan2937/pigpio #288: Snap package

nils-bauer, 2019-08-22:
Hi Joan,
I'm creating a [snap package](https://snapcraft.io/) of pigpio, thereby it can be used, inter alia, on  [Ubuntu Core](https://ubuntu.com/core).
Do you want to include the snapcraft packaging in this repo (I can create a pull request) and [set the automatic build up](https://snapcraft.io/build)?
If you don't, I will maintain the build progress externally.

Best regards,
Nils

joan2937, 2019-08-22:
That's an interesting idea.

I don't want to add this to the repository but I am quite happy to link to your snap from the downloads page (and perhaps other places at your suggestion).

nils-bauer, 2019-08-22:
Okay, thank you.
I'll close this issue now and open it again when it is finished.

closed 2019-08-22 by nils-bauer