joan2937/pigpio #289: Option to disable pid file

nils-bauer, 2019-08-22:
Hi,
I'm creating a snap package for  pigpio (#288). Snaps aren't allowed to write to `/var/run/pigpio.pid`.
Can you please add an option to disable the pid file creation or to change the file's location?

Best regards,
Nils

dkulp, 2019-08-22:
Was just about to log the same issue.   If gpioCfgInterfaces(PI_DISABLE_FIFO_IF | PI_DISABLE_SOCK_IF); is called before gpioInitialise(), there isn't really a way to communicate with the process anyway so having the pid file is relatively pointless.

joan2937, 2019-08-22:
The pid file is used to prevent attempts to run two instances of the daemon.  It has never had any other purpose.

nils-bauer, 2019-08-22:
Yes, I know. But for snapcraft it is needed to disable the pid file or change it's location.

@joan2937 can you please add a parameter to pigpiod to archive this?

nils-bauer, 2019-08-28:
Or would you accept a pull request?

cinderblock, 2019-10-04:
Is there no better way to lock pigpio to only running a single instance than a static file location? Maybe a flag in the DMA module? What happens when it crashes? Does the temp file get left around?

guymcswain, 2020-01-03:
I'm inclined to support the snap package (externally initially).  It would be a great benefit for pigpio.  Can you provide a proposal here (a PR if it is really simple) for insuring only a single instance of the daemon can run?

cinderblock, 2020-03-24:
Did you mean to close this @guymcswain?

guymcswain, 2020-03-24:
Yes.  I hadn't seen any response to my request for a proposal so assumed it had become stale.

cinderblock, 2020-03-24:
I see. Is this still a feature you'd like implemented? Or would you accept a PR if someone fixed this?

If the answer is yes, imho, this "issue" should be left open. But it's your project. Do as you like :)

guymcswain, 2020-03-24:
Perhaps I was too hasty.  Yes, a PR would be welcome.

joan2937, 2020-03-24:
I don't understand why this can't be solved by the snap.  I'm not a fan of making changes to cater for something which should be handled elsewhere.

Why not just grant the snap access to /var/run/pigpio.pid ?

cinderblock, 2020-03-26:
Frankly, I'm interested in this option just to not use the file system

joan2937, 2020-03-26:
Impossible with pigpio.  It makes extensive use of the file system.  I can't think of a GPIO library which does not.

cinderblock, 2020-03-27:
@joan2937 My understanding of how pigpio works is that it does not use the filesystem at all for normal GPIO access. It only uses the "filesystem" for a couple thing, only one of which is a traditional "file":
- `/dev/mem` - to actually control DMA and GPIO outputs
- `/dev/pig{pio,out,err}` - `pigs` control
- `/var/run/piogpio.pid` - DMA Resource mutex

Many other [GPIO libraries](https://docs.google.com/spreadsheets/d/1sFCJuPZ9k5GN0A6j7gHRckvbwIdnwzPH1sTCuPNFNRQ) use SysFS to control GPIO. Is this what you're talking about?

I trust you're saying something accurate, I'm just not following I guess.

guymcswain, 2020-08-29:
Marking this with 'no action' and will close unless a proposal is provided.

cinderblock, 2020-09-01:
I propose using some unused byte/bit of memory to flag that Pigpio is running and not use a pid file as a lock file

guymcswain, 2020-09-01:
Where do you find this bit/byte of persistent memory that is unused?

cinderblock, 2020-09-01:
1. Why close this issue? Is the intention to not ever make a change like this? Removing the dependency on a file system seems perfectly reasonable to me.
2. I'm supposing that there is some bit/byte in ram that could just be taken over by pigpio to be used as a global lock. I'm not familiar enough with how BCM's DMA/Timer peripherals are used but I suspect someone that is familiar would be able to pretty quickly point to one such unused byte/bit in the used peripherals' memory space.

The issue that I'm seeing is the configuration option for which timer/DMA mode is used. This dedicated place to look would have to work for all possible configuration options (DMA allocation mode, DMA Channels, and Clock peripheral).

Maybe there is some "unused" byte in the GPIO peripheral memory space that could be usurped for this purpose. In practice, this is equivalent to a `/var/run/piogpio.pid` file. There just needs to be a way to force delete/ignore a broken "lock" bit.

guymcswain, 2020-09-01:
I think it's clear that it's a non trivial change so it begs the question of what problem are you solving?  The op was offering to contribute a snap package but it seems has moved on.

cinderblock, 2020-09-03:
To put it simply, being able to store a lock state with no dependency on the file system to store the lock file. Frankly, it just feels like a "cleaner" solution to me, assuming there is some location that works.

closed 2020-09-29 by guymcswain