joan2937/pigpio #291: bscXfer problems

Yardie-, 2019-09-17:
Hi Joan
Firstly thank you for pigpio I love it.
However I am having a problem with bscXfer
I was having a problem sending numbers as 4byte unsigned char blocks 
so I went back to basics
so my first question is this (in advance sorry about the arduinoiness of the code.)

the code below is returning a continual stream of 
status = 1030 and txBuf contains hello
I am actually curious why the code even fires at all
as there is no one else on the i2c bus

`#include <pigpio.h>
#include <string.h>
#include <stdio.h>
bsc_xfer_t xfer;
int status = 0;
void begin (uint8_t address)
{
  gpioInitialise();
  xfer.control = (address<<16) | 0x305;
}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()
void requestEvent() {
  // as expected by master
  memcpy(xfer.txBuf, "hello", 5);
  printf("status = %d and txBuf contains %s\n",status, xfer.txBuf);
  xfer.txCnt = 5;
}

void setup()
{
  uint8_t address = 8;
  begin(address);
}

void loop()
{
  status = bscXfer(&xfer);
  if (status >= 0)
  {
    requestEvent();
  }

}

int main()
{
  setup();
  while(1)
    loop();
  return 0;
}
`

joan2937, 2019-09-17:
The returned status should be as given in http://abyz.me.uk/rpi/pigpio/cif.html#bscXfer

Does that make sense for your example?

Yardie-, 2019-09-17:
ok thanks for the quick reply.
I sort of thought it would only return if the buffer changed 
RTFM wins everytime. thanks.
I have some other issues but I will close this for now 
Thank you.

closed 2019-09-17 by Yardie-

joan2937, 2019-09-18:
No problem.  I know the documentation can be misleading. Unfortunately I always find something else I prefer to do rather than make it better! That's life.