joan2937/pigpio #293: Documentation of pigs is difficult for new user

BryanCrotaz, 2019-09-20:
pigs commands are sorted alphabetically, not grouped by function. This can mean that get and set are separated by a long way in the list.

Is there a repo for the docs that I can submit a PR?

joan2937, 2019-09-20:
Pretty much all the documentation (and the web pages) are auto generated from the source code.

pigs is different in that there is no source code as such.

```
INTRO

The socket and pipe interfaces allow control of the GPIO by passing
messages to the running pigpio library.

The normal way to start the pigpio library would be as a daemon during boot.

...
sudo pigpiod
...

pigs is a program and internally uses the socket interface to pigpio
whereas /dev/pigpio uses the pipe interface.

*Features*

o hardware timed PWM on any of GPIO 0-31

o hardware timed servo pulses on any of GPIO 0-31

o reading/writing all of the GPIO in a bank as one operation

o individually setting GPIO modes, reading and writing

o notifications when any of GPIO 0-31 change state

o the construction of output waveforms with microsecond timing

o I2C, SPI, and serial link wrappers

o creating and running scripts on the pigpio daemon

*GPIO*

ALL GPIO are identified by their Broadcom number.

*Usage*

pigs and the socket interface share the same commands and are invoked in
a similar fashion from the command line.

The pigpio library must be running, either by running a program linked
with the library or starting the pigpio daemon (sudo pigpiod).

pigs {command}+

echo "{command}+" >/dev/pigpio

pigs will show the result of the command on screen.

The pigs process returns an exit status (which can be displayed with
the command echo $?).

...
PIGS_OK            0
PIGS_CONNECT_ERR 255
PIGS_OPTION_ERR  254
PIGS_SCRIPT_ERR  253 
...

The results of /dev/pigpio commands need to be read from /dev/pigout,
e.g. cat /dev/pigout (try cat /dev/pigout& so that all subsequent
results are shown on screen).

In both cases if an error was detected a message will have been written
[... 3270 lines cut ...]
RL y x @Rotate left register x bits                  @*y<<=x; F=*y
RLA x  @Rotate left accumulator x bits               @A<<=x; F=A
RR y x @Rotate right register x bits                 @*y>>=x; F=*y
RRA x  @Rotate right accumulator x bits              @A>>=x; F=A
STA y  @Store accumulator in register                @y=A
SUB x  @Subtract x from accumulator                  @A-=x; F=A
SYS str@Run external script (/opt/pigpio/cgi/str)    @system(str); F=A
TAG L  @Label the current script position            @N/A
WAIT x @Wait for a GPIO in x to change state         @A=wait(x); F=A
X y1 y2@Exchange contents of registers y1 and y2     @t=*y1;*y1=*y2;*y2=t
XA y   @Exchange contents of accumulator and register@t=A;A=*y;*y=t
XOR x  @Xor x with accumulator                       @A^=x; F=A

x may be a constant, a parameter (p0-p9), or a variable (v0-v149).

y may be a parameter (p0-p9), or a variable (v0-v149).  If p or v isn't
specified y is assumed to be a variable.

The EVTWT command parameter is a bit-mask with 1 set for events of interest.

The WAIT command parameter is a bit-mask with 1 set for GPIO of interest.

The SYS script receives two unsigned parameters: the accumulator A and
the current GPIO levels.
```

BryanCrotaz, 2019-09-20:
ok, found it in `/pigs.1`

What markdown language is this using?

joan2937, 2019-09-20:
My own.  As is all the other documentation.

The text I gave is the source,  pigs.1 is built from that source as is all the pigs documentation.

The source is used for the man pages and the html.  The man pages and html are generated by a series of bash and Python scripts.

The syntax is simple (just look at the results in the man and html).

If  I did this again I might use something like readthedocs.

closed 2020-06-28 by guymcswain