joan2937/pigpio #296: Error Msg "sigHandler: Unhandled signal 6, terminating" // Newbie code review wanted

dlukacic, 2019-09-25:
Dear community, 

I want to send a 32-bit serial stream to one gpio pin on an Raspberry Pi3 (Bitlenght 800us) using following code:

##########################
#include <stdio.h>
#include <pigpio.h>

//Broadcom Nr. 17, bzw. Hardware Port Nr. 11
// hier habe ich meinen 433MHz Transmitter Input Port angeschlossen
#define myDL433TX_BCMPort 17

//Oase FM Definitionen 
#define oase_baud 1250   //800us Bitlaenge = 1250Hz, Baudrate
#define oase_data_bits 32 //Bitlänge eines Datenwortes
#define oase_stop_bits 2  //  2...8 mögliche Stopbits
#define oase_offset_time 800 //800us Offset bis die Sequenz beginnen soll
#define oase_numBytes 4 //32 data bits => 4 bytes fuer numBytes
#define strOaseFM_CH1_OFF "1001001011001001011001001011001001001011011001011001001001001001001001011"
#define strOaseFM_CH1_ON "1001001011001001011001001011001001001011011001011001001001001001001011001"

int main()
{

  //Bibliothek initialisieren
  if (gpioInitialise()<0) return -1;
  
  gpioSetMode(myDL433TX_BCMPort, PI_OUTPUT); // Set 433 tx pin as output.

  gpioWaveClear();
  gpioWaveAddSerial(myDL433TX_BCMPort,oase_baud,oase_data_bits,oase_stop_bits,oase_offset_time,oase_numBytes,strOaseFM_CH1_OFF);

  int wave_id=gpioWaveCreate();
  //Transmission starten
  gpioWaveTxSend(wave_id,PI_WAVE_MODE_ONE_SHOT);
  gpioWaveTxSend(wave_id,PI_WAVE_MODE_ONE_SHOT);
  gpioWaveTxSend(wave_id,PI_WAVE_MODE_ONE_SHOT);

  gpioWaveDelete(wave_id);

  gpioTerminate();

}
###########
I compile with: gcc -Wall -pthread -o myDL433TX_OaseFM2 myDL433TX_OaseFM_v2.c -lpigpio -lrt

The following message is displayed when running the program:
sudo ./myDL433TX_OaseFM2 
*** stack smashing detected ***: ./myDL433TX_OaseFM2 terminated
2019-09-25 19:11:53 sigHandler: Unhandled signal 6, terminating

Any help or suggestions is warmly welcome.

Thanks,
Davor

joan2937, 2019-09-25:
One logic problem is starting a new wave before the old one has finished.

```C
gpioWaveTxSend(wave_id,PI_WAVE_MODE_ONE_SHOT);
while (gpioWaveTxBusy()) usleep(1000);
gpioWaveTxSend(wave_id,PI_WAVE_MODE_ONE_SHOT);
while (gpioWaveTxBusy()) usleep(1000);
gpioWaveTxSend(wave_id,PI_WAVE_MODE_ONE_SHOT);
while (gpioWaveTxBusy()) usleep(1000);
```

Are you sure you should be sending data formatted as serial data?  That feels wrong for a wireless link.

dlukacic, 2019-09-25:
What is your suggestion to generate a ASK modulated 32bit long 433MHz signal on a GPIO?

joan2937, 2019-09-25:
Have a look at http://abyz.me.uk/rpi/pigpio/examples.html#pdif2__433D and http://abyz.me.uk/rpi/pigpio/examples.html#Python__433_py

dlukacic, 2019-09-26:
Thank you Joan, if I understood correctly, your suggested librariers for 433 transmitter use manchester coded signals, my signal should be NRZ coded. Any further hints? Thank you very much for your support.

joan2937, 2019-09-26:
Not really.  You need to construct the waveform.  I know my pdif doc says Manchester encoding I seem to remember it might not be.  It could even be RTZ for all I know.

Anyhow I am closing this as it's not really a pigpio issue.

You might be better off asking about the encodings on something like the Raspberry Pi forums.

closed 2019-09-26 by joan2937