joan2937/pigpio #298: restarting pigpiod result in "bind to port 8888 failed (Address already in use)"

janvda, 2019-10-04:
I have defined a docker service (in balena) that runs the pigpiod daemon.
The problem is that when I try to restart that service, I am getting the following error:

```
04.10.19 08:35:11 (+0200)  pigpiod  2019-10-04 06:35:11 initInitialise: bind to port 8888 failed (Address already in use)
04.10.19 08:35:11 (+0200)  pigpiod  Can't initialise pigpio library
```

I strongly think that the operating system is not allowing immediately to rebind to the same port.  So you should wait for TIME_WAIT period before a process again can bind to the same port.

janvda, 2019-10-04:
For more details see also https://github.com/janvda/balena-node-red/issues/1

janvda, 2019-10-04:
This issue can be resolved by using the socket option SO_REUSEADDR (see https://stackoverflow.com/questions/21967311/linux-refuse-to-open-listening-port-from-localhost).

So it would be nice if you would consider making this change.

guymcswain, 2020-01-03:
I checked the code.  SO_REUSEADDR is being used.

guymcswain, 2020-01-04:
Please retest and provide more information if its still an issue.  I tried to follow your links but found evidence the problem may have been resolved.

maxnet, 2020-01-04:
You currently have:

```
            int opt;
            setsockopt(fdSock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));
```

But think it should be instead:

```
            int opt = 1;
            setsockopt(fdSock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));
```

guymcswain, 2020-01-05:
@maxnet, You are correct!  Thank you for reviewing the code and finding this oversight.  Meanwhile, I need to bolster the testing around the network features.

guymcswain, 2020-01-10:
@janvda , please retest with the patch for this issue in the 'develop' branch of this repo.

closed 2020-02-03 by guymcswain

dachshund-digital, 2021-05-03:
What this fix implemented?  In version 78 or 79?  I just encountered it on Pi OS buster Debian 10 (fully patched as of today).  Happen to be running 78.

May 03 00:12:29 pi3modelb0 pigpiod[32482]: 2021-05-03 00:12:29 initInitialise: bind to port 8888 failed (Cannot assign requested address)
May 03 00:12:29 pi3modelb0 pigpiod[32482]: Can't initialise pigpio library
M

guymcswain, 2021-05-03:
Yes both 78, 79.  It was implemented in earlier version but don't recall which off the top of my head.

commit bab05ceb (2020-02-03): Merge pull request #321 from joan2937/Release-v74

Release v74: closes #274, closes #298, closes #266, closes #316, closes #223.