joan2937/pigpio #299: Hardware-time PWM on a single PWM channel

nikivanov, 2019-10-18:
Hi,

I'm working on a raspberry pi-enabled open source rover: https://github.com/nikivanov/watney . I'm using PCM for I2S and PWM for sound, so I can only use software-generated PWM signals for controlling the motors and the servo, which makes them rather jittery. I'm only using 1 channel for audio, though. Raspberry Pi has 2 hardware PWM channels, and if I understand it correctly, they are both used for stereo sound. I was wondering if it's possible to run pigpio using only one PWM channel and use the other one for audio. Thank you!

joan2937, 2019-10-18:
The problem as I see it is that the source clock is shared by both PWM channels.  pigpio sets the source clock to PLLD.   I'm not sure what clock audio uses.  If it is PLLD then that's all well and good.  If it isn't that's a point of breakage.

Even if audio does use PLLD it would probably have to be altered to only affect one channel.  That is outside the scope of pigpio.  Apart from setting the clock pigpio will only change the channel it is told to change.  So if told to send PWM on channel 0 any audio on channel 1 would be unaffected.

guymcswain, 2019-11-17:
@nikivanov  I like your robot/Rover!  If you only need single channel audio, I've seen posts on forums showing how to get both mic and speaker using I2S.  If possible, that would leave the PWM peripheral free for DMA timed PWM for your servos.

nikivanov, 2019-11-19:
@joan2937 Sorry, I forgot to follow up on this! Thank you for your reply. I'm afraid your suggestions are beyond my knowledge of hardware clocks and low level linux settings. I'm going to see if I can get help from somebody at work with that knowledge.

@guymcswain Thank you for your suggestion! Low cost is a major concern with Watney - getting both audio input AND output would require me to get different audio hardware, which would drive the cost up and involve significant redesign. I think my best option is to use an angle usb adapter and add Arduino Nano slaved to the Pi.

guymcswain, 2019-11-19:
What about using a PDM microphone, and use the pigpio lib to get timed samples of the audio that you can decode in your app.  PCM is now available to drive pigpio timing for the servos etc.

closed 2020-01-18 by guymcswain