joan2937/pigpio #300: Question example "SPI bit bang MCP3008"

u0078867, 2019-10-22:
I am going to use this great library to read simultaneously from some MCP3008 ADCs.
I don't have to do much more than what is already in the example "SPI bit bang MCP3008", so I was studying it in detail. My target acquisition frequency is 2ksps, so much below the nominal max 25ksps.

I have the following questions:

1) `REPEAT_MICROS` is the inverse of my target sampling frequency. But what is the `BUFFER` for? I don't understand its meaning from: *We need to ensure that the buffer is big enough to cope with any reasonable rescehdule. In practice make the buffer as big as you can.*

2) `usleep(1000);` this should wait for 1ms. But if the sampling frequency of the example is 25 ksps (`#define REPEAT_MICROS 40`), doesn't it create an excessive delay in each sample iteration?

3) Would you suggest to run the example on a isolated RPi core, perhaps even excluding it from the Linux scheduling?

Thanks in advance!!!

joan2937, 2019-10-24:
The example uses DMA to transfer the SPI data.  In effect this isolates it from anything happening in Linux (which is why the samples are read a accurately timed intervals).

DMA writes the samples to a cyclic buffer.  As long as you empty the buffer faster than it is filled you will get good data.

Using this technique there is no advantage in isolating a core.

u0078867, 2019-10-24:
Thanks, [this](https://github.com/hzeller/rpi-gpio-dma-demo/blob/master/gpio-dma-test.c) helped me getting the essential parts of your code.

closed 2020-01-04 by guymcswain