joan2937/pigpio #301: I2C with 16 bit register addresses and 16 bit data with MSB first

DarkJediNinja, 2019-10-24:
Hi, Joan! Thanks for such a wonderful library.
I'm trying to program an audio codec on a Raspberry Pi 3B and using pigpio.c from my C program for interfacing with gpio and i2c.
This audio codec uses 16 bit register addresses and a 16 bit register values. In both cases, the most significant byte goes first. So, I cannot use i2cReadWordData() nor i2cWriteWordData() because they use 8 bit register addresses and LSB of data word goes first.
I tried to write some custom functions equivalent to the previous ones with the needed changes, based on i2cReadDevice() and i2cWriteDevice(), but I cannot make them work.
I think the reason is because a stop bit (P) is inserted between the write and read calls and I don't know how to avoid it.
Could you please tell me how to solve this problem or what is wrong in my code?

And still one more question:
The codec chip also supports fast I2C (400 kbps). How do I tell the I2C driver to use fast speed instead of the standard 100 kbps speed? 

These are the 2 functions I've coded to customize i2cWriteWordData() and i2cReadWordData:

```
// Special version of i2cWriteWordData with 16 bit register addresses
// and MSB first for both register address and data
int i2cWriteWordData16MSB(uint handle, uint i2cRegAddr, uint i2cRegValue) {
	char buf[4];
	buf[0] = (i2cRegAddr >> 8) & 0x00FF;
	buf[1] = i2cRegAddr & 0x00FF;
	buf[2] = (i2cRegValue >> 8) & 0x00FF;
	buf[3] = i2cRegValue & 0x00FF;
	return i2cWriteDevice(handle, buf, 4);
}
```

```
// Special version of i2cReadWordData with 16 bit register addresses
// and MSB first for both register address and data
int i2cReadWordData16MSB(uint handle, uint i2cRegAddr) {
	char buf[2];
	int result;
	buf[0] = (i2cRegAddr >> 8) & 0x00FF;
	buf[1] = i2cRegAddr & 0x00FF;
	if ((result = i2cWriteDevice(handle, buf, 2)) == 0) {
		if ((result = i2cReadDevice(handle, buf, 2)) == 2) {
			result = ((uint) buf[0]) << 8 | (uint) buf[1];
		}
	}
	return result;
}
```

joan2937, 2019-10-24:
You need a `/boot/config.txt` entry to change the I2C bus speed.  Have a look at `/boot/overlays/README` for help.

I suggest you use the raw writes rather than writing your own.

http://abyz.me.uk/rpi/pigpio/cif.html#i2cWriteDevice

DarkJediNinja, 2019-10-24:
Thanks for your quick reply.

Regarding the 2 custom functions, I'm trying to emulate the behavior of i2cWriteWordData and i2cReadWordData by doing calls to the 2 raw functions in pigpio: i2cWriteDevice and i2cReadDevice and handling the swapping of bytes and the 16 bit register addresses inside the 2 custom functions. That is because I had written a lot of code with your standard i2cWriteWordData and i2cReadWordData functions and I preferred to keep that code unchanged.

The point is that I'm using the raw functions but I still don´t know how to swap from the write to the read without adding a stop bit.

pbanasik, 2019-11-01:
I just run into the same issue trying to communicate with a VL6180X sensor using 16-bit register addresses. I'm using the **pigpiod** C I/F. It works with e.g. this function for reading a 16-bit value from a register:

```
uint16_t VL6180X::readReg16Bit(uint16_t reg)
{
  char buf[2];
  buf[0] = reg >> 8;
  buf[1] = reg & 0xFF;
  i2c_write_device(pigpiod_pi_id, i2c_id, buf, 2);
  i2c_read_device(pigpiod_pi_id, i2c_id, buf, 2);
  return (buf[0] << 8) | buf[1];
}

```

closed 2019-11-02 by DarkJediNinja