joan2937/pigpio #302: I2S in slave mode and DMA

DarkJediNinja, 2019-10-24:
Hi, Joan! Back here with another question.
I'm trying to use a codec to get digital audio data into a Raspberry Pi 3B.
To minimize CPU burden, I want to use the I2S interface with DMA support, so that the CODEC writes data directly into a buffer in RPi memory for later processing by the ARM CPU.
Is there any support in pigpio for I2S? And for DMA transfers?

If not, could you please give me a hint of where should I start looking?

Reading some posts, I saw that, in order to use I2S, I have to setup pigpio to use PWM clock instead of PCM clock. In that post, you said that this would disable PWM audio on the RPi. Is that also applicable to the RPi 3 on board headphone jack that uses PWM for audio wave modulation?

joan2937, 2019-10-24:
I can't help you with I2S as I have no knowledge of that area.

I think your best bet would be to outline what you are trying to do as a question on the Raspberry Pi Org forums.  Once you talk about codecs it sounds like you need Linux kernel involvement.

https://www.raspberrypi.org/forums/

DarkJediNinja, 2019-10-24:
Thanks for your help!

closed 2019-10-24 by DarkJediNinja