joan2937/pigpio #303: Doorbell Button Receiver

cslehel, 2019-11-11:
Hello, 

its not really an issue but a question. 
I tried the _433.py example with a key fob and its working great. 
Could be used to read doorbell buttons ? 
I tried with a Legrand doorbell but without success. 

Thank you for your attention.

joan2937, 2019-11-12:
The _433.py example assumes a particular data encoding, in this case the encoding commonly used with key fobs.

See if you can capture the signal with [piscope](http://abyz.me.uk/rpi/pigpio/piscope.html).  If you can and it is repeatable then there is a chance that pigpio could interpret the signal (given some software effort).  If the signal appears to be random noise then you will be wasting your time.

cslehel, 2019-11-12:
Thank you very much for the explanation.

closed 2019-11-12 by cslehel

Kitoko86, 2023-09-22:
Hello,

I try to capture the signal from a kinetic switch for a wireless doorbell on a Raspberry Pi 4 with piscope and the pigpio library. My RF module receiver is a KS-RXB6 shipped with AGC. The noise management by the AGC circuit is my main issue. When there is no signal, the AGC increases the gain level and of course, the noise level also. However, if I press the kinetic switch, I can see a quick drop in gain level but the received signal shows a sequence of short and long pulses in piscope but it doesn't match the protocol from the kinetic doorbell. 

The protocol itself is made of a short pulse of 50 µs as sync bit for "0" followed by a gap of 1550 µs. After that, there is a sequence of short pulses and long pulses of 150 µs for the "1". The signal is repeated 5 times with a delay of 50 µs between every sequence and the modulation is OOK. In summary, we have a protocol of 25 bit length like this: '0011000011100100000001000' or after decoding it : '1100111100011011111110111'.

Thanks to pigs, I enabled noise filter but without success. piscope always shows the same altered sequence.

guymcswain, 2023-09-22:
Have you tried glitch filter?  If neither of these built in filters work, you may need to implement your own custom filter with the callback associated with the gpio events.
`callback(level, tick) {
// your filter here;
}`

guymcswain, 2023-09-22:
I should add; You won’t be able to observe the affects of your filter through piscope unless you produce an output file that piscope can consume on replay.

Kitoko86, 2023-09-23:
> Have you tried glitch filter? If neither of these built in filters work, you may need to implement your own custom filter with the callback associated with the gpio events. `callback(level, tick) { // your filter here; }`

Yes, I have already tried the glitch filter with different values but I see no change in piscope. However, I found something interesting about my signal. It is so strong that the AGC reduces the gain level to 0 and piscope shows a big gap between 30-40 ms. I'll attach some screenshots later.

Maybe I need your help to implement the custom filter because I'm not enough skilled in python programming.

Kitoko86, 2023-09-25:
[image]

guymcswain, 2023-09-26:
The long gap and the long high pulse, appear to be free of 'noise' but are very different from the protocol timings you mentioned above.  A filter is not going to help you until you really understand what you are receiving.  

Also, I assume you have placed the transmitter far away to try to avoid over driving the receiver (AGC).