joan2937/pigpio #304: PWM monitor in c

MungoBerry, 2019-11-15:
Hi,

I'm using version 56 of the library on an rpi model b and I'm trying to replicate the example for python pwm monitor in c.

Using arduino I am generating a frequency of about 5.618hz (controlled using an oscilloscope) on pin 20.

The output of the python example is this, I have only changed the pin and the number of decimal digits for the frequency, from 1 to 4:

f = 5.6172 pw = 1010 dc = 0.57
f = 5.6172 pw = 1010 dc = 0.57
f = 5.6170 pw = 1010 dc = 0.57
f = 5.6172 pw = 1015 dc = 0.57
f = 5.6170 pw = 1010 dc = 0.57
f = 5.6172 pw = 1015 dc = 0.57
...

the program converted to c mi gives this output:

5.6172
5.5787
5.6208
5.6172
5.6149
5.6295
5.6175
5.6156
...

As you can see the reading is much less stable.
I tried to be as faithful as possible to the python code and I am sure that the frequency reading should be done in a much more complex way, which in python is simplified by pigpiod.
I think the solution is in the freq_count_2 example but I got lost ...

The program in c is this:

```
//
// compile with gcc -o my_test my_test.c -lpigpio -lpthread
//
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <unistd.h>
#include <stdbool.h>
#include <pigpio.h>

#define WS_PIN 20

static uint32_t high_tick;
static volatile uint32_t period;

void wsCallback (int gpio, int level, uint32_t tick)
{
    static int init;
    
    if (init)
        period = tick - high_tick;

    if (! init)
        init = 1;
    high_tick = tick;
}

int main (int argc, char * argv [])
{
    int ret;
        
    ret = 1; // defaults to err
    
    gpioCfgBufferSize (PI_DEFAULT_BUFFER_MILLIS);
    gpioCfgClock (PI_DEFAULT_CLK_MICROS, PI_DEFAULT_CLK_PERIPHERAL, 0);
    gpioCfgInterfaces (PI_DEFAULT_IF_FLAGS);
    gpioCfgDMAchannels (PI_DEFAULT_DMA_PRIMARY_CHANNEL, PI_DEFAULT_DMA_SECONDARY_CHANNEL);
    gpioCfgSocketPort (PI_DEFAULT_SOCKET_PORT);
    gpioCfgMemAlloc (PI_DEFAULT_MEM_ALLOC_MODE);

    gpioCfgSetInternals (PI_DEFAULT_CFG_INTERNALS);
    
    if (gpioInitialise () <0)
        goto quit;
    
    gpioSetMode (WS_PIN, PI_INPUT);
    gpioSetISRFunc (WS_PIN, RISING_EDGE, -1, wsCallback);
    
    while (1)
    {
        gpioSleep (PI_TIME_RELATIVE, 2, 0);
        printf ("%. 4f n", 1000000.0 / period);
    }
    
    gpioTerminate ();
    ret = 0;
    
quit:
    return ret;
}

```

Any help ?

Thank you.

guymcswain, 2019-11-15:
Some amount of jitter is introduced by the dma sampling so some variability in your readings are expected.  Have you verified that the python/pigpiod results were obtained with the same sample rate?

Frequency_counter_2 is doing some averaging which will remove the jitter.

MungoBerry, 2019-11-15:
The gpioCfg* function calls in c version code are taken from pigpiod.c, so the sample rate should be the same.
Frequency_counter_2  give as output a fixed value of 6.
I am doing a test to take the values ​​from an anemometer in order to have the minimum, average and maximum speed.
I don't know if there is another more accurate way.

joan2937, 2019-11-15:
The Python is doing quite a bit more than the C code.  I'd be tempted to add in the weighting code and just set the weighting to zero as in the Python example.  There might be some non-weighting related code you missed.
```python
class reader:
   """
   A class to read PWM pulses and calculate their frequency
   and duty cycle.  The frequency is how often the pulse
   happens per second.  The duty cycle is the percentage of
   pulse high time per cycle.
   """
   def __init__(self, pi, gpio, weighting=0.0):
      """
      Instantiate with the Pi and gpio of the PWM signal
      to monitor.

      Optionally a weighting may be specified.  This is a number
      between 0 and 1 and indicates how much the old reading
      affects the new reading.  It defaults to 0 which means
      the old reading has no effect.  This may be used to
      smooth the data.
      """
      self.pi = pi
      self.gpio = gpio

      if weighting < 0.0:
         weighting = 0.0
      elif weighting > 0.99:
         weighting = 0.99

      self._new = 1.0 - weighting # Weighting for new reading.
      self._old = weighting       # Weighting for old reading.

      self._high_tick = None
      self._period = None
      self._high = None

      pi.set_mode(gpio, pigpio.INPUT)

      self._cb = pi.callback(gpio, pigpio.EITHER_EDGE, self._cbf)

   def _cbf(self, gpio, level, tick):

      if level == 1:

         if self._high_tick is not None:
            t = pigpio.tickDiff(self._high_tick, tick)

            if self._period is not None:
               self._period = (self._old * self._period) + (self._new * t)
            else:
               self._period = t

         self._high_tick = tick

      elif level == 0:

         if self._high_tick is not None:
            t = pigpio.tickDiff(self._high_tick, tick)

            if self._high is not None:
               self._high = (self._old * self._high) + (self._new * t)
            else:
               self._high = t

   def frequency(self):
      """
      Returns the PWM frequency.
      """
      if self._period is not None:
         return 1000000.0 / self._period
      else:
         return 0.0

   def pulse_width(self):
      """
      Returns the PWM pulse width in microseconds.
      """
      if self._high is not None:
         return self._high
      else:
         return 0.0

   def duty_cycle(self):
      """
      Returns the PWM duty cycle percentage.
      """
      if self._high is not None:
         return 100.0 * self._high / self._period
      else:
         return 0.0

   def cancel(self):
      """
      Cancels the reader and releases resources.
      """
      self._cb.cancel()
```

MungoBerry, 2019-11-15:
I'm sorry but I completely miss the point.
I've refactored python code to (or what I think it should) reflect the c code I've written:
```
#!/usr/bin/env python

import time
import pigpio
    
high_tick = None
period = None

def callback(gpio, level, tick):
   global high_tick
   global period
   
   if high_tick is not None:
      period = pigpio.tickDiff(high_tick, tick)
   high_tick = tick
    
if __name__ == "__main__":

   import pigpio

   PWM_GPIO = 20
   SAMPLE_TIME = 1.0

   pi = pigpio.pi()
   pi.set_mode(PWM_GPIO, pigpio.INPUT)
   cb = pi.callback(PWM_GPIO, pigpio.RISING_EDGE, callback)

   while True:
      time.sleep(SAMPLE_TIME)
      print("{:.4f}".format(1000000.0 / period))

   cb.cancel()

   pi.stop()   
```
The output seems to be stable as your example.
Thank you for your time, I need another hint!

guymcswain, 2019-11-16:
Your c code is using `gpioSetISRfunc()`

MungoBerry, 2019-11-16:
Yes, I was aware of it. Thank you for pointing me out to right direction.
It was an honest mistake:
The python interface uses EITHER_EDGE as type for GPIO edge.
In c this value is used by gpioSetISRfunc \ gpioSetISRfuncEx.
I thought these interfaces were interchangeable...but
I've not found gpioSetISRfunc (nor gpioSetAlertFunc) calls inside pigpiod or any other python code.

For completeness this is the fixed code:
```

// compile with
//
// gcc -o my_test my_test.c -lpigpio -lpthread
//
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <unistd.h>
#include <stdbool.h>
#include <pigpio.h>

#define WS_PIN      20

static uint32_t high_tick;
static volatile uint32_t period;

void wsCallback(int gpio, int level, uint32_t tick)
{
    if ( 1 == level )
    {
        static int init;
    
        if ( init )
            period = tick - high_tick;

        if ( ! init )
            init = 1;
        high_tick = tick;
    }
}

int main(int argc, char* argv[])
{
    int ret;   
        
    ret = 1; // defaults to err
    
    if ( gpioInitialise() < 0 )
        goto quit;
    
    gpioSetMode(WS_PIN, PI_INPUT);
    gpioSetAlertFunc(WS_PIN, wsCallback);
    
    while ( 1 )
    {
        gpioSleep(PI_TIME_RELATIVE, 2, 0);
        printf("%.4f\n", 1000000.0 / period);
    }
    
    gpioTerminate();
    ret = 0;
    
quit:
    return ret;
}

```

joan2937, 2019-11-16:
@guymcswain Well spotted!

For information.

gpioSetAlertFunc reports the samples captured and time-stamped by the DMA engine.

gpioSetISRFunc reports an interrupt captured by the Linux kernel.

The alert reports will have more latency (half a millisecond say) but will be more accurately timed (+/- 5 µs say).  They will be far more accurate for timing between GPIO events.

The ISR reports will have less latency (50 µs say) but will be less accurately timed (+/- 50 µs say).  They will be far less accurate for timing between GPIO events and can not handle multiple events in close succession.

I would only use the ISR reports when you absolutely need the minimum latency.  I have not found a use for them myself.

MungoBerry, 2019-11-16:
Thank you for explanation!
I'm not very into electronics/low-hardware stuff! Just a hobbyist!
I would like to remove arduino from my setup and do more flexible test. i.e. change frequency.
It is possible to generate a frequency on a pin and read from another pin on the same rpi ?
Should I use gpioWaveCreate func ?

guymcswain, 2019-11-16:
You can generate a frequency on an output pin and read the timing on the very same pin.  Use the pwm APIs to vary frequency and dutycycle.  The APIs are well documented.

MungoBerry, 2019-11-16:
Thank you, I close the issue.

MungoBerry, 2019-11-23:
Hi,

sorry to reopen this issue but I'm not sure that my c conversion reflect the python version.
I've running for a while (2 night, 1 night per test) under same condition (no hdmi and no lan) the python script first and the c code then and I noticed that sometimes the c code give wrong results (I feed the pin with a fixed freq of ~5Hz and it reads 2.8...)
I can guess that it happens 'cause there's many stuff involved like daemons etc, but why python version is stable ?
I believe the answer lies in the implementation of pigpiod but I would like an opinion about this.

Thanks.

guymcswain, 2019-11-23:
Assuming you are using `gpioSetAlertFunc()` then you will receive a callback on either edge of the signal you are measuring.  The c callback function needs to ignore the falling edge to match your Python code.

MungoBerry, 2019-11-23:
The python script has edge setted to `RISING_EDGE`.
The c callback is

```
void wsCallback(int gpio, int level, uint32_t tick)
{
    if ( 1 == level )
    {
		// do stuff
	}
}
```

It is the right way to get only rising edge ?

guymcswain, 2019-11-24:
Seems correct to me.

guymcswain, 2019-11-24:
I notice you are using version 56.  Latest version is 70 ish?

guymcswain, 2020-01-04:
Please retest with the latest version and report back if you still believe there to be an issue.

closed 2020-01-25 by guymcswain