joan2937/pigpio #308: rawWaveAddSPI interleaving

u0078867, 2019-12-11:
Hi,

is it possible to combine SPI waveforms (`rawWaveAddSPI`) for different devices, not sharing SS, CLK, MISO, MOSI? Meaning, having one `rawSPI` structure per device.

In particular, what I need to do is using SPI etherogeneous sensors, but with different sampling rates. At some instant in time, more than one sensor should be sensing its amount, so the `offset` value for different `rawWaveAddSPI` calls will be the same.

The questions arises because, when looking at the internals of `rawWaveAddSPI`, I see **all** GPIO pins are set on or off at the same time, so perhaps overwriting the GPIO on/off pattern set by a previously called `rawWaveAddSPI` with the same `offset`.

joan2937, 2019-12-15:
I'm afraid you are on your own with this one.  I don't have the time to get back up to speed with how these functions work to be able to offer any advice.

guymcswain, 2020-01-25:
The pigpio documentation for this API states:

> Not intended for general use.

This API is also listed under the "expert" catagory. Read between the lines, this means you should consult the source code directly to understand its behaviour.

closed 2020-01-25 by guymcswain