joan2937/pigpio #309: rawWaveAddSPI doc help

u0078867, 2019-12-11:
Hi,

regarding the online example of MCP3008, I don't understand some parts of the following:

`rawWaveAddSPI(&rawSPI, offset, SPI_SS, buf, 2, BX, B0, B0);`

In particular:

1) `2`: what are 2 bits to transfer? 

2) from the MCP3008 datasheet, I see 3 bytes needs to be transmitted, while the buffer here keeps only 2 bytes
[image]

Indeed the example for [bbSPIXfer](http://abyz.me.uk/rpi/pigpio/cif.html#bbSPIXfer) moves 3 bytes.

Perhaps 1) and 2) are closely related.

Thanks!

joan2937, 2019-12-15:
I'm afraid you are on your own with this one.  I don't have the time to get back up to speed with how these functions work to be able to offer any advice.

guymcswain, 2020-01-25:
The pigpio documentation for this API states:

> Not intended for general use.

This API is also listed under the "expert" catagory.  Read between the lines, this means you should consult the source code directly to understand its behaviour.   As Joan states, 'you are on your own with this one'.

closed 2020-01-25 by guymcswain