joan2937/pigpio #310: Make install failing with Error Code 1 on Alpine

pricehatfield, 2019-12-15:
I'm trying to build this inside a Docker container (all commands as root) based on the dockerhub image arm32v6/python:3.8.0-alpine3.10 and running on a Raspberry Pi Zero.

Commands:
> apk add linux-headers build-base
> mkdir /usr/src/pigpio
> cd /usr/src/pigpio
> wget https://github.com/joan2937/pigpio/archive/master.zip
> unzip master.zip
> cd pigpio-master
> make
> make install

Results of last:
> /usr/src/pigpio/pigpio-master # make install
> install -m 0755 -d                             /opt/pigpio/cgi
> install -m 0755 -d                             /usr/local/include
> install -m 0644 pigpio.h                       /usr/local/include
> install -m 0644 pigpiod_if.h                   /usr/local/include
> install -m 0644 pigpiod_if2.h                  /usr/local/include
> install -m 0755 -d                             /usr/local/lib
> install -m 0755 libpigpio.so.1      /usr/local/lib
> install -m 0755 libpigpiod_if.so.1  /usr/local/lib
> install -m 0755 libpigpiod_if2.so.1 /usr/local/lib
> cd /usr/local/lib && ln -fs libpigpio.so.1      libpigpio.so
> cd /usr/local/lib && ln -fs libpigpiod_if.so.1  libpigpiod_if.so
> cd /usr/local/lib && ln -fs libpigpiod_if2.so.1 libpigpiod_if2.so
> install -m 0755 -d                             /usr/local/bin
> install -m 0755 pig2vcd                        /usr/local/bin
> install -m 0755 pigpiod                        /usr/local/bin
> install -m 0755 pigs                           /usr/local/bin
> if which python2; then python2 setup.py install ; fi
> if which python3; then python3 setup.py install ; fi
> /usr/local/bin/python3
> running install
> running build
> running build_py
> creating build
> creating build/lib
> copying pigpio.py -> build/lib
> running install_lib
> copying build/lib/pigpio.py -> /usr/local/lib/python3.8/site-packages
> byte-compiling /usr/local/lib/python3.8/site-packages/pigpio.py to pigpio.cpython-38.pyc
> running install_egg_info
> Removing /usr/local/lib/python3.8/site-packages/pigpio-1.45-py3.8.egg-info
> Writing /usr/local/lib/python3.8/site-packages/pigpio-1.45-py3.8.egg-info
> install -m 0755 -d                             /usr/local/man/man1
> install -m 0644 p*.1                           /usr/local/man/man1
> install -m 0755 -d                             /usr/local/man/man3
> install -m 0644 p*.3                           /usr/local/man/man3
> ldconfig
> make: *** [Makefile:112: install] Error 1

What other info would be helpful?

joan2937, 2019-12-15:
I will not be able to help with this issue as I know nothing about Dockers or Alpine.

Presumably the error is caused by line 112 of the make file.

pricehatfield, 2019-12-15:
Line 112 is simply "ldconfig" - it looks like I may have found something about that under Alpine.  Thanks for at least looking.  My real trials will begin when I can get everything installed and actually write something, haha.

closed 2019-12-15 by pricehatfield

cmisztur, 2019-12-28:
I am interested in pigpiod performance characteristics in docker vs bare metal.
And then before and after RT patch.
https://docs.docker.com/config/containers/resource_constraints/#configure-the-host-machines-kernel

zinen, 2020-06-14:
I agree, I get the same problem when compiling on alpine. I found an explanation on [stackoverflow](https://stackoverflow.com/questions/36990951/ldconfig-seems-no-functional-under-alpine-3-3). The gist of it is that alpine can't use `ldconfig`.

**Solution for me:**
Just edit Makefile and remove `idconfig` before running `make install`, eg. [see my working code here](https://github.com/zinen/docker-alpine-pigpiod/blob/fcd7619780940c906a501f1b7d9f3235e90e2858/Dockerfile#L10).

care4hair, 2021-02-08:
@pricehatfield @zinen Could anyone of you make pigpio run inside an alpine docker container. I am currently trying the same. Also stumbled over the ldconfig thingy and removed it. Now the image builds. However if I run the container on the rpi and then run an executable that should command stuff to the gpio's nothing happens. Do I have to set some special attributes with "docker run"?

zinen, 2021-02-08:
Yes i made one public available here: https://hub.docker.com/r/zinen2/alpine-pigpiod

Usage: `docker run -it -p 8888:8888 --device /dev/gpiochip0 zinen2/alpine-pigpiod`

The image is very simle and tiny in size. It is also mentioned here at the bottom under "Other Languages": http://abyz.me.uk/rpi/pigpio/