joan2937/pigpio #312: Python - Add Keepalive

dupondje, 2019-12-24:
Hi,

First of all thanks for this great project. I use this lib in Home Assistant to read GPIO output of a remote RPi.

Now I noticed that when the RPi reboots for example, the remote GPIO readings stalls.
Seems like this is a known issue, and in fact needs to be fixed at the lowest level, aka pigpio.

Now it seems like fixing this should be no more then adding keepalive to the socket, and reconnect when the socket times out:
See for example https://tewarid.github.io/2013/08/16/handling-tcp-keep-alive.html

I can eventually check to fix this in the code, but as I don't really know the code, it might be quicker to just fix this? :)

Would be a very nice stability improvement!

Thanks

guymcswain, 2019-12-24:
I have dealt with this issue but using a different client (pigpio-client).  Since V68 of pigpio, TCO keep-alive probes are enabled.  I think what you are requesting is that the python client be modified to abort and reconnect once a "dead peer" connection is detected.  This would a question for @joan2937 .  

On the other hand, I think you can solve this issue on the application side: When you open a notification (or alert) on any gpio you will receive a keep alive packet every minute the channel is open.  Your application can monitor this packet and take action like stopping and restarting the python client.

dupondje, 2019-12-24:
I noticed that keepalive is enabled on server side. But it would be nice if the python client can be modified so it reconnects on dead peer indeed.
Cause now the connection stays offline when pigpio was down/restarted/...

The issue could indeed be solved/worked around on the application side, thats correct.
But my idea was that it is preferred that the lowest level lib already has it implemented :) So everybody can enjoy it!

guymcswain, 2019-12-24:
I don't disagree but some modification to your app is still required.  Given the internal state of pigpio got cleared during a restart you want to inform your app so that you can re-initiate its state.  Imagine you were transmitting a waveform but now the wave ID is gone.  Pigpio-client does this by emitting a 'disconnected' event.

guymcswain, 2019-12-24:
Although, if it's just a crappy router that is the cause of your connection problem, it could be solved transparently in the client ;)

joan2937, 2019-12-27:
@guymcswain How does your Pigpio-client know to issue a 'disconnected' event.  I can't see an obvious parallel in the Python module.

guymcswain, 2019-12-27:
While I used the Python module as a guide when implementing pigpio-client, the socket timeout, retry, reconnect and event logic was something I came up with after a user requested the feature.  It was fairly heavy lifting to implement and then afterward I realized work is still needed on the application side.  I would say your time would be better spent on solving other issues.

dupondje, 2019-12-28:
Well I don't think it will be so hard to implement. You should be able to enable keepalive on a socket quite easily in Python. And then you just need some way to trigger a reconnect when its dead.

I tried to implement it like https://github.com/dupondje/pigpio/commit/98dad646412d4b332b0cc8c5b0976d457b0d2e3e , but its not working (yet).

Thing is that pigpio is very unreliable now, as if the daemon becomes unavailable for some reason for some minutes (or even if you just restart pigpiod), the connection is gone, and its dead until manual actions are executed.

joan2937, 2019-12-28:
I will not be making any changes for this issue.

closed 2019-12-28 by joan2937

dupondje, 2019-12-28:
But you are accepting pull requests to get it fixed? :)
Then I'll continue working on it!

guymcswain, 2020-02-10:
@dupondje , I will consider a PR if you come up with a patch that is minimally invasive and aligns with the following philosophy:

A viable solution is not as simple as 

> adding keepalive to the socket, and reconnect when the socket times out,

but rather it should inform the application upon detection and allow the application to decide what action to take - reconnect being one possibility.

Before you submit a PR, please follow the new branch strategy as stated [here](https://github.com/joan2937/pigpio/blob/master/CONTRIBUTING.md).