joan2937/pigpio #314: PIGPIO - Native typescript / javascript

yboujraf, 2020-01-02:
Dear @jdiamond ,  @srounet  , @maihde  , @gyscos  , @ipa-mdl 

There is a way to have in native nodejs typescript / javascript the library ?

Best Regards,
Youssef

srounet, 2020-01-02:
hi @yboujraf , I'm really sorry but first I don't understand what you are asking for, and I have to say that I've nothing to do with this project.

You should have a look to https://github.com/rakeshpai/pi-gpio the project is not maintained but it's pretty low level so should give you the hints to access gpios from a node application.

best regards.

yboujraf, 2020-01-02:
Dear @srounet 

Thanks for your reply.

I am using https://github.com/fivdi/pigpio excellent library in js but depending of your repository.

I followed your repository and I think you will or add a new contributor to maintain and I hope to enhance the library for raspeberry and I hope it will not affect us. that's all. If new functionalities, there are welcome ;-)

and when I have seen you are working in C and there are forked repo from the repository i asked if there is a native library for PI-GPIO.

Best wishes and best regards
Youssef

guymcswain, 2020-01-02:
> I am using https://github.com/fivdi/pigpio excellent library in js but depending of your repository.

The https://github.com/fivdi/pigpio library is dependent on **this** repository, not a fork.

> I followed your repository and I think you will or add a new contributor to maintain and I hope to enhance the library for raspeberry and I hope it will not affect us. that's all. If new functionalities, there are welcome ;-)

I would like to include native bindings of this library - such as /fivdi/pigpio - into our test matrix.  If an automated test can be run whenever we do a push/merge that would be excellent and would alleviate your concern about changes breaking downstream libraries.  I will need help to accomplish this.

yboujraf, 2020-01-02:
@guymcswain 

The owner is @fivdi from this repository.

If you need somebody to test I am here.
I have raspberry PI4 last one with 4GB and beaglebone black for tests with raspian buster.

My colleague is out of the office for 1 month, specialist for wrapper c to any language so because the company develop only in nodejs and typescript and sometime accept javascript or C.

so let me know.

Best Regards,
Youssef

fivdi, 2020-01-04:
@yboujraf 

> I have raspberry PI4 last one with 4GB and beaglebone black for tests with raspian buster.

Just so that there are no misunderstandings, pigpio is a library for the RaspberryPi and will not function on a BeagleBone Black.

@guymcswain 

>  If an automated test can be run whenever we do a push/merge that would be excellent and would alleviate your concern about changes breaking downstream libraries.

Since the release of the first version of the pigpio Node.js module in 2015 there have never been any breaking changes to the pigpio C library that resulted in the pigpio Node.js module breaking. Fortunately, the pigpio Node.js module and pigpio C library have always worked well together. @joan2937 and the other contributors, thankfully, appear to have been meticulously careful not to introduce breaking changes.

guymcswain, 2020-01-04:
Thank you @fivdi .  We intend to keep that track record.

closed 2020-01-04 by guymcswain