joan2937/pigpio #315: Are you interested in tony tables for users switching from wiringPi?

ooxi, 2020-01-06:
Since [wiringPi is deprected](http://wiringpi.com/wiringpi-deprecated/) quite a few users are looking for an alternative. Providing tony tables (before/after) for those users, not that familiar with the C programming language, might yield a lot of new users.

Would you be interested in adding such documentation? I would step forward in providing those, however this might be a strategic decision since wiringPi became deprecated _because of_ those users.

joan2937, 2020-01-07:
I'd have thought that the [bcm2835](https://www.airspayce.com/mikem/bcm2835/) library would be a closer match to wiringPi.

If you think it would be useful for pigpio then perhaps add a wiki entry (assuming its editable). 
 As long as Guy is happy I'd be happy.

guymcswain, 2020-01-07:
Happiness is achieved when contributors provide support for their own contributions :)  So a few questions for @ooxi : 

- Will you accept assignment of all future issues submitted to this repo that relate to wiringPi support?

- If a one-to-one mapping for APIs does not exist, how do you intend to deal with it?  We need to be up front with wiringPI users that existing pigpio APIs will not (ever) change and the work to add a new API is significant (core library, and clients: pigs, python and other 3rd party) - which typically means it likely won't happen.  I suppose one could write an a set of helper wrappers to do the conversion but that requires more work than just documentation.

I'm not trying to dissuade you.  Perhaps you can find support from the wiringPi community to do the required development work.  I would suggest you fork this repo, create a wiki in that fork and get some feedback from other wiringPi users by mentioning your fork.  If there is strong interest in supporting wiringPI from pigpio then we can better evaluate how to merge this support into pigpio.

guymcswain, 2020-01-07:
> perhaps add a wiki entry (assuming its editable)

The wiki is merge-able but you have to follow [this procedure](https://gist.github.com/larrybotha/10650410)

ooxi, 2020-01-07:
@guymcswain since supporting wiringPi users is a significant effort, I agree with you that adding first level support and incentive for wiringPi users to switch should not be done reckless.

I will take on the route you suggested and maybe first publish a blog post or similar. Afterwards we can consider if positioning pigpio as a target for wiringPi users is gainful at al.

closed 2020-01-07 by ooxi