joan2937/pigpio #317: Why opening dev/mem and not dev/gpiomem ?

Kehny91, 2020-01-13:
Hi

I'm using pigpio with a C++ project (using ROS2).
The thing is I can't run ROS2 as sudo but I need ROS2 to run my package, wich use pigpio.

I've tried everything to have a normal user having access to pigpio. (adding the user to the kmem and gpio group; lowering the access restriction on /dev/mem ...)
But everytime, I get the failure message when gpioInitialise call open("/dev/mem, ...)

So I was wondering why do you need to open /dev/mem instead of /dev/gpiomem.
Also, if you have a general clue to help me with my problem, please don't hesitate.

Thank you :)

joan2937, 2020-01-13:
**/dev/gpiomem** only gives access to the GPIO hardware peripheral.

pigpio also makes use of the PWM, PCM, DMA, SYST, SPI, and BSC hardware peripherals.  They can only be accessed via **/dev/mem**.

You might be able to use the Linux capabilities features rather than using root permissions.  I seem to remember I was successful in using capabilities.

I suggest you start with **man capabilities**.

Of course an alternative is to rewrite your code to use the pigpio daemon.

Kehny91, 2020-01-13:
Thank you for your quick answer.
I will keep looking then :)

It might interest some users to have an optional boolean argument "limited" in the gpioInitialize function wich would only require rights on gpiomem instead of mem (with limited functionalities of course) :)

closed 2020-01-18 by guymcswain