joan2937/pigpio #319: question: getting gpioTime() instead of gpioTicks() on callback in python?

sneakers-the-rat, 2020-01-21:
Hello! Hopefully this hasn't been answered, been using the library for awhile and searched issues but couldn't find any.

From python, Is there any way to either 
a) return the system time in a callback function instead of ticks, or
b) return ticks and system time simultaneously so that ticks can be transformed into a timestamp?

I can't see an obvious way to turn ticks into timestamps, I had sketched a solution that added a `get_system_time` function in the .py file that made two `_pigpio_command` calls to, say CMD_TIME1 and CMD_TIME2. I then added two cases to myDoCommand in pigpio.c, `CMD_TIME1` calls gpioTime which assigns the seconds and microseconds ints, and then only returns the seconds. Then `CMD_TIME2` returns the microseconds assigned when CMD_TIME1 was called.

That seems really cumbersome to me, but I'm also a complete C novice. Wondering if you had already built a way to do this or had a better idea. I am at home now and the Pi's in the lab, so can't test my hack just now, will update when I do.

joan2937, 2020-01-21:
All the GPIO events are timestamped by the hardware and that uses ticks.  The tick is read from the free running system clock which is 64-bits and has a million ticks per second.  pigpio uses the bottome 32-bits and so the ticks wrap around every 71.6 minutes.

That's unlikely to change.

You will need to get tick1, grab time.time(), get tick2.  If the difference between tick1 and tick2 is less than a reasonable delta (say a few milliseconds) you can then associate tick1 + ((tick2-tick1)/2) with time.time().

sneakers-the-rat, 2020-01-21:
Alright, fair enough. In the meantime I've hacked together something that can take paired readings of ticks and system time so ticks can be transformed easily to timestamps. Also added option to give unix timestamps/isoformatted timestamps as args rather than ticks in callback functions. mean difference between system clock readings and transformed tick values is 1.15us +/- .9us (sd) for a sample of 1000 timestamps.

really don't know what I'm doing in C, so probably a better way to do this, but it seems to work which is good enough for my purposes.

https://github.com/sneakers-the-rat/pigpio/commit/58c3b914e1ab602593c594bc127bef4e63738014

sneakers-the-rat, 2020-01-21:
PS thanks for your work on this package, really the best out there for GPIO on the pi

closed 2020-01-25 by guymcswain