joan2937/pigpio #320: Pigpiod C i/f: double handle assignment in pigpio_start()

TrevorMuraro, 2020-01-24:
Bug: when calls to `pigpio_start()` are made in rapid succession from different threads, occasionally two of them will return the same value.

Replication:
```cpp
#include <pigpiod_if2.h>
#include <thread>
#include <iostream>
#include <chrono>

using namespace std;

class pg
{
public:
    pg(int _pin)
    {
        pin = _pin;
        pi = pigpio_start(NULL, NULL);
        cout << "pg " << pi << " started" << endl;
    };
    ~pg()
    {
        cout << "pg " << pi << " stopping" << endl;
        pigpio_stop(pi);
        cout << "pg " << pi << " stopped" << endl;
    };

    void test()
    {
        int rc = set_mode(pi, pin, PI_OUTPUT);
        cout << "pg " << pi << " set_mode returned " << rc << endl;

        rc = get_mode(pi, pin);
        cout << "pg " << pi << " get_mode returned " << rc << endl;
    };

private:
    int pi;
    int pin;
};

int f1()
{
    pg to(8);

    this_thread::sleep_for(chrono::milliseconds(490));

    to.test();
}

int f2()
{
    pg to(9);

    this_thread::sleep_for(chrono::milliseconds(490));

    to.test();
}

int f3()
{
    pg to(7);

    this_thread::sleep_for(chrono::milliseconds(490));

    to.test();
}

int f4()
{
    pg to(10);

    this_thread::sleep_for(chrono::milliseconds(592));

    to.test();
}

int f5()
{
    pg to(11);

    this_thread::sleep_for(chrono::milliseconds(592));

    to.test();
}

int main()
{
    thread t1(f1);
    thread t2(f2);
    thread t3(f3);
    thread t4(f4);
    thread t5(f5);

    t1.join();
    t2.join();
    t3.join();
    t4.join();
    t5.join();
}
```

Most of the time, you will get a result like this:
```
pg 2 started
pg 0 started
pg 4 started
pg 1 started
pg 3 started
pg 4 set_mode returned 0
pg 0 set_mode returned 0
pg 4 get_mode returned 1
pg 4 stopping
pg 0 get_mode returned 1
pg 0 stopping
pg 0 stopped
pg 4 stopped
pg 3 set_mode returned 0
pg 3 get_mode returned 1
pg 3 stopping
pg 3 stopped
pg 2 set_mode returned 0
pg 2 get_mode returned 1
pg 2 stopping
pg 2 stopped
pg 1 set_mode returned 0
pg 1 get_mode returned 1
pg 1 stopping
pg 1 stopped
```

Occasionally, however, you will get a result like this:
```
pg 0 started
pg 3 started
pg 2 started
pg 0 started
pg 1 started
pg 3 set_mode returned 0
pg 3 get_mode returned 1
pg 3 stopping
pg 2 set_mode returned 0
pg 2 get_mode returned 1
pg 2 stopping
pg 0 set_mode returned 0
pg 2 stopped
pg 3 stopped
pg 0 get_mode returned 1
pg 0 stopping
pg 0 stopped
pg 0 set_mode returned -2011
pg 0 get_mode returned -2011
pg 0 stopping
pg 0 stopped
pg 1 set_mode returned 0
pg 1 get_mode returned 1
pg 1 stopping
pg 1 stopped
```

The workaround is presumably to mutex-protect all calls to `pigpio_start()`, although I haven't actually done that yet.

Compiled with g++ 8.3.0 with std=c++17 and run on a Raspberry Pi Zero.

guymcswain, 2020-01-24:
What version of the library are you running?
The gpios used have SPI ALT functions.  I assume it replicates on any set of pins.

You may want to run with debug messages turned on pigpiod.  You can try `pigs csi 4` then run your script.

joan2937, 2020-01-24:
As you have noticed pigpio provides no protection.

I would be suspicious of the need to call that function from more than one thread.  I never expected such a usage.  I have never found any need for my applications.

TrevorMuraro, 2020-01-24:
@guymcswain:
V71 (from apt) and V73.

It replicates on any set of pins, and I originally discovered it in an application that communicates with multiple SPI and I2C devices (one thread per device, threads spawn at start and last the lifetime of the application.)  The specific GPIO calls seem like they'd be irrelevant, they're just there to demonstrate the sequence of "multiple threads get the same handle -> one thread closes its handle -> other thread gets `pigif_unconnected_pi` on every subsequent call".

Running after `pigs csi 4` didn't produce any initial output.  Do I need to set it before each script run?

@joan2937:
Can I just get one handle and use it in multiple threads?  I couldn't find any solid statement on pigpio's thread safety, so I figured I'd play it safe by creating one pigpiod connection for each thread.  Is pigpiod designed to have one `pigpio_start()` and one `pigpio_stop()` per process?

joan2937, 2020-01-24:
@TrevorMuraro  All the pigpiod_if2 socket commands are atomic as they are protected by a mutex.  You should be able to safely use a single connection provided by pigpio_start() for commands from multiple threads.

TrevorMuraro, 2020-01-24:
@joan2937 Great to know, thanks!

closed 2020-01-24 by TrevorMuraro

guymcswain, 2020-01-24:
@TrevorMuraro ,  glad you found a solution.  For future reference to get debug output you can issue
`pigs csi 4` (User level)
then `cat /dev/pigerr` to observe messages.