joan2937/pigpio #322: Release naming scheme

github-k8n, 2020-01-27:
Hi,

in order to create the ebuilds for gentoo it would be great to have a consistent naming scheme.
Currently there are 73 and 7301.
Due to the way the version numbers are evaluated in the ebuild this would mean that a future 74 would be considered older than the 7301.
Do you know if this is just a one-off?
Will v74 be released soon? (in that case I would just ignore that version..)
Or would it be possible to release it as something like "73.01" (including the dot to allow for the versioning mechanism of gentoo to correctly identify which version is more recent?)

Thanks,
Daniel

joan2937, 2020-01-27:
The general releases are the two digit numbers.  The four digit numbers are really intended for internal development.

I think Guy intends to release version 74 within a month.

guymcswain, 2020-01-27:
Yes, the four digit version means it came from the 'develop' branch.  Therefore, if you only pull from master it will (currently) be two digits and monotonically increasing over time.  (Until we reach v7xxx and I'm dead.)

V74 is planned to release the first week of February.

guymcswain, 2020-01-27:
If a 7xxx version did get into master (I haven't checked to see if this is the case) then it definitely is a one-off to be ignored.

closed 2020-02-03 by guymcswain