joan2937/pigpio #324: GPCLKx operation on Raspberry Pi 4B

dennowiggle, 2020-02-05:
When I use this gpioHardwareClock() in a C program on Raspberry Pi 4B, or the equivalent with pigs and I see a garbage waveform on my scope.
gpioHardwareClock(4, 16000000);

GPCLK2 is the same. If I bring the frequency down to say 500000 then it is okay.

The PI2B that I have works fine - nice waveform on the scope, and an exact frequency on my frequency counter. I turned on debug and this is what is printed in Pi 4B vs Pi2B.

PI4B
2020-02-04 21:00:46 gpioHardwareClock: gpio=4 frequency=16000000
2020-02-04 21:00:46 initHWClk: ctl=28 div=29 src=6 /I=46 /f=3584 M=1
2020-02-04 21:00:46 gpioHardwareClock: cf=750000000 div=46 frac=3584 mash=1
       status = 0 :
       
PI2B
Set GPIO4 as a 16MHz clock :
2020-02-04 21:05:42 gpioHardwareClock: gpio=4 frequency=16000000
2020-02-04 21:05:42 initHWClk: ctl=28 div=29 src=6 /I=31 /f=1024 M=1
2020-02-04 21:05:42 gpioHardwareClock: cf=500000000 div=31 frac=1024 mash=1

Above 6MHz issues start. e.g. At 16Mhz it sets the right frequency but sends a repeated burst of 8 pulses at 16MHz. It's almost as if at higher frequencies the pinmux combines PWM and CLK functions.

From pigpio via email:

The settings look correct for both.

The Pi4B has a 750MHz clock.

Divide by 16MHz gives you 46.875.  That gives an INT of 46 and a FRAC
of 0.875 * 4096 or 3584.

The Pi4B has undocumented differences.  Until we get the specs (were
expected before the New Year) there is little point in any
investigation.

I suggest you raise a github issue at
https://github.com/joan2937/pigpio/issues

dennowiggle, 2020-02-05:
When I build and run memtester after the GPIO4 GPCLK0 setting has been applied I get a perfect stable 16MHz on my oscilloscope and frequency counter. As soon as memtester stops I get the distorted output of 8x16MHz pulses in a burst. Perhaps another clue? Could it be interrupt related?
Using memtester download from :
http://pyropus.ca/software/memtester/

guymcswain, 2020-02-05:
For completeness:
* What is your hardware version?  `gpioHardwareRevision()`
* What is your pigpio version?  `gpioVersion()`
* What is your OS?  `cat /etc/os-release`
* Kernel?  `cat /proc/version`
* CPU?  `cat /proc/cpuinfo`

dennowiggle, 2020-02-05:
Interestingly "cat /proc/cpuinfo" shows BCM2835????

Here is the full output :

* What is your hardware version? gpioHardwareRevision()

gpioHardwareRevision() = 12595473

* What is your pigpio version? gpioVersion()

gpioVersion() = 74

* What is your OS? cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
    
* Kernel? cat /proc/version

Linux version 4.19.93-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1290 SMP Fri Jan 10 16:45:11 GMT 2020

* CPU? cat /proc/cpuinfo

processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : c03111
Serial          : 1000000081eb7c98
Model           : Raspberry Pi 4 Model B Rev 1.1

guymcswain, 2020-02-05:
> Interestingly "cat /proc/cpuinfo" shows BCM2835?

Is a 'red herring' - why we recently had to change the method of looking up hardware revision.

> gpioHardwareRevision() = 12595473 = 0x00c03111

Shows you have: 4GB memory, made by Sony UK, BCM2711, model 4B, rev 1

guymcswain, 2020-04-19:
@joan2937 , I noticed the BCM2711 document was available on raspberrypi.org.  Does it provide any insight related to this issue?

dennowiggle, 2020-04-19:
Link to Peripheral guide
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf

@joan2937 On page 105 bit 5 KILL has this statement
This is intended for test/debug only. Using this control may cause a glitch on the clock generator output. 

In function initHWClk() the kill bit is instead. Should this be changed to use ENAB = 0?

guymcswain, 2020-04-20:
@dennowiggle , interesting find.  If the potential fix is a simple bit flip, can you give it a quick test?  If not, I have a RPi4B that I can repurpose and an oscilloscope.  Can you send me your code snippet that demonstrates the issue?

dennowiggle, 2020-04-20:
I ran my old test code (no changes) and the 16MHz clock now appears as it should do. I can't see glitches on the scope and the frequency counter shows 15.999990MHz. I did NOT change initHWClk() code as suggested above.

I powered down and back up and ran pigs and that too worked (it did not before)
pigs hc 4 16000000

To double check I found my backup SD card that has a boot image bcm2711-rpi-4-b.dtb dated 01/20/2020. I loaded this and ran the pigs hc command again. The waveform was as originally reported on this issue - 8 clock pulses at a time.

Both SD card images used Pigpio version 74.

I then put the newer SD card back in but with bcm2711-rpi-4-b.dtb dated 01/20/2020 instead of the 03/05/2020 dated one. The waveform was good. So it does not seem to be the boot image.

With the newer SD card image I see the Linux version is newer

cat /proc/version
Linux version 4.19.97-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1294 SMP Thu Jan 30 13:21:14 GMT 2020

While I have the two SD card images to play with any suggestions to root cause this issue? I don't know how to check the Raspberry PI image changelog to look for any clock issues. Given that the original description of this problem showed that if you loaded the CPU the problem went away, looking at changelogs may be fruitless.

Anyway good news is, the issue on Pi4B version is no longer there.

guymcswain, 2020-04-20:
> While I have the two SD card images to play with any suggestions to root cause this issue?

If it's a one off combination of system software components unlikely to be run across again, then I wouldn't bother with it. But I'll leave the issue open for now.

joan2937, 2020-04-20:
@guymcswain I haven't found anything relevant to this problem in the new BCM2711 document.  It seems to repeat the detail from the earlier spec.

Re the KILL/ENA bits.  I seem to remember that if you followed the documentation (and didn't use kill) the process would sometimes hang waiting for the ena bit to clear - it never cleared.

guymcswain, 2020-05-01:
@dennowiggle , please reopen if you believe an issue exists in the pigpio implementation.

closed 2020-05-01 by guymcswain