joan2937/pigpio #325: connection monitor, failsafe

Toeedev, 2020-02-10:
Is there any way to monitor connection to remote GPIO ? 
Currently, when loosing connection (after connection was established by pigpio.pi() ),
pi.write() hangs forever, until connection is established again.
Is there any way to create some timeout ? 
Is there a way to close connection pigpio.pi() ?  so it can be retested again for connection?

Also is there a way to check if there is device connected  to local device via pigpio ? 

My Idea is, I need to detect if I lost remote connection to other PI, or other device lost connection to me.  But it seems like it's not possible with pigpio.

guymcswain, 2020-02-10:
Are you using pigpiod C I/F?

guymcswain, 2020-02-10:
... and how do you invoke the deamon?

Toeedev, 2020-02-10:
yes, on remote device I am using pigpiod 
I only did that on remote device:
`systemctl start pigpiod`
if that is what's what you mean.

guymcswain, 2020-02-10:
> Currently, when loosing connection (after connection was established by pigpio.pi() ),
pi.write() hangs forever, until connection is established again.

Where are you running this from?

guymcswain, 2020-02-10:
Are you running the python client?

Toeedev, 2020-02-10:
yes, sorry I didn't specify. 
I have remote rasbperry pi with pigpiod ( no script yet). 
And I have another raspberry pi which is my master device. And here I have python script.
Remote works fine and no problem here, however I wanted to apply some failsafe, so if for some reason remote rpi will loose connection to network, whole script won't just hang and wait for remote rpi to comeback. 

I've made test script:
```python
from time import sleep
import pigpio

try:
    PI = pigpio.pi('raspislave')
except Exception:
    print('Connection to PI failed')
else:
    isConnected = True

while (True):
    try:
        PI.write(17,1)
        sleep(1)
        PI.write(17,0)
    except:
        print('Error sending')
        sleep(0.1)
    else:
        sleep(2)
```

So let's say I got initial connection,  but wile in the while loop, remote disconnects. 
Pi.write will hand until raspislave will be back in network, which might be never. 

I was thinking to disconnect  after each loop, and test for pigpio.pi() again on the beginning of the loop. But I don't know how to disconnect from pigpio.pi() 

In the future I would like to know on raspislave, that my master is connected or not, if not it will run some local script. 

Am I missing something obvious?

guymcswain, 2020-02-10:
Thanks for the fuller description.   No the python module doesn't have a mechanism for timing out on idle socket and either attempting to reconnect or alerting the application.  You are not the first to ask for this functionality - see issue #312.

uneasymusic, 2020-02-11:
That's seems like something essential for that kind of library. I will have a look on pigpio-client then, as I feel more comfortable with nodejs

closed 2020-02-23 by guymcswain