joan2937/pigpio #330: Pigpio i2c broadcast from master to all slaves

TregliaRaphael, 2020-03-07:
Hi,
I'm currently working with i2c and i'm using bscxfer for being able to use multiples slave.
The fact is that i need to send a instruction to all slave in same time at one given time.
My master is controlling all the slaves to make them takes photo in "real time" (synchronized as fast as i can).
So i need broadcast to be able to trigger photo to all slave in the same time.

The thing is that i try to talk to 0x00 from my master to slaves but slaves do not receive anything.
Maybe that broadcast is on another addr?
Maybe that bscxfer do not handle it?
I'll just like to know if broadcast is handled because if it is not, i wont be able to use bscxfer anymore :)

Thanks in advance
Raphael

joan2937, 2020-03-08:
I2C does not support a broadcast address.  Perhaps there is a confusion with the use of the general call address to respond to slave generated interrupts.

TregliaRaphael, 2020-03-08:
Ok thanks then did you know how can i do a general call to all slaves from the master with pigpio?
Or did you know another way to process?

I appreciate your help

guymcswain, 2020-03-09:
If the code for master and slaves is all under your control, could you just send a command to each slave telling them to listen at a particular "broadcast" address for the synchronization signals.  Slaves would continue to receive sync signals until told to go back and listen at their assigned address.

How tight is your timing requirement?  I would expect there could up several milliseconds of skew between the slaves due to the OS dependent latencies in polling and responding to I/O.

TregliaRaphael, 2020-03-09:
I'm using bscxfer for slaves.
I'm trying to be as fast a possible.
A send/receive from master to slaves takes me 2ms et i can't be slower really.
 My slaves number is not fixed. I could get 2 slaves like 10 or 16 so i want to be sure that trigger a photo will take same time every time.

My sync time between cam is 500micro second and faster i can send instructions to slaves, more i'll be able to take photo.
For now, i'm just trying to take a photo every second but i dont want to be bottleneck by my communication solution.

Instead, i prefer to add a gpio wire to trigger all photo in same time.

joan2937, 2020-03-09:
@guymcswain The suggestion of using a common call address (not 0) e.g. 0x53 for all slaves makes sense.  They could change to that address prior to expecting the broadcast and return to their individual address afterwards.  Don't request a read of the devices when they are in common address mode (or make sure each slaves transmit buffer is empty).

You will only be able to synchronise to a millisecond or so since the slaves will not be signalled about the common call address message at the same time.

To get better sync use a GPIO or send a message saying take a photo at an absolute time in the future (the slaves could busy spin waiting for that time).

TregliaRaphael, 2020-03-09:
Ok i get it.
To be sure, for example:
    - to start all slave have 0x53 addr (for example),
    - when they receive the order to get photo, they Switch to their own addr to be able to ask them if they take the picture and if they not,
    - after that, they go back to 0x53  and the loop restart.

Did you know if there is a way in bscxfer fir slave to send data without waiting master info?

joan2937, 2020-03-09:
The Pi's BSC I2C slave peripheral has no way of sending data unless requested by the master.

TregliaRaphael, 2020-03-10:
Ok so i tried your solution. I'm working on pi 0 w.
Since now, i just test it with one master and one slave.
I just tried to add one slaves just plugging my pi on i2c bus and with same ground and now, the master write always fail.
When i plugged out the second slave, it works again. Idk why but i can't even add a slave for now.
When it ill be fixed, i'll come back to you.
If you have any idea of the problem, i'll take that

joan2937, 2020-03-10:
Sorry I do not understand.

There is nothing to be fixed.  You have not shown any problem with the way pigpio is working.

TregliaRaphael, 2020-03-11:
Here is my master's code 
```
#include "master.hpp"

long i2c_test(unsigned sl, int to_send, char to_receive){
  int state = 0, data = 0;
  long time = 100000;
  long unit = 100000;
  char buf[2] = "0";
  state = i2cWriteByte(sl, to_send);
  utils::my_nano_sleep(2000000);
  if (state >= 0){
    std::cout << "SENDED" << std::endl;
    return 2;
  }
  std::cout << "FAIL" << std::endl;
  return 3;
}

void start_master(RPiGPIOPin gpio_led)
{
  gpioInitialise();
  unsigned sl = i2cOpen(1, 0x10, 0);
  
  long tot = 0;
  
  //test receive A
  utils::ctrl_act_led(500, 100);
  tot += i2c_test(sl, 0x00, 'A');
  //test receive B
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x01, 'B');
  //test receive C
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x02, 'C');
  //test receive D
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x03, 'D');
  //test receive E
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x10, 'E');
  //test receive F
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x11, 'F');

  //test receive A
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x00, 'A');
  //test receive B
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x01, 'B');
  //test receive C
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x03, 'D');
  //test receive E
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x10, 'E');
  //test receive F
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x11, 'F');

  //test receive A
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x00, 'A');
  //test receive B
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x01, 'B');
  //test receive C
  utils::ctrl_act_led(500,100);
  tot += i2c_test(sl, 0x02, 'C');
[... 148 lines cut ...]
        this->xfer->txBuf[0] = 'E';
        this->xfer->txBuf[1] = '\0';
        this->xfer->txCnt = 1;
        break;

        case 0x11:
        this->xfer->txBuf[0] = 'F';
        this->xfer->txBuf[1] = '\0';
        this->xfer->txCnt = 1;
        break;

        
        default:
        break;
      }
      std::memset(this->xfer->rxBuf, '\0', BSC_FIFO_SIZE);
      */
      utils::ctrl_act_led(1,0);
    }
    bscXfer(this->xfer);
  }
}
```

I'm getting error when i'm adding second slave and i'm selecting addresses of slaves with pull up.

TregliaRaphael, 2020-03-11:
i'm back... it was just a burned pi 0.... your solution works you can close the subject :)
thanks a lot

guymcswain, 2020-03-11:
I'll leave open for a day to two before closing.  Your application is interesting.  It may be worthy of mentioning it in the repo's wiki (if I ever get the time).

TregliaRaphael, 2020-03-11:
good, if you need any help, i'll work more on it. I'm on a real time os and it still not optimized so i'm very confident for the future.

joan2937, 2020-03-11:
@TregliaRaphael That's good.  Best wishes for your project.

TregliaRaphael, 2020-03-11:
Thanks, i'm now under half a micro second for a send receive!

closed 2020-03-22 by guymcswain

TregliaRaphael, 2020-03-31:
Hi, i'm back
Afterall, i'm trying to change i2c adresse of the slave dynamically but it is not working. I need to unplug replug master to make it works.
Here is the slaves code
```
void I2CSlave::slaveWorking(){
  uint8_t order = 0x00;
  bool last_state = false;
  bool be_broadcast = false;

  while(1){
    if(this->xfer->rxCnt > 0) {
      //HERE YOU WRITE I2C COMMUNICATION CODE
      std::bitset<8> rx(this->xfer->rxBuf[0]);
      rx[7] && !rx[6] && !rx[5] && !rx[4] ? order = TAKE_PIC: order = this->xfer->rxBuf[0];

      switch(order)
      {
        case DETECT_SLAVE:
          this->xfer->txBuf[0] = DETECT_SUCCESS;
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1;
          be_broadcast = true;
          break;

        case TAKE_PIC:
          //take pic, can't answer => broadcast
          if (!rx[1] && !rx[0]){
            //take pic no settings changed
            this->xfer->txBuf[0] = TAKE_PIC;
          }
          else if (!rx[1] && rx[0]){
            //take pic settings1
            this->xfer->txBuf[0] = TAKE_PIC_SET_1;
          }
          else if (rx[1] && !rx[0]){
            //take pic settings2
            this->xfer->txBuf[0] = TAKE_PIC_SET_2;
          }
          else if (rx[1] && rx[0]){
            //take pic settings3
            this->xfer->txBuf[0] = TAKE_PIC_SET_3;
          }
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1; 
          be_broadcast = false;
          break;

        case PIC_ASK:
          this->xfer->txBuf[0] = PIC_SUCCESS; //Or PIC_ERROR
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1;
          be_broadcast = true;
          break;
        
        default:
          break;
      }
      std::memset(this->xfer->rxBuf, '\0', BSC_FIFO_SIZE);
    }
    bscXfer(this->xfer);
    /*if (be_broadcast == true && last_state == false){
      this->xfer->control = getControlBits(this->slave_address, false);
      bscXfer(this->xfer);
      this->xfer->control = getControlBits(this->broadcast_address, true);
      last_state = be_broadcast;
      bscXfer(this->xfer);
    }
    else if (!be_broadcast && last_state){
      this->xfer->control = getControlBits(this->broadcast_address, false);
      bscXfer(this->xfer);
      this->xfer->control = getControlBits(this->slave_address, true);
      last_state = be_broadcast;
      bscXfer(this->xfer);
    }*/
  }
}
```

With my master, i'm first asking to my slave if he's existing or not.
When he answer me, i want him to go in another addresse but it is not working (the code that is doing it is quoted at the end of the switch)
I need to unplug replug to do it.
So for now, i'll to reopen the subject because it is not working for me right now.

Thank you