joan2937/pigpio #334: Cross Compiling Problems - Visual Studio + Raspberry Pi

AdriiDev, 2020-03-24:
Hello!

I have been reading about many libs and i think this one fits really well to my future projects.

I have been trying to compile it with CMake to create a Visual Studio (VS) project, but I failed.

I would like to know if someone could help me to compile it in Windows so I can start using the lib api.

* Expected environment:
    * Devices and programs/OS:
        * Visual Studio Community 2019 in a Windows 10 machine.
        * Raspbian in a Raspberry Pi
    * Communications:
        * Same LAN net
        * SSH communication

* Desired Workflow:
    1. Get libraries to link to the project (2 possible options):
        * Compile in CMake to build VisualStudio Project and compile libs 
        * Get libraries from remote device (RPi) where are installed
      The problem is that I can't link libraries in VS so I can't solve the externs included.
    2. With libraries, programm in VS IDE and debug in RPi.
    3. Compile and copy to RPi, to execute the code.

* Adittional questions:
    1. I have read that there are 3 methods to use the api:
        * C directly from api
        * C through daemon (desired)
        * Python through daemon
      I would like to use C and second option to be able to check the output with piscope. If is there an easier option with first option i can dispense with piscope, because I could get a better performance.

PD:
    I don't know if there was an open/closed issue related with this but i have read so many issues and have not get a solution.
    If there is no solution, I would have to programm in RPi (unless someone knows another option), but i would like to give it a try.

Thank You! And have a nice day!

guymcswain, 2020-03-24:
> I have been trying to compile it with CMake to create a Visual Studio (VS) project, but I failed.

You'll need to be much more specific.  CMake support in pigpio was contributed from the community so the lib's author probably can not help you.  The latest release, 75, had significant updates in this area - see PR #290.  Which version are you running?

AdriiDev, 2020-03-24:
I think I have solved it. Gonna post the tutorial in next days to help everyone who need it if it is correctly solved

guymcswain, 2020-05-01:
I hope you have solved your issue.  If not, and you believe there is a problem with this implementation then feel free to reopen.

closed 2020-05-01 by guymcswain

engineertrooper, 2023-05-17:
> I think I have solved it. Gonna post the tutorial in next days to help everyone who need it if it is correctly solved

@AdriiDev Any new information? Where I can find the tutorial?
Thank you.