joan2937/pigpio #335: [I2C][BSCXfer] Can't swap address of slaves without needed to reboot

TregliaRaphael, 2020-04-08:
Hi, i'm back
Afterall, i'm trying to change i2c adresse of the slave dynamically but it is not working. I need to unplug replug master to make it works.
Here is the slaves code
```
void I2CSlave::slaveWorking(){
  uint8_t order = 0x00;
  bool last_state = false;
  bool be_broadcast = false;

  while(1){
    if(this->xfer->rxCnt > 0) {
      //HERE YOU WRITE I2C COMMUNICATION CODE
      std::bitset<8> rx(this->xfer->rxBuf[0]);
      rx[7] && !rx[6] && !rx[5] && !rx[4] ? order = TAKE_PIC: order = this->xfer->rxBuf[0];

      switch(order)
      {
        case DETECT_SLAVE:
          this->xfer->txBuf[0] = DETECT_SUCCESS;
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1;
          be_broadcast = true;
          break;

        case TAKE_PIC:
          //take pic, can't answer => broadcast
          if (!rx[1] && !rx[0]){
            //take pic no settings changed
            this->xfer->txBuf[0] = TAKE_PIC;
          }
          else if (!rx[1] && rx[0]){
            //take pic settings1
            this->xfer->txBuf[0] = TAKE_PIC_SET_1;
          }
          else if (rx[1] && !rx[0]){
            //take pic settings2
            this->xfer->txBuf[0] = TAKE_PIC_SET_2;
          }
          else if (rx[1] && rx[0]){
            //take pic settings3
            this->xfer->txBuf[0] = TAKE_PIC_SET_3;
          }
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1; 
          be_broadcast = false;
          break;

        case PIC_ASK:
          this->xfer->txBuf[0] = PIC_SUCCESS; //Or PIC_ERROR
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1;
          be_broadcast = true;
          break;
        
        default:
          break;
      }
      std::memset(this->xfer->rxBuf, '\0', BSC_FIFO_SIZE);
    }
    bscXfer(this->xfer);
    /*if (be_broadcast == true && last_state == false){
      this->xfer->control = getControlBits(this->slave_address, false);
      bscXfer(this->xfer);
      this->xfer->control = getControlBits(this->broadcast_address, true);
      last_state = be_broadcast;
      bscXfer(this->xfer);
    }
    else if (!be_broadcast && last_state){
      this->xfer->control = getControlBits(this->broadcast_address, false);
      bscXfer(this->xfer);
      this->xfer->control = getControlBits(this->slave_address, true);
      last_state = be_broadcast;
      bscXfer(this->xfer);
    }*/
  }
}
```

With my master, i'm first asking to my slave if he's existing or not.
When he answer me, i want him to go in another addresse to simulate broadcast address but it is not working (the code that is doing it is quoted at the end of the switch)
I need to unplug replug to do it.

Thank you

guymcswain, 2020-04-10:
If, as you say, a reboot is required to change the address, this suggests the control register bits may need to be reset.  Have you tried:
* Set/clear of xfer.control bit 7 (BK,  abort operation and clear FIFOs)
* Clear all of xfer.control, then re-enable.
  `xfer.control = 0;`
  `xfer.control = address (slave-addr << 16 ) | 0x305;`

TregliaRaphael, 2020-04-11:
Between each changes of xfer.control i need to bscxfer right?
I'll test it this weekend

TregliaRaphael, 2020-04-13:
Here is my getcontrolbit fonction between quotes, wheb putting false to it, it is putting BK to the control parameter:
```
int I2CSlave::getControlBits(int address /* max 127 */, bool open) {
    /*
    Excerpt from http://abyz.me.uk/rpi/pigpio/cif.html#bscXfer regarding the control bits:
    22 21 20 19 18 17 16 15 14 13 12 11 10 09 08 07 06 05 04 03 02 01 00
    a  a  a  a  a  a  a  -  -  IT HC TF IR RE TE BK EC ES PL PH I2 SP EN
    Bits 0-13 are copied unchanged to the BSC CR register. See pages 163-165 of the Broadcom 
    peripherals document for full details. 
    aaaaaaa defines the I2C slave address (only relevant in I2C mode)
    IT  invert transmit status flags
    HC  enable host control
    TF  enable test FIFO
    IR  invert receive status flags
    RE  enable receive
    TE  enable transmit
    BK  abort operation and clear FIFOs
    EC  send control register as first I2C byte
    ES  send status register as first I2C byte
    PL  set SPI polarity high
    PH  set SPI phase high
    I2  enable I2C mode
    SP  enable SPI mode
    EN  enable BSC peripheral
    */

    // Flags like this: 0b/*IT:*/0/*HC:*/0/*TF:*/0/*IR:*/0/*RE:*/0/*TE:*/0/*BK:*/0/*EC:*/0/*ES:*/0/*PL:*/0/*PH:*/0/*I2:*/0/*SP:*/0/*EN:*/0;

    int flags;
    if(open)
        flags = /*RE:*/ (1 << 9) | /*TE:*/ (1 << 8) | /*I2:*/ (1 << 2) | /*EN:*/ (1 << 0);
    else // Close/Abort
        flags = /*BK:*/ (1 << 7) | /*I2:*/ (0 << 2) | /*EN:*/ (0 << 0);

    return (address << 16 /*= to the start of significant bits*/) | flags;
}
```
here is what i tested instead of the quoted part in first comment:
```
/*if (be_broadcast == true && last_state == false){
      //YOUR FIRST POINT
      this->xfer->control = getControlBits(this->slave_address, false);
      bscXfer(this->xfer);

      //YOUR SECOND POINT
      this->xfer->control = 0;
      bscXfer(this->xfer);
      this->xfer->control = getControlBits(this->broadcast_address, true);
      last_state = be_broadcast;
      bscXfer(this->xfer);
      //END SECOND POINTS
    }
    else if (!be_broadcast && last_state){
      //YOUR FIRST POINT
      this->xfer->control = getControlBits(this->broadcast_address, false);
      bscXfer(this->xfer);

      //YOUR SECOND POINT
      this->xfer->control = 0;
      bscXfer(this->xfer);
      this->xfer->control = getControlBits(this->slave_address, true);
      last_state = be_broadcast;
      bscXfer(this->xfer);
      //END SECOND POINTS
    }*/

```

Then, what you said doesn't change anything to me :/
Your first points is done with my getcontrolbits and the second point is done here with
`this->xfer->control = getControlBits(this->slave_address, true);`
equal to
`xfer.control = (this->slave_address << 16 ) | 0x305;`
 and still need to unplug replug to make it changed :(
Did someone have any example of a swap address of a slave that is working?

TregliaRaphael, 2020-04-20:
Here is my way to bypass lack of broadcast with one more wire!
```
void I2CSlave::slaveWorking(){
  uint8_t order = 0x00;
  int pic_before = 0;
  int pic_last = 0;
  int cpt = 0;

  bool front_montant = false;
  bool wait_pic_order = false;
  bool pic_ok = false;

  while(1){
    //DO YOUR TAKE_PIC THINGS HERE
    if (wait_pic_order && front_montant){
      if (cpt == 0){
        pic_ok = true;
        cpt = 1;
      }
      else {
        pic_ok = false;
        cpt = 0;
      }
      wait_pic_order = false;
      front_montant = false;
    }
    
    //I2C RESPONSES
    if(this->xfer->rxCnt > 0 && !wait_pic_order){
      switch(this->xfer->rxBuf[0])
      {
        case DETECT_SLAVE:
          this->xfer->txBuf[0] = DETECT_SUCCESS;
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1;
          wait_pic_order = true;
          break;

        case PIC_ASK:
          //i2c response if picture is taken successfully
          if (pic_ok == true)
            this->xfer->txBuf[0] = PIC_SUCCESS;
          else
            this->xfer->txBuf[0] = PIC_ERROR;
          wait_pic_order = true;
          this->xfer->txBuf[1] = '\0';
          this->xfer->txCnt = 1;
          break;
        
        default:
          break;
      }
      std::memset(this->xfer->rxBuf, '\0', BSC_FIFO_SIZE);
    }
    else {
      //if master's message not used, set it to 0 or your answer will be always the same....
      // not written on the doc
      this->xfer->txBuf[0] = '\0';
      this->xfer->txCnt = 0;
      std::memset(this->xfer->rxBuf, '\0', BSC_FIFO_SIZE);
    }

    //reload slave's state
    bscXfer(this->xfer);

    // if slaves is waiting picture trigger and that he still not detect the take picture order
    if (wait_pic_order && !front_montant) {
      pic_before = pic_last;
      pic_last = gpioRead(GPIO_TAKE_PIC);
      //if state of gpio change upper => front montant detected => lets take picture
      if (pic_last == 1 && pic_before == 0)
        front_montant = true;
    }
  }
}

```

guymcswain, 2020-05-01:
I don't believe this is an issue with pigpio - no documented feature or API is shown to be at fault.  The suggestion to use a 'broadcast' slave address was a suggested workaround to you original predicament but unfortunately it did not work.  However, it now appears you have a working solution.  Thanks for sharing!

closed 2020-05-01 by guymcswain

TregliaRaphael, 2020-05-01:
Ok... What about 
https://www.i2c-bus.org/specification/
I'm letting you take look at **3.1.13 General call address p17** 
You should check out what you said and reconsidered the fact that it is not coming from pigpio.
Maybe you wasn't aware of it but i don't think so.
I'm here to help you find how to resolve issues of the lib and you are just saying that there is nothing wrong with your lib but that it was mine...

Then to answer to your last message, i believe that the main problem is coming from your lib that should own the broadcast feature.
As you says fews month ago about broadcast was that i2c broadcast doesn't exist.
Here is the link :
https://github.com/joan2937/pigpio/issues/330#issuecomment-596240716