joan2937/pigpio #337: Audio not working even after pigpio script ends

maheshhchandra, 2020-04-11:
Hi any way to release PCM and PWM for audio
or restating audio after sending
gpioWaveTxSend();
audio not working even after
gpioTerminate();

documentation says pigpio will use both peripherals(PWM and PCM) if waves or the hardware PWM function is used. so audio will not work with it

I don't wan't to use audio along with pigpio waves
but audio after the pigpio c has terminated
on Raspberry pi 4

don't want to add usb peripheral for audio

Thanks

joan2937, 2020-04-11:
I suggest you make a note of the modes of the affected GPIO before pigpio is started.  Perhaps the gpio utility will do that for you.  Set the GPIO back to the previous modes after pigpio has finished.  Again perhaps the gpio utility will do that.

If the gpio utility does not provide that feature you will have to find another utility.

That may be enough.  If not **you** will have to write a program to get the full state of the affected GPIO and peripherals before pigpio is started and restore that state after pigpio is finisshed.

guymcswain, 2020-04-11:
If you are asking for pigpio to somehow 'share' a peripheral with an audio application, the answer is no.  But there are ways to finesse the configuration of pigpio to allow both to work.  For example, you could turn off alerts, thereby freeing up the PCM, and send waves using the PWM.  (Caveat: I have not verified this configuration.)

closed 2020-05-01 by guymcswain