joan2937/pigpio #338: Address offset between DMA channels 0-14

kripton, 2020-04-16:
Hi, I'm trying to learn from pigpio's source code in order to write a small utility to display the status of the Pi's DMA channels. Current status: Displays for each DMA channel if it is reserved by the firmware (via mailbox property request): https://github.com/kripton/PiDMAMon/

Next I would like to read the DMA channel control registers via direct memory read access (after using mmap).

Acording to the BCM2711's data sheet (https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf, chapter 4.2, page 41), the DMA registers for channels 0-14 are 0x100 bytes apart.
However, in your source code they are 0x40 bytes apart: https://github.com/joan2937/pigpio/blob/master/pigpio.c#L7334

Could you hint me on what I'm getting wrong? Thanks!

joan2937, 2020-04-16:
The source code is saying they are 0x40 words apart.  There are four bytes per word.

kripton, 2020-04-16:
Oh, yes, of course. Thanks!

closed 2020-04-16 by kripton