joan2937/pigpio #339: Consider O_CLOEXEC when opening FIFOs?

fandahao17, 2020-04-19:
Hi there! Thanks for your awesome project!

This is how I run into this problem: In `main()`, I started notification on FIFO `/dev/pigpio0` and forked another process which runs `pig2vcd` to output the captured waveform. But I found that in the child process, I can't close `/dev/pigpio0` even if I called `gpioNotifyClose`.

After going through the source code I found that by default, the [actual close is done in the alert thread](https://github.com/joan2937/pigpio/blob/7852103d13df7b061bb8afb2229e6cba7a120a93/pigpio.c#L12053), but `fork()` only copies the main thread. I read the man page and found that this problem can be fixed by including `O_CLOEXEC` when calling `open()`. Will you consider adding this flag when opening FIFOs?

guymcswain, 2020-04-19:
As an alternative approach, you could use `pigpiod_if2` to open notifications via the socket then fork `pig2vcd` from there.

joan2937, 2020-04-20:
I don't have any problem with the use of this flag.  Mind you I have only had a quick skim through the man pages.

guymcswain, 2020-05-01:
@fandahao17 , if you have a working proposal with test cases, please submit a pull request.