joan2937/pigpio #340: Getting segfault relating to libpigpio.so.1 when I run pigpiod or x_pigpio

AndyWas, 2020-04-19:
I'm building pigpio from source on Fedora Server 31 (64-bit kernel ver. 5.5.16-200) on RPi 3B+ following the procedure here: [.](http://abyz.me.uk/rpi/pigpio/download.html)  I need to use GPIO Zero, which uses pigpio as its pin factory (RPi.GPIO seems to be a non-starter due to the way Fedora declares itself in /proc/cpuinfo compared to Raspbian).  I had installed the Fedora packages **python3-gpiozero-1.4.1-3** and **python3-pigpio-1.40-3.post1** but these only install the Python modules; there is no pigpiod executable, and there doesn't appear to be any Fedora package that contains it.

**make** and **make install** completed cleanly, although libpigpio.so.1 was installed to /usr/local/lib and I had to edit /etc/ld.so.conf in order to avoid "error while loading shared libraries" when running x_pigpio or pigpiod (see [).](https://github.com/joan2937/pigpio/issues/113#issuecomment-362918666)  Now however when I run x_pigpio or pigpiod I get a segmentation fault - which still seems to implicate libpigpio.so.1:-

                                                
                                                Stack trace of thread 40757:
                                                #0  0x0000ffffbe718378 _IO_fclose@@GLIBC_2.17 (libc.so.6)
                                                #1  0x0000ffffbe887ef0 n/a (libpigpio.so.1)
                                                #2  0x0000ffffbe8ad620 gpioHardwareRevision (libpigpio.so.1)
                                                #3  0x0000ffffbe8add4c initInitialise (libpigpio.so.1)
                                                #4  0x0000ffffbe8b06c4 gpioInitialise (libpigpio.so.1)
                                                #5  0x0000000000401474 n/a (pigpiod)
                                                #6  0x0000ffffbe6d622c __libc_start_main (libc.so.6)
                                                #7  0x00000000004016d4 n/a (pigpiod)
                                                #8  0x00000000004016d4 n/a (pigpiod)

Grateful for any suggestions on where I may be going wrong or how to fix this.

Thanks,
Andrew

guymcswain, 2020-04-19:
Fedora on raspi + GPIO.zero + compiling pigpio from source - not a common usage so unlikely to get much help.

AndyWas, 2020-04-19:
I know...however I use Fedora for everything else so as they have a aarch64 build it seemed worth a try, and everything else works OK on Fedora.  I'm puzzled as to why Fedora has a maintained package for the pigpio Python components but not pigpiod itself as I can't see how they can work without it.

joan2937, 2020-04-20:
I have nothing to offer on this issue.  I know nothing of Fedora or 64-bit operation.

bennuttall, 2020-04-20:
> I'm puzzled as to why Fedora has a maintained package for the pigpio Python components but not pigpiod itself as I can't see how they can work without it.

Because you can run the client on one machine (doesn't have to be a Pi) and remotely control another Pi running the daemon. So you could run Python code on a Pi, or a PC running Windows/Mac/Linux and control a Pi running Raspbian that has the pigpio daemon running.

That said, in theory there's no reason why pigpiod couldn't run on Fedora on a Pi. Just an edge case here really - would take work from a keen third party to make it happen.

guymcswain, 2020-04-20:
To work-around Fedora /proc/cpuinfo issue you can try [this simple hack](https://www.raspberrypi.org/forums/viewtopic.php?p=1428987#p1428987).

closed 2020-05-01 by guymcswain