joan2937/pigpio #344: Trying to get minimal minimal_clk.c working on Raspberry Pi 4

bootrino, 2020-04-30:
Hi Joan,

Thanks for your great work.

In this thread, you posted "the minimum required to generate a 10MHz clock."
https://www.raspberrypi.org/forums/viewtopic.php?t=22274

Which I understand to be a cut down version of your minimal_clk.c program.

I have modified your code to try to make it work with the Raspberry Pi 4.

After I made the modifications I ran it and it doesn't do anything on the Pi 4.

So I then ran your unmodified code from the thread above on a Pi Zero and that didn't do anything either - the program just ends.

I wonder if you would mind seeing what I am doing wrong?

Here is my modification of your super minimal code, to work with the Pi 4:

```
/*
https://www.raspberrypi.org/forums/viewtopic.php?t=22274
"I cut down the code to (almost) the minimum required to generate a 10MHz clock."

10MHzClock.c
2014-10-24
Public Domain

gcc -o 10MHzClock 10MHzClock.c
sudo ./10MHzClock

*/

/*

This allows the setting of the three general purpose clocks.

The clocks are named GPCLK0, GPCLK1, and GPCLK2.

The clocks are accessible from the following gpios.

gpio4  GPCLK0 ALT0
gpio5  GPCLK1 ALT0 B+ and compute module only (reserved for system use)
gpio6  GPCLK2 ALT0 B+ and compute module only
gpio20 GPCLK0 ALT5 B+ and compute module only
gpio21 GPCLK1 ALT5 Not available on Rev.2 B (reserved for system use)

gpio32 GPCLK0 ALT0 Compute module only
gpio34 GPCLK0 ALT0 Compute module only
gpio42 GPCLK1 ALT0 Compute module only (reserved for system use)
gpio43 GPCLK2 ALT0 Compute module only
gpio44 GPCLK1 ALT0 Compute module only (reserved for system use)

Clock sources

0     0 Hz     Ground
1     19.2 MHz oscillator 
2     0 Hz     testdebug0
3     0 Hz     testdebug1
4     0 Hz     PLLA
5     1000 MHz PLLC (changes with overclock settings)
6     500 MHz  PLLD
7     216 MHz  HDMI auxiliary
8-15  0 Hz     Ground

The integer divider may be 2-4095.
The fractional divider may be 0-4095.

There is no 25MHz cap for using non-zero MASH
(multi-stage noise shaping) values.
[... 419 lines cut ...]
         "Bad, clkReg mmap failed\n");
      return -1;
   }
   fprintf(stderr, "end gpioInitialise\n");

   return 0;
}

/* ------------------------------------------------------------------------ */

int main(int argc, char *argv[])
{
   if (gpioInitialise() < 0) return 1;

   /* int clock, int source, int divI, int divF, int MASH */
   fprintf(stderr, "do initClock\n");
   initClock(0, 0, 4000, 0, 0);

   fprintf(stderr, "do gpioSetMode\n");
   gpioSetMode(4, PI_ALT0);

   return 0;
}

```

guymcswain, 2020-04-30:
Putting aside the fact that you are referring to code that is 5-6 years old and possibly not representative of what is in pigpio, have you tried to use this library to generate your desired clock frequency?

bootrino, 2020-04-30:
@guymcswain the value of this code for me is that Joan has cut it down to the absolute minimum needed to get the job done.

That's a big help for me in trying to understand how it works and modify it.  Frequency generation on the Raspberry Pi is pretty challenging stuff to wrap my head around and after alot of research and experimentation this piece of code is the key for me. 

I've got a goal of being able to understand and drive the clocks and frequency generation - this is the closest I've found to a "pure" solution that does nothing else which means I'll be able to understand it.

guymcswain, 2020-04-30:
We'll see what Joan comes back with.  I'm also trying to understand if this is a lib issue that we should track.

bootrino, 2020-04-30:
>> have you tried to use this library to generate your desired clock frequency

@guymcswain as a separate question though I would be appreciative & interested if you could direct me to which bit of the library that does the same should I investigate?

guymcswain, 2020-05-01:
> if you could direct me to which bit of the library ...

Sorry to say that I'm not up to speed on this area of the code.  I recommend you read the Broadcom 2835 document and then search the source for the labels associated with the clock generator base address.

closed 2020-06-28 by guymcswain

ddouglass, 2022-10-28:
For the benefit of anyone else who might end up reading these comments after a fruitless Google search:
https://github.com/LCOGT/minimal_clk