joan2937/pigpio #346: Cannot set GPIOs simultaneously in pigpiod

MarcelBeining, 2020-05-09:
I am brandnew to Pigpio. I want to set different colors of an RGB LED. The red channel is connected to GPIO19, the green to GPIO13 and the blue to GPIO12.

My expectation when I do
`
pigs m 19 w m 13 w m 12 w
`
and then
`pigs w 19 1 w 13 1 w 12 0`

that I should see some kind of yellow, however only RED is on. When I turn off red with
`pigs w 19 0`
suddenly GREEN is on.

I tested a bit, it seems pigpiod can always only set one GPIO and the rest is in some kind of queue???? How can I set multiple GPIOs at the same time?

Thanks in advance!

I have a Raspberry3b with Debian Stretch and pigpio v64

joan2937, 2020-05-09:
That should work.

A couple of possibilities.

- wrong connections
- LEDs need far more power than can be supplied so only one can be on at a time.

What does `pigs r 19 r 13 r 12` report

MarcelBeining, 2020-05-09:
Thanks for your answer!
I found the problem, I thought I can save 2  resistors for the LED circuit by placing only one resistor between the LED and Ground, thereby ignoring the fact that the current chooses the easiest way through the LED colors :-D Sorry for your time

closed 2020-05-09 by MarcelBeining