joan2937/pigpio #347: Consider more user friendly *Open functions

JakeSays, 2020-05-11:
The various Open() functions take a bit packed flag value that defines the options to be used during open. The usage of flags is a pain for the casual user, and generally an all around inconvenience.

I propose additional apis that provide a more convenient experience.

As an example I will use `spiOpen()`:

From the documentation:

spiFlags consists of the least significant 22 bits.
```
21 20 19 18 17 16 15 14 13 12 11 10  9  8  7  6  5  4  3  2  1  0
b  b  b  b  b  b  R  T  n  n  n  n  W  A u2 u1 u0 p2 p1 p0  m  m
```

This is extremely frustrating to have to deal with, especially when there is no need for it. Even on the smallest of architectures there are better options.

My suggestion would be a struct holding the individual parameters:

```
// For brevity not all necessary documentation is included in the comments.

typedef struct _spiParameters
{
	// If true the clock phase is leading edge.
	bool clockPhaseLeadingEdge;

	// If true the clock polarity is idle high.
	int clockPhaseIdleHigh;

	// If true CE is active high
	bool chipEnableActiveLow;

	// If true CE is active high
	bool reserveChipEnableGpio;

	// If true use the main SPI bus.
	bool useMainSpi;

	// If true the bus is 3 wire. Main SPI only.
	bool busIsThreeWire;

	// Number of bytes (0-15) to write before switching the MOSI line to MISO to read data.
	// This field is ignored 3 wire is not enabled. Main SPI only.
	int bytesToWriteBeforeRead;

	// If true the least significant bit is transmitted on MOSI first.
	// Auxiliary SPI only.
	bool writeLsbFirst;

	// If true the least significant bit is received on MISO first.
	// Auxiliary SPI only.
	bool readLsbFirst;

	// Defines the word size in bits (0-32). 
	// Auxiliary SPI only.
	int wordSize;

} SpiParameters;

int spiOpenParameters(int channel, int baud, SpiParameters& params)
{
	// convert parameters to flags
	unsigned int flags = ...
	return spiOpen(channel, baud, flags);
}

```

joan2937, 2020-05-13:
C may have changed in the years since I started.  Bit fields in structs were/are not standardised.  In fact as far as I know they are implementation defined.

What you suggest would not work across different compilers or languages (the current API can be used by any compiler and language).

closed 2020-05-13 by joan2937

JakeSays, 2020-05-13:
@joan2937 first, that's not true - I wasn't suggesting bit fields. I was suggesting bool (C99) or int if that won't work. Bit fields are an unnecessary optimization.
second, pretty much any language on the planet can handle C structs.
third, i wasn't suggesting replacing the existing api - just adding to it - so the impact is zero.