joan2937/pigpio #348: SPI transfer size limitation?

dom11990, 2020-05-14:
Hello,

I'm trying to use this library to write an FPGA bitfile in one go, but it appears that the maximum transfer size is limited by 

#define PI_MAX_SPI_DEVICE_COUNT (1<<16) -> 65k

I need to send roundabout 200kB. Is there any particular reason that this limitation is in the code? Since unsigned is used as the count type 2^32-1 bytes should be "supported" (even though the ram will likely limit this)

cheers
dom

guymcswain, 2020-05-14:
Are you using the daemon or linking the library in C?  I believe there is a hard limit of 65K imposed by the daemon for all buffer transfers (not just SPI).

dom11990, 2020-05-14:
I'm linking the library in C.

joan2937, 2020-05-14:
Why not just change the constant and rebuild pigpio and see if it works?

The software should run ok once that change is made.

The only point of the constant is to try and trap errors so the range is limited to what I thought would be a reasonable maximum size.

I'm neutral as to whether it's worth changing the constant in pigpio proper for one known use case.

dom11990, 2020-05-14:
@joan2937 I can certainly change it and see if it works and would also be happy to create a PR for you. I just wanted to check if perhaps there was some hardware compatibility reason that required this to be hard coded. As per making the change in the library proper, I would argue in favor of it since it only expands the applicability of the library with no detriment. It is standard operating procedure for a C library to allow you to shoot yourself in the foot if you pass a bad buffer size that you didn't intend, after all :) Thoughts?

closed 2020-06-28 by guymcswain