joan2937/pigpio #349: python pi.get_current_tick() always return 0

wang-bc, 2020-05-20:
I'm using python connect to pigpiod sever on local machine. Doesn't work correctly.
Library [ircodec](https://github.com/kentwait/ircodec) couldn't got response from pigpiod.
And always return zero when i run pi.get_current_tick().

joan2937, 2020-05-20:
Please provide a short Python script which shows this problem.

wang-bc, 2020-05-20:
Script like follow
```python
import time
import pigpio

pi = pigpio.pi()
print(pi.connected)

for i in range(10):
    print(pi.get_current_tick())
    time.sleep(1)
```
And i got result like this
[image]

joan2937, 2020-05-20:
Can you add the result of `cat /proc/cpuinfo` and `pigs pigpv` ?

wang-bc, 2020-05-20:
```sh
cat /proc/cpuinfo

processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03112
Serial		: 10000000af73b517
Model		: Raspberry Pi 4 Model B Rev 1.2

```

```sh
pigs pigpv

2020-05-20 21:24:11 pthSocketThread: Connection accepted on socket 4
2020-05-20 21:24:11 pthSocketThread: SO_KEEPALIVE enabled on socket 4

68
2020-05-20 21:24:11 pthSocketThreadHandler: Socket 4 closed
```

wang-bc, 2020-05-20:
My pigpio is installed by apt-get. May be its version is too old.

wang-bc, 2020-05-20:
Problem solved after upgrade to latest verison.

closed 2020-05-20 by wang-bc